indilib/indi #1262: Primaluce Sesto Senso 2 - user profile / hold current control support

gilesco, 2020-11-17:
**Is your feature request related to a problem? Some telescopes require a Hold Current to be set in order to prevent slippage. The Sesto Senso 2 supports setting such a current, but it is not available to be set in the driver.
Currently the driver loads the factory preset "Slow" profile and this is hardcoded into the driver source code.

Ideally, I would like to be able to configure the all available settings through the driver, and also be able to load and save user profiles (There are 3 user profile slots available).
If not possible, at least be able to select an existing user profile (we can use the Wifi connection to create and configure these profiles.

I am yet to use the Sesto Senso 2, but I intend to create my own profile and then modify the code to select it instead of the "slow" profile.

knro, 2020-11-17:
Thank you for the feedback. Can you submit a PR for this?

gilesco, 2020-11-18:
Ah, I'm not really sure of how to implement any functionality, all I know is that if I set up a user profile called "Esprit" outside of the driver that I can substitute "slow" in line 1028 of sestosenso2.cpp for "Esprit" and it will probably load my profile instead of the slow profile:

bool CommandSet::loadSlowPreset(char *res)
{
    return sendCmd("{\"req\":{\"cmd\":{\"RUNPRESET\":\"Esprit\"}}}", "RUNPRESET", res);
}

On my unit {"req":{"get": ""}}} results with:

{"res":{"get":{"SWID":"001","MODNAME":"SESTOSENSO2","SN":"SESTOSENSO20716","HOSTNAME":"SESTOSENSO20716","MACADDR":"70:B3:D5:32:95:2F","SWVERS":{"SWAPP":"1.3","SWWEB":"1.3"},"LANCFG":"ap","LANSTATUS":{"LANCONN":"disconnected","PINGGW":0,"PINGEXTHOST":0},"WIFIAP":{"SSID":"SESTOSENSO20716","PWD":"primalucelab","IP":"192.168.4.1","NM":"255.255.255.0","CHL":2,"MACADDR":"70:B3:D5:32:95:30","STATUS":{"HIZ":0,"UVLO":0,"TH_SD":0,"TH_WRN":0,"OCD":0,"WCMD":0,"NOPCMD":0,"BUSY":0,"DIR":"In","MST":"stop"}},"WIFISTA":{"SSID":"MySSID","PWD":"MyPassword","IP":"","NM":"","GW":"","DNS":"","MACADDR":"70:B3:D5:32:95:2F"},"EXT_T":"-127.00","VIN_12V":"13.98","ARCO":0,"LEDBLINKSTATUS":"PowerInLess11Volts","PRESET_1":{"NAME":"","M1POS":0},"PRESET_2":{"NAME":"","M1POS":0},"PRESET_3":{"NAME":"","M1POS":0},"PRESET_4":{"NAME":"","M1POS":0},"PRESET_5":{"NAME":"","M1POS":0},"PRESET_6":{"NAME":"","M1POS":0},"PRESET_7":{"NAME":"","M1POS":0,"M2POS":0},"PRESET_8":{"NAME":"","M1POS":0,"M2POS":0},"PRESET_9":{"NAME":"","M1POS":0,"M2POS":0},"PRESET_10":{"NAME":"","M2POS":0},"PRESET_11":{"NAME":"","M2POS":0},"PRESET_12":{"NAME":"","M2POS":0},"DIMLED_1":200,"DIMLED_2":200,"DIMLED_3":200,"MOT1":{"ABS_POS":18075,"EL_POS":0,"MARK":0,"SPEED":0,"ACC":5,"DEC":5,"MAX_SPEED":4,"MIN_SPEED":0,"TVAL_HOLD":2,"TVAL_RUN":20,"TVAL_ACC":20,"TVAL_DEC":20,"T_FAST":25,"TON_MIN":41,"TOFF_MIN":41,"FS_SPD":1023,"OCD_TH":2,"STEP_MODE":4,"CONFIG":11904,"ALARM":255,"STATUS":{"HIZ":1,"UVLO":0,"TH_SD":0,"TH_WRN":0,"OCD":0,"WCMD":0,"NOPCMD":0,"BUSY":0,"DIR":"Out","MST":"stop"},"NTC_T":"37.12","CAL_MAXPOS":17639,"CAL_BKLASH":0,"CAL_MINPOS":0,"CAL_HOMEP":1000,"CAL_NUMOF":0,"CAL_DIR":"normal","LASTDIR":0,"LOCKMOV":1,"HOLDCURR_STATUS":1,"FnRUN_ACC":1,"FnRUN_DEC":1,"FnRUN_SPD":2,"FnRUN_CURR_ACC":7,"FnRUN_CURR_DEC":7,"FnRUN_CURR_SPD":7,"FnRUN_CURR_HOLD":3},"RUNPRESET_L":{"RP_NAME":"light","M1ACC":10,"M1DEC":10,"M1SPD":10,"M1CACC":3,"M1CDEC":3,"M1CSPD":3,"M1HOLD":1},"RUNPRESET_M":{"RP_NAME":"medium","M1ACC":6,"M1DEC":6,"M1SPD":6,"M1CACC":5,"M1CDEC":5,"M1CSPD":5,"M1HOLD":1},"RUNPRESET_S":{"RP_NAME":"slow","M1ACC":1,"M1DEC":1,"M1SPD":2,"M1CACC":7,"M1CDEC":7,"M1CSPD":7,"M1HOLD":1},**"RUNPRESET_1**":{**"RP_NAME":"Esprit"**,"M1ACC":1,"M1DEC":1,"M1SPD":2,"M1CACC":7,"M1CDEC":7,"M1CSPD":7,"M1HOLD":3},"RUNPRESET_2":{"RP_NAME":"User 2","M1ACC":6,"M1DEC":6,"M1SPD":6,"M1CACC":5,"M1CDEC":5,"M1CSPD":5,"M1HOLD":1},"RUNPRESET_3":{"RP_NAME":"User 3","M1ACC":1,"M1DEC":1,"M1SPD":2,"M1CACC":7,"M1CDEC":7,"M1CSPD":7,"M1HOLD":1},"MOT2":{"ABS_POS":0,"EL_POS":0,"MARK":0,"SPEED":0,"ACC":35,"DEC":35,"MAX_SPEED":30,"MIN_SPEED":0,"TVAL_HOLD":1,"TVAL_RUN":18,"TVAL_ACC":18,"TVAL_DEC":18,"T_FAST":25,"TON_MIN":41,"TOFF_MIN":41,"FS_SPD":1023,"OCD_TH":2,"STEP_MODE":4,"CONFIG":11904,"ALARM":255,"STATUS":{"HIZ":1,"UVLO":0,"TH_SD":0,"TH_WRN":0,"OCD":0,"WCMD":0,"NOPCMD":0,"BUSY":0,"DIR":"In","MST":"stop"},"CAL_MAXPOS":0,"HSENDET":1,"HEMISPHERE":"northern","COMPENSATION_DEGS":0.00000,"CAL_STATUS":"stop","CAL_MINPOS":0,"CAL_HOMEP":1000,"CAL_NUMOF":0,"CAL_DIR":"normal","LASTDIR":0,"LOCKMOV":1,"HOLDCURR_STATUS":0},"DIMLEDS":"on","LOGLEVEL":"no output"}}}

So I suppose a good starting point would be to somehow parse that and obtain the "RUNPRESET_1", "RUNPRESET_2" an d "RUNPRESET_3" RP_NAME strings, and populate them into some drop down list in the INDI control panel. perhaps together with the builtin light, medium and slow profiles.

This would at least allow a user to first create a profile by other means, store it in the Sesto Senso 2, and then load it in the INDI configuration.

It is somewhat beyond my programming capabilities.

gilesco, 2021-01-12:
I think https://github.com/indilib/indi/pull/1319 solves this issue.

closed 2021-04-17 by knro