indilib/indi #1263: Allow for periodic automatic astrometry alignment

federicoparra, 2020-11-18:
Long exposure astrophotography is overrated and complicated. 
Short exposure astrophotography can reach great results even with an Alt-Az mount, see for example this pic of Orion's nebula the fruit of 100 stacked 10 sec pics at 800 ISO with a Nikon D5100 through a SkyWatcher 127mak and an AZ-GTI mount in Alt-Az. 

[image]

I believe Ekos could accommodate folks doing short exposure better by adding some simple and very useful features.

In this case, I ask for the capacity to add periodic astrometry alignments to keep the object centered throughout.

Since short exposure astrophotography typically is done without guiding, and since each exposure is short, it would be cool that every say 20 pics Ekos runs an alignment and then continues.

To loose as little time as possible, plate solving could be performed on a given picture without stopping the routine (i.e. keep taking pictures as astrometry resolves). Once resolved, Ekos waits for last picture taken to download from the camera, then slews and enters an alignment cycle until alignment threshold reached, then shooting continues.

This would allow for setting a short exposure routine lasting hours yet keeping the object always centered.

bluthen, 2020-11-19:
I think you can do this with the scheduler. And specify a existing fits file so it goes to that exact coordinates.

TallFurryMan, 2020-11-19:
As @bluthen says, that workflow is already possible with the Scheduler. But, agreed, it is not as efficient as a dedicated workflow could be.

This situation should improve when we rework the Capture pipeline to better accommodate for custom processes, such as electronically assisted astronomy.

If you want to discuss and propose improvements on Ekos and KStars, you should go to https://invent.kde.org/education/kstars. This tracker is for INDI.

closed 2020-11-19 by knro