indilib/indi #1270: indi_lx200_10micron: setting Long in telescope location

fenriques, 2020-11-30:
On a 10Micron mount, but probably on other drivers as well, when entering a Longitude value in 'Site Management', the driver updates the mount coordinates but skips the seconds (ss) value.
For example, entering '02:38:30' in Ekos, will results in 'E002 39 00' (and not 'E002 38 30') on the 10micron handpad.

Latitude behavior is different: seconds are stored correctly.

Looking in the INDI log file, the reason seems to be different decimal places for Long (5) and Lat (only 2) when setting the **Observer location** (second line of the log below).

`[2020-11-27T19:03:52.453 CET INFO ][           org.kde.kstars.indi] - LX200 10micron :  "[INFO] Site location updated to Lat  37:44:24 - Long    2:38:30 "`
`[2020-11-27T19:03:52.458 CET INFO ][           org.kde.kstars.indi] - LX200 10micron :  "[INFO] Observer location updated: Longitude (2.63556) Latitude (37.74) "`

Tested using Kstars 3.50, Indi 1.8.7 but was also on previous versions.

I wasn't able to find in the source code where the Observer location is set.
Ferrante

knro, 2020-11-30:
Thank you for the report. The reason is that 10Micron is based off LX200 command set. If you check in [lx200driver:1020](https://github.com/indilib/indi/blob/5abd3d5cb7f39f1411c4647b8f5cfc51e82f0b82/drivers/telescope/lx200driver.cpp#L1020) you'll see it only sets degrees and minutes. The 10Micron driver should probably override [updateLocation](https://github.com/indilib/indi/blob/5abd3d5cb7f39f1411c4647b8f5cfc51e82f0b82/drivers/telescope/lx200telescope.h#L89) function to send the full sexagesimal representation correctly to the mount.

fenriques, 2020-11-30:
thanks Jasem. But if nothing else breaks, wouldn't it be better to fix  lx200driver::setSiteLongitude method directly? 
I'm not much into this class hierarchy, but other lx200-based drivers are affected as well by this issue.

knro, 2020-11-30:
The Deg:Min convention is used by the official LX200 command set. It's possible to perhaps check if long format is supported and then send the seconds accordingly like it's done for the equatorial coordinates.

knro, 2020-12-12:
@d33psky Please check this out

d33psky, 2020-12-23:
I'll be using the `controller_format` to switch styles (short, long, longer formats currently)

d33psky, 2020-12-23:
First part of the fix https://github.com/indilib/indi/pull/1296

Can I get access to link the PR to this ticket on the right ?

d33psky, 2020-12-30:
and the 3rdparty PR -> https://github.com/indilib/indi-3rdparty/pull/272

fenriques, 2021-01-04:
I've been testing the fix on the last few night. Everything works fine (thank Hans) so I'm closing this issue.

closed 2021-01-04 by fenriques