indilib/indi #1274: EQ8-R autohome - permanent fix found?

xthestreams, 2020-12-04:
Posting @seescho solution to the problem - mount code 0x20
https://sternenkarten.com/2020/11/23/umstellung-auf-die-eq8-r-pro/

_Originally posted by @xthestreams in https://github.com/indilib/indi/issues/1162#issuecomment-738545878_

knro, 2020-12-09:
It's fixed in 3rd party now.

closed 2020-12-09 by knro

xthestreams, 2020-12-09:
amazing!

xthestreams, 2020-12-10:
working great from what i'v been able to test

seescho, 2020-12-12:
great!

xthestreams, 2020-12-13:
I've had to do a rebuild of my system, I've noticed that the driver that comes with teh disc-upgrade does not have the fix, I'll check with teh source and advise if still not working.

xthestreams, 2020-12-13:
I just rebuilt from source, does not appear to be fixed, has the commit happened?

knro, 2020-12-14:
Make sure you're rebuilding indi-eqmod driver as well.

fcasarramona, 2021-06-25:
Just installed version 1.9.1~202106241219~ubuntu18.04.1 and it seems not fixed.

2021-06-25T07:23:40: [WARNING] Warning: Invalid command -> Incorrect mount type 
2021-06-25T07:23:40: [INFO] AutoHome phase 1: reading home position indexes to set directions 
2021-06-25T07:23:40: [INFO] AutoHome phase 1: resetting home position indexes 
2021-06-25T07:23:40: [INFO] AutoHome phase 1: turning off aux encoders 
2021-06-25T07:23:40: [INFO] Starting Autohome. 
2021-06-25T07:23:39: [WARNING] *** AutoHome NOT TESTED. Press PERFORM AGAIN TO CONFIRM. ***