indilib/indi #1276: indi_lx200_10micron add interface to manage unattended flip setting

d33psky, 2020-12-08:
Note to self: use these two 10micron extensions to the LX200 protocol:

> :Guaf#
> Returns the unattended flip setting.
> Returns: 0 disabled
> 1 enabled
> Available from version 2.11.
> Note: unattended flip didn't work properly in firmware versions up to 2.13.8 included.
> 
> :SuafN#
> Note
> Enables or disables the unattended flip. Use N=1 to enable, N=0 to disable. This is set
> always to 0 after power up.
> Returns: nothing
> Available from version 2.11.
> unattended flip didn't work properly in firmware versions up to 2.13.8 included.

This is needed to let software like EKOS manage the meridian flip. Thanks to Ferrante for bringing this to my attention.

d33psky, 2020-12-25:
PR -> https://github.com/indilib/indi/pull/1298

fenriques, 2021-01-04:
Suggest to close also this issue.
Based on my tests the unattended flip now works just fine.

closed 2021-01-04 by knro