indilib/indi #1291: High speed GPS tracking with Losmandy Gemini mount

livi0, 2020-12-18:
This is related to the code made by Jasem Mutlaq:
https://indilib.org/telescopes/losmandy.html
I'm planning to track with a G-11 Losmandy mount a light source on a radiosonde balloon. It moves with an angular speed of a few thousands arcsec/sec near the ground to few hundreds arcsec/sec at higher altitudes. The radiosonde sends wirelessly its GPS lat-long coordinates, that I will convert in real-time into RA/DEC coordinates. In order to keep it in my field of view the light has to stay within 30 arcsec of the real GPS coordinates. I need therefore the mount to update its pointing every 0.1 sec. If there is a pointing bias near the zenith, I can take that bias into account to keep the light centered near that particular pointing direction. Any idea, please, if this software is able to do 0.1 sec pointing updates, while disengaging the sidereal tracking? Is there any manual for this software, please? I plan to install it on the Windows 10 Subsystem for Linux – I guess it’s based on Ubuntu. Any idea if it works there, please? Thanks!

charlesmarseille, 2021-08-04:
Hi Liviu,
I managed to control the Gemini1 with the serial commands (:P#, :GD#, etc) as per the user manual directly from windows 10 with the usb to serial adapter. Considering the mount can now move with simple output to the serial port and readings from the gemini are read, I can test your requirement of 10hz movements, although I doubt the mount can indeed move that fast (back and forth instead of the usual continuous movements). I will post back the results.
Connection and control from KStars also works in Ubuntu 18.04, but a deep dive in the code is needed to make the simple movements we are looking for. Might as well control via serial directly.

knro, 2021-11-24:
I believe for this to work, there needs to be support at the firmware level as you can't expect clients to send rapid updates without significant lag over serial.

closed 2021-11-24 by knro

charlesmarseille, 2021-11-24:
thank you knro!