indilib/indi #1294: Nikon and Gphoto drivers broken with last update in beta channel

federicoparra, 2020-12-20:
I'm a Stellarmate user and just got the latest OS/app updates that include latest versions of Indi drivers (using the beta channel).
For the first time since using Indi, DSLR Nikon connection is broken, with both the dedicated Nikon driver and the generic Gphoto driver.
Broken means, INDI connects to the camera, detects the right model, but can't take a picture or do live streaming - I'm using of course the same camera, Nikon D5100 as before, and I tried all obvious things (changing USB port in Raspberry, disconnecting and reconnecting a million times, and so forth).
I never got this problem or anything like it before - since the very first time using INDI, connecting to this Nikon was just connect and boom, it could take pics and do live video.

knro, 2020-12-21:
Stable INDI release works fine though? No changes were made to INDI GPhoto since the last stable release of INDI though. Let me check against latest libgphoto2 and latest INDI.

federicoparra, 2020-12-21:
Have not tried going back to the stable channel, but if I do I'll lose the ability to use the latest AZ-GTI alt-az driver right?

federicoparra, 2020-12-22:
The error is "Exposure failed to save image...Unspecified error"

federicoparra, 2020-12-22:
Unrelated but important @knro  I'm getting this error a LOT since the updates (second screenshot shows that Ekos is running properly)
[image]
[image]

federicoparra, 2020-12-29:
Nevermind, the problem is that the new version of nikon/gphoto driver's force bulb switch does work, whereas the previous versions the switch did not had an effect for me. Thus since my Nikon D5100 does not allow bulb from USB cable, it was causing errors; once I switched force bulb to off, it started working fine again.

closed 2020-12-29 by federicoparra