indilib/indi #1299: Mount device configuration is always saved twice

d33psky, 2020-12-26:
This happens probably for all LX200 mounts. Below paste is from a 10Micron mount.

2020-12-25T20:10:15: [INFO] Device configuration saved. 
2020-12-25T20:10:15: [INFO] Saving device configuration... 
2020-12-25T20:10:15: [INFO] Device configuration saved. 
2020-12-25T20:10:15: [INFO] Saving device configuration... 

Just once is enough really ;-)

knro, 2020-12-26:
I think one reason could be perhaps that there saveConfig for specific properties and saveConfig for all properties.

knro, 2021-03-12:
The issue was tracked down to Serial::processHandshake since it always saves config when it is successful. A solution is to check whether the device port and baud rate are different and then skipping the saveConfig call altogether.

knro, 2021-05-28:
This is fixed in master.

closed 2021-05-28 by knro