indilib/indi #1306: https://www.indilib.org/ down

rickbassham, 2020-12-30:
**Describe the bug**
The https://www.indilib.org/ website appears to be down.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Visit https://www.indilib.org/

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

xsnrg, 2020-12-30:
It is fine if you do not use "www". Just go to https://indilib.org

rickbassham, 2020-12-30:
Is this new? Wonder how my bookmark became bad... Might just be user error.

xsnrg, 2020-12-30:
Not sure if it happened when the cert was renewed or not. I have always just used it without "www".

rickbassham, 2020-12-30:
Closing. We'll call this user error. I've updated my bookmark.

closed 2020-12-30 by rickbassham