indilib/indi #1311: No EQUATORIAL_COORD standard property in mount drivers

marcocipriani01, 2021-01-04:
I'm working on the [INDIForJava](https://github.com/INDIForJava/INDIForJava) and [IPARCOS](https://github.com/INDIForJava/IPARCOS) (an INDI client for Android) projects and I've noticed that the INDI EQMod driver doesn't have the `EQUATORIAL_COORD` [standard property](https://indilib.org/develop/developer-manual/101-standard-properties.html). Instead, it only has the `EQUATORIAL_EOD_COORD` property. The difference between the two is the epoch of the equatorial coordinates, **J2000** and **JNow** respectively.
The problem is that IPARCOS currently uses the NGC and Common Stars catalogs to point the telescope (the app has a simple go-to interface). Those catalogs use J2000 equatorial coordinates, and setting them to the driver's `EQUATORIAL_EOD_COORD` property means pointing the telescope in a slightly different place, which is what the app does at the moment.
I can't test other INDI mount drivers because I only have an EQMod mount, but I suspect that most of the drivers don't implement `EQUATORIAL_COORD` as well.
Would it be possible to add this missing standard property to all the mount drivers? If no, does anyone know the formula to convert from J2000 to JNow?

Kind regards,
Marco Cipriani

TallFurryMan, 2021-01-05:
There are some conversion tests in KStars : https://github.com/KDE/kstars/blob/f5e4e983604e62f8b462d0444b378b28b820f962/Tests/skyobjects/test_skypoint.cpp

marcocipriani01, 2021-01-05:
Those tests use KStars' internal utilities and calculators, which I can't use from INDIForJava

TallFurryMan, 2021-01-05:
Yes, certainly, but it leads you to the actual algorithm that you may then port in your environment. Else, the ASCOM documentation, if I remember correctly, also has the conversion procedures between JNow and J2000.

This said I agree the keyword could be added from INDI.

marcocipriani01, 2021-01-05:
It took me some time but, in the the end, I decided to study "Astronomical Algorithms" by Jean Meeus instead of looking for the formulas in the KStars code. Now I get precise coordinates when using IPARCOS.

> This said I agree the keyword could be added from INDI.

I agree too, it could be useful for other INDI clients or for somebody who wants to point the telescope using his own catalog.

TallFurryMan, 2021-01-06:
Ah you can't get wrong with Meeus 👌

closed 2021-01-11 by marcocipriani01