indilib/indi #1329: celestrongps reports wrong guide rates (to PHD2) at wrong scale

interplanetarychris, 2021-01-20:
I'm not quite sure where to reference it in the code, but it would appear that the Celestron GPS INDI driver is reporting RA/DEC guide rates at a percentage number from 0-100, when applications like PHD2 are expecting it as a fraction between 0-1. 

While it doesn't prevent guiding from working, PHD2 ends up using much smaller pulses than it needs to, and reports incorrect information about the capability of the mount.

https://github.com/indilib/indi/blob/d6e9d74a82bbf6914709bab7063bb9bd68f45687/drivers/telescope/celestrongps.cpp#L1250

I've reported this to the PHD2 developers, but they indicate that it is not compliant with the "interface standards" and that it should be fixed in the INDI driver.

knro, 2021-01-20:
Right, the guide rate scale should be from 0 to 1. Can you submit a PR to fix that?

interplanetarychris, 2021-01-20:
> Right, the guide rate scale should be from 0 to 1. Can you submit a PR to fix that?

Sorry, I don't follow the code well enough to distinguish what's used internally, what's interfaced to the mount itself, and what's reported externally.

interplanetarychris, 2021-01-20:
I took my best attempt @knro  - hopefully a code review is quicker than making the change yourself!

Chris-Rowland, 2021-01-21:
The guide rate as a percentage of sidereal is done to match what Celestron use in their HC.  I think this should remain for the UI at least.

I've had a look in the INDI documentation and can see nothing about how the guide rate should be specified, or even any mention of guide rate.  I could have missed this, where is this defined?

Where are the "interface standards" that PHD mention defined and published?  Is this "standard" just what PHD require?

It is quite likely that this is not implemented according to the "standard" for other drivers as well.

Chris

interplanetarychris, 2021-01-21:
@Chris-Rowland 
I've asked Bruce with PHD2 to point to their preferred standard.  He had initially pointed to the INDI documentation "SLEW_RATE.  I don't see that the INDI API clearly indicates guide rate as a variable, or its units.

Also, can you give me a pointer on INDI logs?  In Kstar/Indi, if I enable Debug mode in the driver, after the log shows it being "Enabled", it immediately becomes "Disabled."

interplanetarychris, 2021-01-21:
Updated the original Issue description with the PHD2-referenced standard: https://indilib.org/developers/developer-manual/101-standard-properties.html#h3-telescopes

ken-self, 2021-01-21:
Looks to me like GUIDE_RATE is a de facto standard. I cannot see it in Telescope:: class so there will be inconsistency across drivers.  eg. In indi-avalon it is implemented as GUIDING_SPEED.
It would make sense to make a published standard and include in Telescope::

interplanetarychris, 2021-01-22:
I've tested the updated code against my mount and confirmed that i appropriately updates RA and DEC guiding rates per INDI settings, and that it looks appropriate in the Hand Controller (HC).    It seems to show up 1% lower than the setting due to the 8-bit scaling implementation currently in the INDI driver.

After trying it in all the various Track/Initialization states, the only time that the Guide rate is ever updated in INDI is when it is explicitly set using the GUI.  It does not appear that in my configuration (a Celestron CGX with HC version 5.31) that it ever reads the mount values into INDI variables, even though the following code in `CelestronGPS::updateProperties() ` should accomplish the read:

https://github.com/indilib/indi/blob/fa59b727825146091206ea9c7e9f3c0d7372ed2b/drivers/telescope/celestrongps.cpp#L429-L444

Furthermore, PHD2 appears to use whatever value is "current" in the largest value in the `GuideRateNP` vector.  If a guide value has not been set in the INDI/Ekos GUI, GuideRateNP=0, and PHD2 uses its minimum guide rate value of 0.10x

I'm currently examining the code to try and figure out how to appropriately implement a READ of the mount Guide rate values for my configuration, although my capabilities in the object-oriented aspects of C++ are nil, so its slow-going.

interplanetarychris, 2021-01-22:
After some investigation, it appears that the following function is called at appropriate times, but ALWAYS returns 0 for my configuration.  Not sure why yet...

https://github.com/indilib/indi/blob/09420ebb4bbfe971eba077d5c3e164d0248e7be7/drivers/telescope/celestrondriver.cpp#L563-L579

Chris-Rowland, 2021-01-22:
The TelescopeSlewRate enum has a SLEW_GUIDE enumerator but there doesn't seem to be any definition of the rate in any of the common code.
The best definition is in telescopesimulator.cpp where the guide rate switches define rates from 0.1 to 1 times sidereal.  That's not really a standard.
If this is to be defined it needs to be in the GuiderInterface definition, with the other guide stuff.

nrackwit, 2021-02-07:
It looks as if PHD2 is now receiving too small guide rates.  I incidentally came across this yesterday while trying to guide my AVX against a recent build of Indi master. I noticed that after PHP connected using my normal  profile there was an unusual notification of a change in settings and PHD2 requested a calibration. The calibration subsequently failed for to too little west movement. At first I was quite confused until I found that the guide rates in the Indi mount control tab were initialized to zero and had changed from a 0 - 100 scale to a 0 - 1 scale and then also your brief discussion around the odd Celestron guide rates from the PHD2 forum came to my mind so I figured there likely was a change to the telescope driver. I didn't look further into this at the time but switched to another computer that had a stable release with which I was able to calibrate and guide my setup. Thus unfortunately I didn't gather much helpful debug logs, screen shots, etc.

The one thing that do have to share are the PHD2 logs from the failing session with Indi compiled some time last week from master (failed guiding - https://openphdguiding.org/logs/dl/PHD2_logs_sCLd.zip) and a successful one from Indi out of the  OS X KStars 3.5.1 Stable release (successful guiding - https://openphdguiding.org/logs/dl/PHD2_logs_FC7e.zip)

With your updated code I can see that the guide speed looks 'correct' (7.5as/s), but PHD2 produces no (only random) star movement:

**Calibration Begins at 2021-02-06 19:40:33
....
RA Guide Speed = 7.5 a-s/s, Dec Guide Speed = 7.5 a-s/s
...**

The old code seems to list the classical 'incorrect' guide speed, but here I see the star noticeably changing position based on PHD2's guide steps and the calibration succeeds.

**Calibration Begins at 2021-02-06 21:04:49
....
RA Guide Speed = 748.0 a-s/s, Dec Guide Speed = 748.0 a-s/s
...**

Hope this helps!

Kind regards
Niels

knro, 2021-02-08:
That's exactly related to this change, it's two orders of magnitude. @interplanetarychris can you please check what's going on?

interplanetarychris, 2021-02-10:
I believe I've found and fixed this bug, including some brief night-sky testing last night.  The guiding rate command is included with the `send_guide_pulse` command, and the scale of that rate was not correct, effectively resulting in a zero-length guide pulse.

I'm currently investigating why get_guide_rates is reporting zero, and will post a PR sometime today with the fixes/work-arounds.

nrackwit, 2021-02-11:
I took a brief look at the code, and before you go down the wrong path I wanted to share what I'm seeing. I think the problem is not so much scale of rate but implicit casting which is setting your guide rate to '0'. It happens in two places. The first one is in CelestronGPS::Guide(..):

[image]

This code is casting a float/double which is between 0..1 to uint8 so that is where I'd expect the zero guide rate to originate from. Your code pasted above notes in the comment that 'rate' takes values between 0..255 (and that's also what I gather from the code), thus you'll probably need to do something like this

            rate = guideRateDec = static_cast<uint8_t>(std::min(GuideRateN[..].value * 256.0, 255.0));

I'm a bit puzzled by the old code, too. I'm not sure about the intention to initialize rate = 50.

Another similar effect may take place in CelestronGPS::updateProperties():

[image]

By dividing two int here the compiler would implicitly cast the result which is in 0..1 again also to int thus now the UI rate is set to zero as well. Overall, with the rates now being between 0..1 the code gets tricky to read code so I would be as explicit as possible:

                GuideRateN[AXIS_RA].value = static_cast<double>(rate) / 255.0;

On another thought - I think it might make sense to somehow provide a console readout or such to make the user aware of the rates reporting to the guider having changed in scale when connecting to the guider/telescope. I think that in my case the reason why PHD2 had asked me to recalibrate was that it had correctly detected a change in guide rate and figured that the calibration data it had would no longer work. That is probably going to be the case for any guider software which has the ability to retain and reuse calibration data and would use the updated driver. I could see how that could leave users confused as they probably would not expect to needed to recalibrate their guiding after having upgraded Indi. 

Hope I am helping.

Cheers
Niels

interplanetarychris, 2021-02-15:
Closed with merge of PR https://github.com/indilib/indi/pull/1353

closed 2021-02-15 by interplanetarychris