indilib/indi #1330: CelestronGPS and Nexstar+ vs Starsense and location/time setting

enolan94, 2021-01-20:
Not sure this would be considered a bug or enhancement or clarification request.
In the CelestronGPS driver there are checks for isAligned if either location or time set is attempted.
The comment in these sections calls out for Starsense but does not actually check if the HC is Starsense.
Quoting source:
// Update Location and time are disabled if the mount is aligned.  This is because
// changing either will change the mount model because at least the local sidereal time
// will be changed. StarSense will set the mount to unaligned but it isn't a good idea even
// with the NexStar HCs

The issue I ran into is when using PARK with Hibernate enabled is on the next powerup, how do I get the correct time into the mount if trying to run remotely.

Yes, I can unpark and enable tracking but the time in the HC is still the 'last' time power was removed.

I have found, using the latest Nexstar+ firmware, that setting ''time' works correctly even if mount is reporting 'isAligned'.

What I have found is on powerup and prior to unparking/enable tracking I connect and set the time, the HC takes the  set but indi emits the confusing message (to me anyway):

08:39:45.018: [INFO] Mount supports guiding.
08:39:45.018: [INFO] Mount UTC offset: -5.00. UTC time: 2021-01-20T13:39:43. DST: Off  == ('OLD TIME)
08:39:45.020: [WARNING] Mount is NOT aligned. You must align the mount first before you can use it. Disconnect, align the mount, and reconnect again.

If I then disconnect, reconnect and unpark/enable tracking all is good with new time.

How should all this work using Indi?

Thank you all who make Indi and all supported pieces possible!!!

Gene

knro, 2021-01-20:
@Chris-Rowland Thoughts?

Chris-Rowland, 2021-01-21:
Changing the time will, I believe, cause the mount alignment to change because a time change will change the hour angle, which is a vital part of the conversion between hour angle - which is what the mount uses - to Ra.   It is easy to check this by noting the Ra on the HC, then changing the time using the HC and seeing how the Ra changes.  This will affect ALL HC types, NexStar, NexStar+ and StarSense.  The StarSense will generate warnings and will set the mount to unaligned in the HC, The NexStar HCs will allow the time change but will change the position reported.

As for the unpark process this should work by the mount time being set, then the wake up command being sent.  I am not sure if a mount without a RTC will get a time update, I tested on an AVX and may have relied on it's RTC for the time.
It also relies on the mount being parked using the Hibernate command.

These hibernate and wake up commands are vital to the park and unpark process working and retaining alignment.  I am not sure if the StarSense HC implements these.  It would not surprise me if it doesn't or if the implementation is faulty in some way.

I would recommend avoiding the StarSense HC for INDI control, especially remotely.  It is a lot less developed than the NexStar+ HC and has significant bugs around Sync at least.

I can't comment more without a log that shows all the communication between INDI and the mount at start up, with a commentary outlining what was done and what was seen, for example if the actual position reported to INDI is correct.

Chris

enolan94, 2021-01-21:
Hi Chris,
You are correct, setting the time after coming out of park/hibernate does change the hour angle/dec, to the correct values for the given time set sent down to mount.
With the + handset anyway on latest firmware a 'goto' after wakeup/timeset gets me within error limits (e.g. <1deg) of normal coming out of hibernate, considering I move the mount after shutdown and try to place it back exactly where it was on park/hibernate.

The time and location are not forcefully set on indi startup. It takes a  higher level program to set them. in my case CCDCiel) provides option to set time on connect.

If I attempt to connect prior to unpark, errors throw in driver but the time is forced into HC so a subsequent unpark starts with time configured.
08:39:45.020: [WARNING] Mount is NOT aligned. You must align the mount first before you can use it. Disconnect, align the mount, and reconnect again

For my personal use I took the check out of the driver for isAligned and just sent the time towards the mount. This way I can unpark, start tracking and then connect with CCDCiel forcing a time set and all is happy (for me anyway!)

Chris-Rowland, 2021-01-22:
It all needs to work correctly for the more usual case of a user who does a manual alignment before connecting to INDI.  We don't want to screw up their alignment.  Allowing the time or position to be changed after an alignment can/will do this.

The warning that the mount is not aligned is by design, the mount isn't aligned.  An align or wake up is one way to get the mount to be aligned.

It might be worth sending the time from the driver as part of the unpark or align process but this relies on the driver having a good source of time.  Can we rely on this in every case?

enolan94, 2021-01-22:
I believe nothing breaks currently. There is no force set of time in INDI itself. It would be up to the user to enable a client to forcefully send time, as the checkbox enable in CCDCiel does.
Now if we want the INDI driver to stop the user from doing this, as is currently coded, that is a design decision to 'protect' the user, but in my case it becomes confusing for a 'user'
For current code base, if I connect prior to unpark/tracking enable, the time is sent but an error pop's but time is set. If I connect post unpark/tracking enable an error pops but time is not sent.
The code mod I did just sends the time if a higher level program commands it, regardless of isAligned.
Yes, it does require the user to understand what they are doing, but so does current code base :-)

knro, 2021-01-22:
While I hate adding more options to make the driver even more complex to use, is this a case worth of adding such an option to override the currently behavior and ignore isAligned?

enolan94, 2021-01-22:
This may come down to more intelligence in the driver based on the HC type and version.
I only have a sample of one, a Nexstar+ with latest firmware.
The comments in the current code base indicate this has been examined and/or evaluated previously, especially with respect to Starsense HC's
// Update Location and time are disabled if the mount is aligned. This is because
// changing either will change the mount model because at least the local sidereal time
// will be changed. StarSense will set the mount to unaligned but it isn't a good idea even
// with the NexStar HCs

enolan94, 2021-01-23:
For what it may be worth, in my private local copy  I added an Enable/Disable for allowing time set when aligned to the OPTION's screen and backing code to support.
Attached are the diff's for *gps.cpp and *gps.h along with a screen shot based on Indi 1.8.7

The original source I started with had a setVer to 3.4 which matches
https://indilib.org/telescopes/celestron/celestron-nexstar.html

Note: The *gps.h was diff'ed from 1.8.8 which seemed to add a comment to end of line 71

Also, I update the displayed version,  setVersion to (3,5) but did not touch
libindi/drivers.xml

Which I found in 1.8.7 to contain 
    <version>3.3</version>

There are two diff's for celestrongps.cpp.
One is for against v1.8.8 and the other for v1.8.7

[celestrongps_settime.zip](https://github.com/indilib/indi/files/5860613/celestrongps_settime.zip)

knro, 2021-01-24:
Thanks, how would user figure out how to enable this option? Is it possible to have this option enabled for Starsence HCs only by default? I'm not sure if this make sense though.

enolan94, 2021-01-24:
It is a design philosophy issue. I for one think a user should be allowed to do anything the underlying system provides, but of course it requires the user to understand what implications are. In this particular case, since setting time or location is known to break Starsense alignment one aligned, they should not do it. With other HC's it does not break alignment but rather works correctly, but the user is blocked for doing it. Of course it is possible to have a hybrid, if controller is Starsense then do not even present the option, but then I do not have enough knowledge of Celestron HC's to say if anything breaks on v2.X, 4.X, etc so should the option be presented to them? I like being able to tweak things and learning the implications of the tweaks. If I cannot tweak then, as in this case, the source is available I will make it do what I personally want it to. I even have local tweaks on PHD2, CCDCiel and Skychart. Example for PHD2 is the minimum exposure time available only went down to 0.01 seconds, for some testing I was doing in daytime I needed 0.001 so added .005 and .001 choices to the source.

Chris-Rowland, 2021-01-24:
Changing the time or location will break the alignment with ALL Celestron HCs.  This is because time and location are used as part of the alignment process to convert between Ra and hour angle.  This is because the localsidereal time depends on time and location.

This is true for ALL Celestron HCs, Try it and see.  You will find that the Ra will change if you change the time, time zone, DST or longitude.

The StarSense HCs don't allow changing the time or location and if you do the HC will become unaligned.

The NexStar HCs are in some senses worse, the alignment is still broken but the HC alignment is not lost, the HC still thinks it is aligned but it isn't.

Putting a "Break the Alignment" in the driver doesn't seem very sensible to me.

Trust me on this. I've been writing mount control software for the Celestron HCs for nearly 20 years, from the days of the original NexStar version 1.6 onwards and have good contact with the Celestron developers.

You are free to do what you want with your own code base but dumping your personal and broken code on all the users is I think a bad idea.

enolan94, 2021-01-24:
Chris, I am using it daily on a wedge mounted SE with a C6 and Nexstar+
Enable Hibernate, Park, shutdown scope
Next day
Powerup scope, Unpark, Enable tracking.
At this point the RA?DEC matches the time of the shutdown
Push Time to HC (via capability in Ccdciel), RA and DEC change to match the 'new' time. Do a 'Goto' and be within a degree of the target.
Remember I physically move the scope from shutdown to power up so accuracy is dependent on how well I place it back where I started.

Now all I am changing is the Time, not DST or Lat/Long, just Tiime

enolan94, 2021-01-24:
Chris, attached are series of pics showing the drivers screens during phases from power up to after time set.
The NexStar+ with latest firmware does the right thing, LST and all.

Your statement "This is true for ALL Celestron HCs," is not correct for based on my sample of one.

[NexstarPlusTimeSetting.zip](https://github.com/indilib/indi/files/5863254/NexstarPlusTimeSetting.zip)

enolan94, 2021-01-24:
And the comment on LST, it updates correctly, two images
Poweron after hibernate for hours
Unpark, start tracking, take pic of HC LST
Send time down via driver, take second PIC of LST after time set

[LSTaftertimechange.zip](https://github.com/indilib/indi/files/5863476/LSTaftertimechange.zip)

enolan94, 2021-01-25:
[image]
And this is my c6 SE rig
Pi4 controlled via INDI.
DIY wedge, guide scope/cam, main scope/cam, 3d-printed focus adapter with stepper and my stepper controller.

My normal astro community identity is astrogene1000_stuff and I do build a lot of stuff for scopes.

astrogene1000, 2021-01-25:
Hi Chris, the following link shows how I use the driver, scripted for a small automated search via CCDciel.
Prior to each session, which may script 25 to a hundred objects a night, I use the time set 'hack'.
It all works flawlessly on my HC
[http://www.astrogene1000.com/supernova/c6/webpage.html](url)

astrogene1000, 2021-01-25:
Serial sniff logs attached
INDI driver to HC and AUX bus communications
I repurposed my DIY GPS AUX port module as an AUX sniffer and captured all AUX comms

During both logs
Power on, unpark, start tracking
Then CCDciel connects through driver and sends time and whatever else goes back and forth from driver to app and HC

[auxser1.txt](https://github.com/indilib/indi/files/5870076/auxser1.txt)
[celestron-hc-sniff.log](https://github.com/indilib/indi/files/5870079/celestron-hc-sniff.log)

astrogene1000, 2021-01-28:
Ok,, so no activity in 4 days, closing issue

closed 2021-01-28 by astrogene1000

Chris-Rowland, 2021-01-31:
Sorry for delay.  I think that closing this is correct.

I tried and see the issue I meentioned:-
Scope: AVX HC NexStar+ Version 5.31.9200, this is the current released version, just downloaded from Celestron.
Start, set time correctly do a Quick Align
Slew to object (M12) position reached correctly.
Using Cartes du Ceil and the position is shown as expected.
Use HC Scope Setup/Setup Time and Site tochange the time by an hour.
On completing the setup the position shown in CdC changes by an hour in Ra as predicted.
Changing the time back returns the position to what was initially set.

This is why I disabled time and location setting when the mount reports it is aligned, it breaks the alignment.

Chris

astrogene1000, 2021-02-02:
Hi Chris, thank for the response.
I am a bit confused though, if start at Time0 RA0/DEC0 then set to TIME1 get RA1/DEC1 then set back to TIME0 I would expect it to now say RA0/DEC0

Earlier in the thread you mentioned seeing captures. What I saw was the motor's only know about ALT/AZ so no matter what the time is the motor locations are not updated. Here is the communications I posted in the indilib forum

Then via INDI driver, Park at specific ALT/AZ position with Hibernate enabled
Power-off mount
Power-on mount some time later
HC sitting there saying 'Hit Enter to wake up'

Using INDI driver, Un-park and enable tracking
HC shows RA/DEC of the last time the HC controller knew about at the 'park position'

Update time to the HC
HC then shows the 'new' time and the RA/DEC matching the park position at the new time.

Since it seems the motor cards work in ALT/AZ mode and not RA/DEC mode the HC now has the correct RA and DEC and LST and as far as the motor cards are concerned, they are still at the same ALT/AZ with respect to the mount

Now what internal model of the HC would change considering the one-star north align and sidereal tracking ?

Here is an AUX port sniff of all commands on the AUX bus from scope pawer up
On top of the indi driver is a program that will force set time to the mount when it is already aligned. (CCDCiel in my case on Connect)

Notice below the motor cards start up at position 0 ALT, 0 AZ
When scope is un parked they are told the ALT/AZ of the park position

0x3b 0x03 0x0d 0x11 0x05 0xda To ALT, CMD = 5 ???
0x3b 0x05 0x11 0x0d 0x05 0x0c 0x82 0x4a
0x3b 0x03 0x0d 0x10 0x05 0xdb To AZ, CMD = 5 ???
0x3b 0x05 0x10 0x0d 0x05 0x0c 0x82 0x4b
0x3b 0x03 0x0d 0x10 0xfe 0xe2 Get AZM version
0x3b 0x05 0x10 0x0d 0xfe 0x05 0x0e 0xcd
0x3b 0x03 0x0d 0x10 0xfc 0xe4 Get AZM approach
0x3b 0x04 0x10 0x0d 0xfc 0x00 0xe3
0x3b 0x03 0x0d 0x11 0xfc 0xe3 Get ALT approach
0x3b 0x04 0x11 0x0d 0xfc 0x00 0xe2
0x3b 0x03 0x0d 0x10 0x05 0xdb To AZM, CMD = 5 ???
0x3b 0x05 0x10 0x0d 0x05 0x0c 0x82 0x4b
0x3b 0x03 0x0d 0x10 0xfe 0xe2 Get AZ version
0x3b 0x05 0x10 0x0d 0xfe 0x05 0x0e 0xcd
0x3b 0x03 0x0d 0x11 0xfe 0xe1 Get ALT Version
0x3b 0x05 0x11 0x0d 0xfe 0x05 0x0e 0xcc
0x3b 0x03 0x0d 0xb0 0x33 0x0d Try GPS
0x3b 0x03 0x0d 0xb0 0x33 0x0d
0x3b 0x03 0x0d 0xb0 0x33 0x0d
0x3b 0x03 0x0d 0xb0 0x33 0x0d
0x3b 0x03 0x0d 0xb0 0x33 0x0d
0x3b 0x03 0x0d 0x12 0xfe 0xe0 Try focuser?
0x3b 0x03 0x0d 0x12 0xfe 0xe0
0x3b 0x03 0x0d 0x12 0xfe 0xe0
0x3b 0x03 0x0d 0x12 0xfe 0xe0
0x3b 0x03 0x0d 0x12 0xfe 0xe0
0x3b 0x03 0x0d 0x10 0x47 0x99 AZ get AUTOGUIDE RATE
0x3b 0x04 0x10 0x0d 0x47 0xff 0x99
0x3b 0x03 0x0d 0x11 0x47 0x98 ALT get autoguide rate
0x3b 0x04 0x11 0x0d 0x47 0xff 0x98

0x3b 0x03 0x0d 0x10 0x01 0xdf AZ Get position, scope has been parked and hibernated and still parked
0x3b 0x06 0x10 0x0d 0x01 0x00 0x00 0x00 0xdc
0x3b 0x03 0x0d 0x11 0x01 0xde ALT get position
0x3b 0x06 0x11 0x0d 0x01 0x00 0x00 0x00 0xdb
0x3b 0x03 0x0d 0x10 0x01 0xdf
0x3b 0x06 0x10 0x0d 0x01 0x00 0x00 0x00 0xdc
0x3b 0x03 0x0d 0x11 0x01 0xde
0x3b 0x06 0x11 0x0d 0x01 0x00 0x00 0x00 0xdb
0x3b 0x03 0x0d 0x10 0x01 0xdf
0x3b 0x06 0x10 0x0d 0x01 0x00 0x00 0x00 0xdc
0x3b 0x03 0x0d 0x11 0x01 0xde
0x3b 0x06 0x11 0x0d 0x01 0x00 0x00 0x00 0xdb
0x3b 0x03 0x0d 0x10 0x01 0xdf
[... 76 lines cut ...]
0x3b 0x03 0x0d 0x10 0x01 0xdf
0x3b 0x06 0x10 0x0d 0x01 0x7b 0x0f 0xac 0xa6
0x3b 0x03 0x0d 0x11 0x01 0xde
0x3b 0x06 0x11 0x0d 0x01 0xe6 0xe3 0xd2 0x40
0x3b 0x03 0x0d 0x10 0x01 0xdf
0x3b 0x06 0x10 0x0d 0x01 0x7b 0x10 0x7e 0xd3
0x3b 0x03 0x0d 0x11 0x01 0xde
0x3b 0x06 0x11 0x0d 0x01 0xe6 0xe3 0xd2 0x40
0x3b 0x03 0x0d 0x10 0x01 0xdf
0x3b 0x06 0x10 0x0d 0x01 0x7b 0x11 0x4c 0x04
0x3b 0x03 0x0d 0x11 0x01 0xde
0x3b 0x06 0x11 0x0d 0x01 0xe6 0xe3 0xd2 0x40
0x3b 0x03 0x0d 0x10 0x01 0xdf
0x3b 0x06 0x10 0x0d 0x01 0x7b 0x12 0x14 0x3b
0x3b 0x03 0x0d 0x11 0x01 0xde
0x3b 0x06 0x11 0x0d 0x01 0xe6 0xe3 0xd2 0x40
0x3b 0x03 0x0d 0x10 0x01 0xdf
0x3b 0x06 0x10 0x0d 0x01 0x7b 0x12 0xe6 0x69
0x3b 0x03 0x0d 0x11 0x01 0xde
0x3b 0x06 0x11 0x0d 0x01 0xe6 0xe3 0xd2 0x40
0x3b 0x03 0x0d 0x10 0x01 0xdf
0x3b 0x06 0x10 0x0d 0x01 0x7b 0x13 0xbb 0x93
0x3b 0x03 0x0d 0x11 0x01 0xde
0x3b 0x06 0x11 0x0d 0x01 0xe6 0xe3 0xd2 0x40
0x3b 0x03 0x0d 0x10 0x01 0xdf

Chris-Rowland, 2021-02-03:
Hi Gene, replies inline.

>
> Hi Chris, thank for the response.
> I am a bit confused though, if start at Time0 RA0/DEC0 then set to 
> TIME1 get RA1/DEC1 then set back to TIME0 I would expect it to now say 
> RA0/DEC0
I would expect it to reach RA1/DEC0, not DEC1, but that's not the 
point.  If you change the time on an aligned mount you can loose the 
alignment.
>
> Earlier in the thread you mentioned seeing captures. What I saw was 
> the motor's only know about ALT/AZ so no matter what the time is the 
> motor locations are not updated. Here is the communications I posted 
> in the indilib forum
I replied in the INDI forum.  I don't see any need to repeat that reply.

astrogene1000, 2021-02-04:
Sorry, still confused.

If time is set time to 9:00:00 and given RA = 18:00:00 for scope at AZ=180,ALT=14
Now change time to 10:00:00 but not move the scope, RA now equals 19:00:00 because scope is still at AZ=180, ALT = 14
Then again change time back to 9:00:00 and not move the scope, RA now equals 18:00:00, the scope still pointing to AZ=180, ALT-14.
[image]

And sorry, I do not see any replies in indilib for subject
[https://indilib.org/forum/general/8616-celestron-nexstar-remote-powerup.html#66077](url)

astrogene1000, 2021-02-05:
Hi Chris,

Try this, with time set enabled even if isAligned = true
Set first time to match current time
Set second time to be + or - one hour
Set third time to match current time

And watch the RA move +/- one hour and then back to original

#/!bin/bash
port=7624

indi_setprop -p $port "Celestron GPS.CONNECTION.CONNECT=On"
indi_setprop -p $port "Celestron GPS.TIME_UTC.UTC;OFFSET=2021-01-05T18:55:12;-5"
sleep 5
indi_setprop -p $port "Celestron GPS.TIME_UTC.UTC;OFFSET=2021-01-05T17:55:12;-5"
sleep 5
indi_setprop -p $port "Celestron GPS.TIME_UTC.UTC;OFFSET=2021-01-05T18:55:12;-5"

astrogene1000, 2021-02-06:
=====================================
Set time = 2021-01-05T20:35:12;-5
Value of RA = 23.555851936340332031
Value of DEC = -19.235944747924804688
=====================================
Set time = 2021-01-05T19:35:12;-5
Value of RA = 22.552522659301757812
Value of DEC = -19.235944747924804688
=====================================
Set time = 2021-01-05T20:35:12;-5
Value of RA = 23.554666042327880859
Value of DEC = -19.235944747924804688

Chris-Rowland, 2021-02-06:
I'm not sure what point you are trying to make, this demonstrates what 
I'm saying - changing the time changes the position reported.  It breaks 
the alignment.
Your first change, by an hour, changes the Ra by about an hour.
Your second change, also by an hour, changes the Ra by about an hour.  
None of this is exact because the time change, from the mount HC's 
internal time to the new time will not be exactly an hour.

Adding a break the alignment option to the driver seems like a dumb idea 
to me.

Sending an explicit set time operation in the unpark function just 
before the unpark command is sent will I think do what you want and not 
break anything else.

Chris

astrogene1000, 2021-02-08:
Chris, not discussing adding to driver, just trying to understand -how- it is broken.
My point is if at time 0 with scope at alt/az 0 I turn it off at time 0 and hibernate
When I turn it back on, HC still thinks it is time 0 and scope is still at alt/az 0
Now if I unpark, HC still thanks the same
If I then set the time, the alt/az has not changed but the HC will update the ra to match the time and alt/az position of the scope.

Now under your premise, all is broken. My premise is if it was broken how come if I do a goto the goto lands correctly.
This should not be possible if broken.

I see the same results if I hibernate with GPS enabled
Then turn on and just accept the time shown on the HC without waiting for GPS sync and then
Do and do a GPS sync, the RA changes to match the time because scope has not moved.
Same thing if I hibernate with GPS disabled, turn on and just accept the time then subsequently enable GPS and time sync GPS, RA changes because scope has not moved.

Gene