indilib/indi #1332: meade-dsi-iii-colour-camera-unsupported-bayer-pattern.

Geoff1801, 2021-01-20:
**Did you make sure there is no existing driver for your device? Please check [supported devices portal](https://indilib.org/devices].**
Please confirm you verified no drivers exist for your device.      Incorrect driver for my imager

**Provide link to product page**
http://www.hamiltonastronomy.com/ASTRO_ARTICLES/DSI_Pro_III.pdf 
Provide URL to the product description and details.

**Describe device family** Cameras: ccd
All INDI devices fall under one or more of the following categories:
* Mounts
* Cameras (CCDs + CMOS)
* Focusers
* Filter Wheels
* Domes
* Weather Stations
* Adaptive Optics
* Receivers
* Spectrographs
* Auxiliary (Anything else that does not fit above).

**Provide device communication protocol**
All devices use some sort of communication language between them and the host system. This could be called *Command Set* or *SDK*. It is best to provide a link to a document or PDF with such details from the manufacutrer. It needs to include the following information:
* Communication Method: Serial, USB, TCP...etc. along with the connection parameters.
* Communication Protocol: Detailed description of the language used to communicate with the device along with examples if available.

**Hardware Sample**
Would it be possible to send the developer a sample hardware unit to develop the driver? If that's not possible, would it be possible to setup a remote environment with the devie attached?

TallFurryMan, 2021-01-22:
This issue report needs to be reformatted and reviewed for details (can't edit myself).

The original forum thread is https://indilib.org/forum/ccds-dslrs/8639-re-re-re-re-meade-dsi-iii-colour-camera-unsupported-bayer-pattern.html.

The problem is support for the sensor properties of the Meade DSI III. The INDI driver is marking Bayer requirement as soon as the camera is color, but the DSI III has a standard RGB pattern.

The relevant part of the driver log is the following, which helps locating the issue in the driver:

```
[...INFO ][ org.kde.kstars.indi] - Meade Deep Sky Imager : "[INFO] Found a DSI with an unknown CCD: ICX285AQ "
[...INFO ][ org.kde.kstars.ekos] - "Meade Deep Sky Imager" Version: "0.3" Interface: 6 is connected.
```

I will provide an `amd64` Linux binary of the `indi_dsi_ccd` driver for test purposes, then a PR fixing the problem if successful.

TallFurryMan, 2021-01-22:
@knro please assign to me. Is it OK to provide executables in an Issue report?

Geoff1801, 2021-01-22:
Yes that is OK with me. Unfortunately I am a complete ludite with this sort of thing so you may need to guide me through this.

⁣Get TypeApp for Android ​

TallFurryMan, 2021-01-22:
The 0.4 driver I attach to this post has a very simple fix that clears the Bayer flag if the driver sees an ICX285AQ sensor.

[indi_dsi_ccd.zip](https://github.com/indilib/indi/files/5858263/indi_dsi_ccd.zip)

On the computer where the DSI is connected, uncompress the zip archive to a subfolder of your home folder. If you want to verifiy the integrity of the executable file, you may use `md5sum` in a terminal in this subfolder. I obtain the following MD5 checksum from my end, so will you:
```
$ md5sum ./indi_dsi_ccd
105b9b65dc4ab9de80f5ab2f7e5e4d66  ./indi_dsi_ccd
```

From that same subfolder, execute `indiserver` with that driver as argument:
```
$ indiserver ./indi_dsi_ccd 
2021-01-22T19:58:53: startup: indiserver ./indi_dsi_ccd 
```
Note the argument of `indiserver` is explicitly the file that was uncompressed, and not a driver installed elsewhere in the system. The `startup` log is produced by `indiserver`.

With Ekos, create an equipment profile and connect to the `indiserver` running on `localhost`, default port `7624`.

[image]

The device is version 0.4, while the latest released version is 0.3:

[image]

When the device is connected with the driver, paste the log seen in the INDI panel. In my case, obviously, no device.

[image]

If the device is recognised, please try to run a capture and check the result.

After you disconnect Ekos, you may stop `indiserver` by pressing Ctrl-C in the terminal where it is running.
```
$ indiserver ./indi_dsi_ccd 
2021-01-22T19:58:53: startup: indiserver ./indi_dsi_ccd 
^C
$
```

TallFurryMan, 2021-01-22:
Looking deeper in the driver, support for DSI Color III was introduced by @bgilsrud in 2016, I wasn't looking closely enough!

So oops, I'll be checking the driver again.

Geoff1801, 2021-01-23:
Firstly Eric, Thanks very much for assisting me with this.

Below is a copy of my terminal log after entering the command you have forwarded to me. You can see that it didn't accept. The only variation to what you sent is the date/time numbers which would be correct.

geoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdparty-Desktop-Debug$ indiserver ./indi_dsi_ccd
2021-01-23T02:55:21: startup: indiserver ./indi_dsi_ccd
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1790 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: stderr EOF
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: restart #1
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1792 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: restart #2
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1793 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: stderr EOF
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: restart #3
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1794 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: stderr EOF
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: restart #4
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1795 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: stderr EOF
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: restart #5
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1796 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: stderr EOF
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: restart #6
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1797 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: stderr EOF
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: restart #7
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1798 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: stderr EOF
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: restart #8
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1799 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: restart #9
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1800 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: stderr EOF
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: restart #10
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: 2021-01-23T02:55:21: Driver ./indi_dsi_ccd: execlp: No such file or directory
Child process 1801 died
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: stderr EOF
2021-01-23T02:55:21: Driver ./indi_dsi_ccd: Terminated after #10 restarts.
2021-01-23T02:55:21: good bye
geoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdparty-Desktop-Debug$geoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdparty-Desktop-Debugeoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdparty-Desktop-Degeoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdparty-Desktop-geoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdparty-Desktogeoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdpartgeoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdparty-Desktop-geoff@ggeoff@ggeoff@ggeoff@geoff-HPgeoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdparty-Desktop-Dgeoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdparty-Desktop-Debug$ 2021-01-22T19:58:53: startup: indiserver ./indi_dsi_ccd ^C
geoff@geoff-HP-ProBook-6560b:~/Documents/build-indi-3rdparty-Desktop-Debug$

Oh! I just noticed your later email. but thought I would send you the terminal extract anyway it may help.

Regards Geoff

On 23/1/21 7:14 am, TallFurryMan wrote:
> indiserver ./indi_dsi_ccd 
> 2021-01-22T19:58:53: startup: indiserver ./indi_dsi_ccd

TallFurryMan, 2021-01-23:
A new iteration of the driver, which advertises a RGGB pattern.

[indi_dsi_ccd.zip](https://github.com/indilib/indi/files/5861056/indi_dsi_ccd.zip)

```
$ md5sum indi_dsi_ccd
ce7697650c9b6178d468af81d7be79a8  indi_dsi_ccd
```

The error you get about `execlp` is probably because the `indi_dsi_ccd` driver executable is not present at the folder level you are executing `indiserver`. The notation `./` means "in the current folder". Make sure you can list the driver with `ls ./indi_dsi_ccd` before executing `indidriver ./indi_dsi_ccd`.

TallFurryMan, 2021-01-23:
If you want to paste a log excerpt in this thread, write three backticks (` ``` `) on the line before and three backticks on the line after. This will tell github you want to render a pre-formatted block.

TallFurryMan, 2021-01-23:
Also, if you can paste (zipped) FITS files taken with the various driver iterations in this thread, that will be helpful.

Geoff1801, 2021-01-24:
The execlp error appears to have gone however the logs still shows
```Imager :  "[INFO] Found a DSI with an unknown CCD: ICX285AQ "
[2021-01-24T14:58:51.184 AEDT INFO ][           org.kde.kstars.ekos] - "Meade Deep Sky Imager" Version: "0.3" Interface: 6 is connected
`Attached is an extract terminal.``
```
geoff@geoff-HP-ProBook-6560b:~$ cd Documents
geoff@geoff-HP-ProBook-6560b:~/Documents$ ls
 build-indi-3rdparty-Desktop-Debug  'Screenshot 1.png'
 indi_dsi_ccd                       'Screenshot 2.png'
'No change to driver.fits.tar.gz'    Various
geoff@geoff-HP-ProBook-6560b:~/Documents$ ls ./indi_dsi_ccd
./indi_dsi_ccd
geoff@geoff-HP-ProBook-6560b:~/Documents$ indiserver ./indi_dsi_ccd
2021-01-24T03:36:56: startup: indiserver ./indi_dsi_ccd 
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: stderr EOF
Child process 7447 died
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: restart #1
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: stderr EOF
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: restart #2
Child process 7448 died
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: stderr EOF
Child process 7449 died
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: restart #3
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: stderr EOF
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: restart #4
Child process 7450 died
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: stderr EOF
2021-01-24T03:36:56: Driver ./indi_dsi_ccd: restart #5
Child process 7451 died
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
Child process 7452 died
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: stderr EOF
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: restart #6
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: stderr EOF
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: restart #7
Child process 7453 died
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: stderr EOF
Child process 7454 died
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: restart #8
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
Child process 7455 died
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: restart #9
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: stderr EOF
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: restart #10
Child process 7456 died
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: ./indi_dsi_ccd: symbol lookup error: ./indi_dsi_ccd: undefined symbol: _ZThn24_N4INDI3CCD10GuideNorthEj
Child process 7457 died
2021-01-24T03:36:57: Driver ./indi_dsi_ccd: Terminated after #10 restarts.
2021-01-24T03:36:57: good bye
geoff@geoff-HP-ProBook-6560b:~/Documents$ 
```
Regards Geoff

TallFurryMan, 2021-01-24:
Ah, which version of indi do you have installed on this machine? You rebuilt it from sources?

Geoff1801, 2021-01-24:
No I'm not that clever. I just downloaded Kstars from the indilib site via the terminal commands contained within indilib site. The Kstars version I am running is currently Version 3.5.1 Stable Build: 2021-01-08T07:17:10Z. Hoping that is what you are after. 
Regards Geoff

TallFurryMan, 2021-01-24:
Because you ran the test from a build-indi-3rdparty folder, I thought you were rebuilding the two repositories yourself. I was actually asking you about the version of the "libindi" package. If you successfully rebuilt indi and indi-3rdparty, we can do this differently than with posted executables. 

What I can also do is to push the change and let the build chain provide the packet to you (this is longer though). You will need to switch to the ppa providing nightly builds to obtain it.

Geoff1801, 2021-01-24:
Given my capabilities ( lack of)  computer wise, I would prefer to download the nightly build from the INDILIB site. If possible could you advise me when the packet required has been added to the nightly build. 
Thanks again 
Geoff

TallFurryMan, 2021-01-25:
I pushed a PR at https://github.com/indilib/indi-3rdparty/pull/306. When it is merged, I'll monitor the PPA builder and give you the package version the change will be in.

TallFurryMan, 2021-01-26:
I fixed a wrong commit message and replaced the PR with https://github.com/indilib/indi-3rdparty/pull/307.

Geoff1801, 2021-01-27:
Thanks Eric, Will await your advice. Did a nightly but the change obviously not in yet.

TallFurryMan, 2021-01-30:
@Geoff1801 I believe `indi-dsi 0.2~202101292329` should contain my change now.

Geoff1801, 2021-01-30:
Reloaded the latest version of INDILIB Kstars then updated to the indi-dsi 0.2~202101292329 Via the update manager, rebooted the computer.  When attempting to connect the camera,  Kstars tells me that the INDI Driver Meade Deep Sky Imager has crashed.    So no connection at all.

TallFurryMan, 2021-01-30:
Do you have a log? Could you run the driver with the same method as described earlier?

Geoff1801, 2021-01-31:
Hi Eric, Here is the latest log.
```
[2021-01-31T13:02:01.434 AEDT INFO ][                org.kde.kstars] - Welcome to KStars 3.5.2 Beta
[2021-01-31T13:02:01.435 AEDT INFO ][                org.kde.kstars] - Build: 2021-01-26T17:53:19Z
[2021-01-31T13:02:01.435 AEDT INFO ][                org.kde.kstars] - OS: "linuxmint"
[2021-01-31T13:02:01.435 AEDT INFO ][                org.kde.kstars] - API: "x86_64-little_endian-lp64"
[2021-01-31T13:02:01.435 AEDT INFO ][                org.kde.kstars] - Arch: "x86_64"
[2021-01-31T13:02:01.435 AEDT INFO ][                org.kde.kstars] - Kernel Type: "linux"
[2021-01-31T13:02:01.435 AEDT INFO ][                org.kde.kstars] - Kernel Version: "5.4.0-65-generic"
[2021-01-31T13:02:01.435 AEDT INFO ][                org.kde.kstars] - Qt Version: 5.12.8
[2021-01-31T13:02:01.443 AEDT INFO ][                org.kde.kstars] - KStars is started in paused state.
[2021-01-31T13:02:01.718 AEDT DEBG ][                org.kde.kstars] - Opened the User DB. Ready.
[2021-01-31T13:02:01.943 AEDT INFO ][                org.kde.kstars] - Processing  "unnamedstars.dat" , HTMesh Level 3
[2021-01-31T13:02:01.943 AEDT INFO ][                org.kde.kstars] -   Sky Mesh Size:  512
[2021-01-31T13:02:01.967 AEDT INFO ][                org.kde.kstars] - Loaded DSO catalog file:  "unnamedstars.dat"
[2021-01-31T13:02:01.967 AEDT INFO ][                org.kde.kstars] - Processing  "deepstars.dat" , HTMesh Level 3
[2021-01-31T13:02:01.967 AEDT INFO ][                org.kde.kstars] -   Sky Mesh Size:  512
[2021-01-31T13:02:01.967 AEDT INFO ][                org.kde.kstars] - Loaded DSO catalog file:  "deepstars.dat"
[2021-01-31T13:02:01.973 AEDT INFO ][                org.kde.kstars] - Processing  "USNO-NOMAD-1e8.dat" , HTMesh Level 6
[2021-01-31T13:02:01.983 AEDT INFO ][                org.kde.kstars] -   Sky Mesh Size:  32768
[2021-01-31T13:02:01.990 AEDT INFO ][                org.kde.kstars] - Loaded DSO catalog file:  "USNO-NOMAD-1e8.dat"
[2021-01-31T13:02:02.310 AEDT WARN ][                org.kde.kstars] - "Star HD20,794 not found."
[2021-01-31T13:02:02.325 AEDT WARN ][                org.kde.kstars] - "Star HD98,230 not found."
[2021-01-31T13:02:02.339 AEDT DEBG ][                       default] - File opened:  "/usr/share/kstars/ngcic.dat"
[2021-01-31T13:02:02.339 AEDT INFO ][                org.kde.kstars] - Loading NGC/IC objects
[2021-01-31T13:02:02.844 AEDT DEBG ][                       default] - File opened:  "/home/geoff/.local/share/kstars/comets.dat"
[2021-01-31T13:02:03.011 AEDT WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-31T13:02:03.013 AEDT WARN ][                org.kde.kstars] - "Object named NGC 6050A not found"
[2021-01-31T13:02:03.014 AEDT WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-31T13:02:03.019 AEDT WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-31T13:02:03.019 AEDT WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-31T13:02:03.019 AEDT WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-31T13:02:03.019 AEDT WARN ][                org.kde.kstars] - "Object named Ceres not found"
[2021-01-31T13:02:03.019 AEDT WARN ][                org.kde.kstars] - "Object named Pallas not found"
[2021-01-31T13:02:03.019 AEDT WARN ][                org.kde.kstars] - "Object named Vesta not found"
[2021-01-31T13:02:03.019 AEDT WARN ][                org.kde.kstars] - "Object named Cybele not found"
[2021-01-31T13:02:03.019 AEDT WARN ][                org.kde.kstars] - "Object named Psyche not found"
[2021-01-31T13:02:03.019 AEDT WARN ][                org.kde.kstars] - "Object named Juno not found"
[2021-01-31T13:02:03.019 AEDT WARN ][                org.kde.kstars] - "Object named Themis not found"
[2021-01-31T13:02:03.020 AEDT WARN ][                org.kde.kstars] - "Object named Astraea not found"
[2021-01-31T13:02:03.020 AEDT WARN ][                org.kde.kstars] - "Object named Lumen not found"
[2021-01-31T13:02:03.020 AEDT WARN ][                org.kde.kstars] - "Object named Ida not found"
[2021-01-31T13:02:03.020 AEDT WARN ][                org.kde.kstars] - "Object named Mathilde not found"
[2021-01-31T13:02:03.020 AEDT WARN ][                org.kde.kstars] - "Object named Eros (1898 DQ) not found"
[2021-01-31T13:02:03.020 AEDT WARN ][                org.kde.kstars] - "Object named Hektor (1907 XM) not found"
[2021-01-31T13:02:03.025 AEDT WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-31T13:02:03.026 AEDT WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-31T13:02:03.026 AEDT WARN ][                org.kde.kstars] - "Object named Pluto not found"
[2021-01-31T13:02:03.026 AEDT WARN ][                org.kde.kstars] - "Object named Ceres not found"
[2021-01-31T13:02:03.026 AEDT WARN ][                org.kde.kstars] - "Object named Pallas not found"
[2021-01-31T13:02:03.026 AEDT WARN ][                org.kde.kstars] - "Object named Vesta not found"
[2021-01-31T13:02:03.026 AEDT WARN ][                org.kde.kstars] - "Object named Cybele not found"
[2021-01-31T13:02:03.026 AEDT WARN ][                org.kde.kstars] - "Object named Psyche not found"
[2021-01-31T13:02:03.026 AEDT WARN ][                org.kde.kstars] - "Object named Juno not found"
[... cut ...]
[2021-01-31T13:02:31.737 AEDT DEBG ][           org.kde.kstars.indi] - Removing device "Meade Deep Sky Imager"
[2021-01-31T13:02:31.738 AEDT DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2021-01-31T13:02:31.738 AEDT CRIT ][           org.kde.kstars.indi] - INDI driver  "indi_dsi_ccd"  crashed!
[2021-01-31T13:02:31.765 AEDT DEBG ][           org.kde.kstars.indi] - INDIListener: Removing device "Meade Deep Sky Imager"
[2021-01-31T13:02:31.766 AEDT INFO ][           org.kde.kstars.ekos] - "Meade Deep Sky Imager is offline."
```

TallFurryMan, 2021-01-31:
Well, I would have preferred you to run the `indiserver -vv indi_dsi_ccd` method to obtain a cleaner log, but I could follow up on what is happening anyway. I see that the `indi_dsi_ccd` driver started initialisation, up to the point it takes a preliminary image.

I don't understand yet what is happening, but it is unrelated to the code I added. I can read that the driver detected the sensor type 'ICX285AQ', then probably switched to the right sub-driver for the DSI III (which is named `DSI111` unexpectedly, and also there are running exchanges that may not be expected here). The last lines are about the preliminary exposure ran as an internal test, and the frame is read in interlaced mode (even then odd frames) and log states it starts with the odd frame (while it should start with the even frame).

Would it be possible to run a few attempts to see if the result is consistent, in order to understand if there is a driver instability or if the amount of logs is too high for the client? Start `indiserver` with the following command:
```
indiserver -vv $(which indi_dsi_ccd)
```
And connect Ekos to `localhost` port `7642`. Then paste the logs output by `indiserver` for each attempt (connect, exposure, disconnect, if any of these steps are possible).

Also, always make sure you update the whole INDI stack, core + third-party drivers, to nightly builds so that your installation is coherent. Updating KStars/Ekos is not needed at this stage, we are focusing on the INDI part.

Geoff1801, 2021-02-01:
I hope that this is what you are seeking. 
I do not know how to  update the whole INDI stack, core + third-party drivers, to nightly builds
In kstars I am now getting the message " unable to find DSI. has firmware been loaded?" for both port configerations. Also unable to get logs for connect, exposure, or disconnect as we are now not getting to that point.

Logs on changing to port 7642.

```
geoff@geoff-HP-ProBook-6560b:~$ indiserver -vv -p 7642 $ meade dsi 111
2021-02-01T00:05:55: startup: indiserver -vv -p 7642 $ meade dsi 111 
2021-02-01T00:05:55: Driver $: pid=10085 rfd=3 wfd=6 efd=7
2021-02-01T00:05:55: Driver meade: pid=10086 rfd=4 wfd=9 efd=10
2021-02-01T00:05:55: Driver dsi: pid=10087 rfd=5 wfd=12 efd=13
Child process 10085 died
Child process 10086 died
2021-02-01T00:05:55: Driver 111: pid=10088 rfd=8 wfd=15 efd=16
2021-02-01T00:05:55: listening to port 7642 on fd 11
2021-02-01T00:05:55: Driver 111: sending <getProperties version='1.7'/>

2021-02-01T00:05:55: Driver dsi: sending <getProperties version='1.7'/>

2021-02-01T00:05:55: Driver meade: 2021-02-01T00:05:55: Driver meade: execlp: No such file or directory
2021-02-01T00:05:55: Driver meade: stdin EOF
2021-02-01T00:05:55: Driver meade: restart #1
Child process 10087 died
Child process 10088 died
2021-02-01T00:05:55: Driver meade: pid=10089 rfd=0 wfd=10 efd=14
2021-02-01T00:05:55: Driver 111: 2021-02-01T00:05:55: Driver 111: execlp: No such file or directory
2021-02-01T00:05:55: Driver 111: stdin EOF
2021-02-01T00:05:55: Driver 111: restart #1
Child process 10089 died
2021-02-01T00:05:55: Driver 111: pid=10090 rfd=4 wfd=15 efd=16
2021-02-01T00:05:55: Driver 111: sending <getProperties version='1.7'/>

2021-02-01T00:05:55: Driver dsi: 2021-02-01T00:05:55: Driver dsi: execlp: No such file or directory
2021-02-01T00:05:55: Driver dsi: stdin EOF
2021-02-01T00:05:55: Driver dsi: restart #1
Child process 10090 died
2021-02-01T00:05:55: Driver dsi: pid=10091 rfd=5 wfd=12 efd=13
2021-02-01T00:05:55: Driver 111: 2021-02-01T00:05:55: Driver 111: execlp: No such file or directory
2021-02-01T00:05:55: Driver 111: stdin EOF
2021-02-01T00:05:55: Driver 111: restart #2
Child process 10091 died
2021-02-01T00:05:55: Driver 111: sending <getProperties version='1.7'/>

2021-02-01T00:05:55: Driver dsi: 2021-02-01T00:05:55: Driver dsi: execlp: No such file or directory
2021-02-01T00:05:55: Driver dsi: stdin EOF
2021-02-01T00:05:55: Driver dsi: restart #2
Child process 10092 died
2021-02-01T00:05:55: Driver dsi: pid=10093 rfd=5 wfd=12 efd=13
2021-02-01T00:05:55: Driver 111: 2021-02-01T00:05:55: Driver 111: execlp: No such file or directory
2021-02-01T00:05:55: Driver 111: stdin EOF
2021-02-01T00:05:55: Driver 111: restart #3
2021-02-01T00:05:55: Driver 111: pid=10094 rfd=4 wfd=15 efd=16
2021-02-01T00:05:55: Driver 111: sending <getProperties version='1.7'/>

2021-02-01T00:05:55: Driver dsi: 2021-02-01T00:05:55: Driver dsi: execlp: No such file or directory
Child process 10093 died
2021-02-01T00:05:55: Driver dsi: write: Broken pipe
2021-02-01T00:05:55: Driver dsi: restart #3
2021-02-01T00:05:55: Driver dsi: pid=10095 rfd=5 wfd=12 efd=13
2021-02-01T00:05:55: Driver 111: 2021-02-01T00:05:55: Driver 111: execlp: No such file or directory
Child process 10094 died
2021-02-01T00:05:55: Driver dsi: sending <getProperties version='1.7'/>

2021-02-01T00:05:55: Driver meade: 2021-02-01T00:05:55: Driver meade: execlp: No such file or directory
2021-02-01T00:05:55: Driver meade: stdin EOF
2021-02-01T00:05:55: Driver meade: restart #2
2021-02-01T00:05:55: Driver meade: pid=10096 rfd=0 wfd=10 efd=14
2021-02-01T00:05:55: Driver 111: stderr EOF
2021-02-01T00:05:55: Driver 111: restart #4
[... 407 lines cut ...]
Child process 10057 died
2021-02-01T00:03:03: Driver $: restart #8
2021-02-01T00:03:03: Driver $: pid=10058 rfd=0 wfd=5 efd=6
2021-02-01T00:03:03: Driver $: sending <getProperties version='1.7'/>

2021-02-01T00:03:03: Driver $: 2021-02-01T00:03:03: Driver $: execlp: No such file or directory
Child process 10058 died
2021-02-01T00:03:03: Driver $: stderr EOF
2021-02-01T00:03:03: Driver $: restart #9
2021-02-01T00:03:03: Driver $: pid=10059 rfd=0 wfd=5 efd=6
2021-02-01T00:03:03: Driver $: sending <getProperties version='1.7'/>

2021-02-01T00:03:03: Driver $: 2021-02-01T00:03:03: Driver $: execlp: No such file or directory
Child process 10059 died
2021-02-01T00:03:03: Driver $: stderr EOF
2021-02-01T00:03:03: Driver $: restart #10
2021-02-01T00:03:03: Driver $: pid=10060 rfd=0 wfd=5 efd=6
2021-02-01T00:03:03: Driver $: sending <getProperties version='1.7'/>

2021-02-01T00:03:03: Driver $: 2021-02-01T00:03:03: Driver $: execlp: No such file or directory
Child process 10060 died
2021-02-01T00:03:03: Driver $: stderr EOF
2021-02-01T00:03:03: Driver $: Terminated after #10 restarts.
2021-02-01T00:03:03: good bye
```

TallFurryMan, 2021-02-01:
I am not sure you entered the command I gave you properly:
```
indiserver -vv $(which indi_dsi_ccd)
```
In any case, I realised there was an error in the driver I posted, that did impact other versions of the device :disappointed:. I pushed an update to fix the problem. I'll ping you when the updated driver is available in the nightly builds.

Geoff1801, 2021-02-01:
Thanks Eric. my mistake.

⁣Get TypeApp for Android ​

TallFurryMan, 2021-02-02:
@Geoff1801 Please try with `0.2~202102020955`.

@knro Should I update the package version, still retained at `0.2`? Should it read `0.4` now? This is minor for now, of course.

knro, 2021-02-02:
@TallFurryMan Yes go ahead.

Geoff1801, 2021-02-03:
having internet issues unable to access landline access. However unsure how to update indi with package 0.2-202102020955.

TallFurryMan, 2021-02-03:
Refer to https://indilib.org/download.html and install "indi-full". When you execute "apt show indi-dsi" you should see this version.

Geoff1801, 2021-02-03:
After installing indi full and running "apt show indi-dsi" this is a copy of my terminal .

```geoff@geoff-HP-ProBook-6560b:~$ apt show indi-dsi
Package: indi-dsi
Version: 0.2~202102031131~ubuntu20.04.1
Priority: extra
Section: science
Maintainer: Jasem Mutlaq <mutlaqja@ikarustech.com>
Installed-Size: 236 kB
Depends: libc6 (>= 2.14), libgcc-s1 (>= 3.0), libindi1, libstdc++6 (>= 5.2), libusb-1.0-0 (>= 2:1.0.9), fxload
Download-Size: 55.0 kB
APT-Sources: http://ppa.launchpad.net/mutlaqja/indinightly/ubuntu focal/main amd64 Packages
Description: INDI Meade DSI Pro I/II Driver
 Driver for Meade DSI Pro I/II camera.
 .
 This driver is compatible with any INDI client such as KStars or Xephem.

N: There are 2 additional records. Please use the '-a' switch to see them.
```

Geoff1801, 2021-02-03:
After installing the latest nightly version and running "apt show indi-dsi" this is a copy of my terminal.
```geoff@geoff-HP-ProBook-6560b:~$ apt show indi-dsi
Package: indi-dsi
Version: 0.2~202102031131~ubuntu20.04.1
Priority: extra
Section: science
Maintainer: Jasem Mutlaq <mutlaqja@ikarustech.com>
Installed-Size: 236 kB
Depends: libc6 (>= 2.14), libgcc-s1 (>= 3.0), libindi1, libstdc++6 (>= 5.2), libusb-1.0-0 (>= 2:1.0.9), fxload
Download-Size: 55.0 kB
APT-Manual-Installed: no
APT-Sources: http://ppa.launchpad.net/mutlaqja/indinightly/ubuntu focal/main amd64 Packages
Description: INDI Meade DSI Pro I/II Driver
 Driver for Meade DSI Pro I/II camera.
 .
 This driver is compatible with any INDI client such as KStars or Xephem.

N: There is 1 additional record. Please use the '-a' switch to see it
geoff@geoff-HP-ProBook-6560b:~$ 
```
Using KStars to take an image I still get "unsupported Bayer pattern"

TallFurryMan, 2021-02-04:
`20210203` is indeed a newer timestamp than `20210202`, so that's the right driver.

Could you attach a FITS frame captured with the driver to this thread? I'd like to see if and how the Bayer pattern was written to the header.

Geoff1801, 2021-02-04:
Fits frame :
[Light_1_secs_2021-02-04T14-06-38_005.fits.tar.gz](https://github.com/indilib/indi/files/5922963/Light_1_secs_2021-02-04T14-06-38_005.fits.tar.gz)

Kstars log:
```
[log_14-33-43.txt.tar.gz](https://github.com/indilib/indi/files/5923038/log_14-33-43.txt.tar.gz)

I hope this is what you are looking for.

TallFurryMan, 2021-02-04:
OK there is a Bayer pattern keyword, but it is empty.

However, if I add it manually, I get a proper RGB frame (although I have no idea if the pattern is correct). We're nearly there.

[image]

This is zoomed, of course.

Geoff1801, 2021-02-04:
That is sounding positive. As you can probably tell that was taken with the lens cap on.  Would a landscape image help?

TallFurryMan, 2021-02-04:
So while going through the forum, I realise I missed @knro's comment at https://indilib.org/forum/ccds-dslrs/608-meade-dsi-load-firmware.html?start=18#3280, which gave the right answer from the beginning. Next time I'll be more thorough.

I pushed #314 to make sure the Bayer pattern is written to the FITS header. @Geoff1801, from the forum thread, it seems you may be able to test various Bayer patterns (RGGB, BGGR...) with landscape images and see which one is correct from the INDI panel (see the post just before the one I linked, in the screenshot the Bayer field is "Filter").

You also need to test the 2x2 binning mode, because I believe that the driver outputs a monochrome frame in that case, thus the Bayer pattern should be cleared by the driver when this mode is set.

I'll ping you when the change is available from nightly.

Geoff1801, 2021-02-04:
[GBRG.fits.tar.gz](https://github.com/indilib/indi/files/5928738/GBRG.fits.tar.gz)
The pattern appears to be GBRG.

TallFurryMan, 2021-02-04:
Updated PR #314 to reflect your findings.

TallFurryMan, 2021-02-04:
I'll fix the version later this week.

Geoff1801, 2021-02-05:
I sent you the wrong image yesterday. My error. These should show the image at the GBRG pattern
[GRBG-test1.tar.gz](https://github.com/indilib/indi/files/5935093/GRBG-test1.tar.gz)
[GRBG-test2.tar.gz](https://github.com/indilib/indi/files/5935098/GRBG-test2.tar.gz)

Geoff1801, 2021-02-06:
As you predicted the binning 2x2 is still mono
[Bin2x2.tar.gz](https://github.com/indilib/indi/files/5936162/Bin2x2.tar.gz)
 None of the images (binned 1x1 or 2x2) are shown as colour in my image processing programs.  Including Siril  in which I altered the rgb pattern to GBRG. Despite the fits header giving the correct GBRG pattern.  Just wondering if you have any thoughts on that.

TallFurryMan, 2021-02-06:
Yes the documentation states that 2x2 binning will return a mono frame, because it consolidates the full Bayer pattern into one photosite. So I'll need to remove the Bayer pattern header keyword when switching to this mode, for this sensor. Thanks for testing.

Could you confirm you ran the test with the latest driver, and that as a result you did not have to configure the pattern to get the frame to display in the viewer?

Geoff1801, 2021-02-06:
I haven't updated the driver for a couple of days, so that was the same version as the last log I forwarded. I did have to replace the RGGB pattern with GBRG.
Do you want me to update to the latest nightly version?

TallFurryMan, 2021-02-06:
Yes, please do. There will be another iteration in the next days, but I'd like you to confirm the behaviour.

Geoff1801, 2021-02-06:
I updated to the latest nightly build however cannot connect the dsi. I get the message    "Indi driver meade dsi crashed". Kstars log attached.

[log_07-34-01.txt.tar.gz](https://github.com/indilib/indi/files/5937603/log_07-34-01.txt.tar.gz)

TallFurryMan, 2021-02-07:
Thanks for reporting. My fault again :disappointed: I'll issue a fix. Sorry again @pcxz

TallFurryMan, 2021-02-07:
I pushed #323 to fix that regression.

Geoff1801, 2021-02-07:
Tried to connect again, however nightly version still the same 2021.02.06 T00::32Z     I will keep trying until nightly version is updated. Bad luck I was going to try the dsi out tonight supposedly a relatively clear night for a change supposedly.

TallFurryMan, 2021-02-07:
Yes that's unfortunate. In this situation remember you can uninstall the nightly build and revert to the stable branch.

Geoff1801, 2021-02-07:
Of course 😊. Thanks

⁣Get TypeApp for Android ​

Geoff1801, 2021-02-09:
Attempted to install the Nightly version however it failed to update with the message "to many redirects".  I don't know what this means.

Geoff1801, 2021-02-12:
Ignore the last  message. I have been having Broadband internet problems so that message was related to that not INDI. I have fixed the internet problems for now at least.

TallFurryMan, 2021-02-12:
Thanks, let us know how it goes.

Geoff1801, 2021-02-16:
Hi Eric,
Hopefully the internet problems are behind me.
I have two identical lap tops on which I do my Astronomy. They are both the same model units and running on the latest version of Linux mint and are close to identical. The one that I have been following your adjustments on ( my No.2 ) is imaging in colour providing that I put the GBRG colour pattern in 1x1 binning. Still mono in 2x2 binning. Attached is copy of the log.
[log_12-43-30-Test2.txt.tar.gz](https://github.com/indilib/indi/files/5985531/log_12-43-30-Test2.txt.tar.gz)

With the other laptop (No.1) which I was keeping  as a comparison, I have updated to Nightly version. 
Build: 2021-02-12T06:03:15Z
This is all mono, no colour showing at all. I attach a copy of the log and an image .
[log_12-26-16-Mono-No.1.txt.tar.gz](https://github.com/indilib/indi/files/5985585/log_12-26-16-Mono-No.1.txt.tar.gz)
[Test1_Light_001.fits.tar.gz](https://github.com/indilib/indi/files/5985587/Test1_Light_001.fits.tar.gz)

I am happy with the one computer working , it is just that I thought with the changes it probably should have brought this one to the same stage as the other.  I am just waiting for a clear night to test it.
Regards Geoff

TallFurryMan, 2021-02-16:
Thanks, I will check your results.

SteveBz, 2021-02-20:
Hi Guys,
  
  
   
  
  
   I also have been playing with the mono version of this camera for the last few weeks (on the very occasional clear nights we've had). I thought I was immune from the problem that Geoff describes because I was using the mono, but it turns out it is not so.
  
  
   
  
  
   I've unclicked the auto-debayer checkbox in the 'fits options' tab, and then what I get is an un-debayered image, ie an image with apparently large pixels! But it's mono! Not sure what more I can say, except it's a bit strange.  I would also go further and say it's a problem with the fits file and not the viewer, as  I get the same issue if I look at the images with alternative viewers such as Siril.
  
  
   
  
  
   I'm using an old mono DSI to do some spectroscopy and in fact there are a myriad of problems, this is only one. However here...:
  
  
   
  
  
   https://stargazerslounge.com/topic/372310-how-to-improve-these-sa100-results/
  
  
   
  
  
   ...you can see examples of the mega-pixilation I'm getting.
  
  
   
  
  
   In the mean time I'm just going to 'bin' my way around it, but it would be nice if I could get this resolved.
  
  
   
  
  
   Have a good day.
  
  
   
  
  
   Regards,
  
  
   
  
  
   Steve.

TallFurryMan, 2021-02-20:
I had a really quick look at the results from @Geoff1801 and not that much time to look into them. Without hardware it's kind of difficult, but I think there should be four 1Kx1K frames in the 1x1 dsi3 color payload, one for each  color-filtered photosite. With 2x2, there should be a single 1Kx1K frame with the four pixels accumulated, and this frame will always be mono. What are de dimensions of the dsi3 mono version? I'd expect 2Kx2K.

SteveBz, 2021-02-20:
Hi People,

The sensor is 1360 x 1024 by 6.45 μm pixels.  If I look at the .fits header it says the same thing.  Is that what you are asking?

NAXIS = 2
NAXIS1 = 1360 
NAXIS2 = 1024

Regards

Steve.

Geoff1801, 2021-02-20:
[M_42_Light_10_secs_2021-02-20T22-09-31_003.zip](https://github.com/indilib/indi/files/6016863/M_42_Light_10_secs_2021-02-20T22-09-31_003.zip)
Attached is a image of M42, it was just a 10 second shot. As you can see it all worked well. Even if the optics on this scope are dirty.
I was really pleased that Kstars remembered the GBRG pattern.  No problem with the 2x2 being mono. Now I just need to get my alternate laptop recognizing colour.

SteveBz, 2021-02-21:
Hi Guys,

I don't really understand what I'm see here.  It looks like mine.  An undebayered image with the same dimensions as mine. 

And I can see the debayering pattern.  Is it that mine is actually colour?  Is that what my problem is?  It could be, I got it from a friend who didn't know what it was and hadn't used it.

Kind regards

Steve.

[6 more replies]

closed 2021-04-17 by knro