indilib/indi #1333: celestrongps: Allow unattended recording for multiple PEC cycles

interplanetarychris, 2021-01-21:
**Is your feature request related to a problem? Please describe.**
Currently, the "Record" option for a PEC cycle terminates after recording one worm cycle.

**Describe the solution you'd like**
Recording multiple PEC cycles (to support averaging/smoothing) requires the user to "notice" that a recording cycle has been completed, and then to press Record again to initiate a new cycle.  This prevents recording multiple cycles in an unattended mode.

**Describe alternatives you've considered**
Currently, this requires user interaction.

**Additional context**
A potential solution to implementing this feature request would be a checkbox (default, unchecked, to preserve current behavior) which instructions INDI to "Continue new Record cycle after current cycle completes."

To terminate collection after the current cycle, the user could uncheck the box, or click the "Stop" button.

Chris-Rowland, 2021-01-22:
I think this belongs in an application, not the driver.
The application would monitor the PEC record and when it finishes download the recording and start a new record.  It would also manage guiding control so PEC could be aborted when the guiding fails.

Finally the application would average the multiple PEC records and upload them to the motor controller.

This is similar to the Celestron PECTool application.

The application could work with other drivers - probably needs an INDI PEC standard.

Chris

interplanetarychris, 2021-02-11:
I have no opinion about where it would be implemented - having spent some time in the celestron driver code, it was clear out to approach it within that context.   

I'll take a look at your idea for the application level - and consider levels of enhancement that get towards PECTool functionality.  I'll also gander at other mount driver PEC functions and see how they typically run their cycles.

interplanetarychris, 2021-05-18:
As I no longer have a Celestron mount, I'm no longer interested in the functionality, nor can I test it, so I am closing the request.

closed 2021-05-18 by interplanetarychris