indilib/indi #1336: Program readout mode of qhy268c camera per job in sequence queue / fix streaming issue

SebNik1, 2021-01-23:
**Problem:** I tried to program taking a larger amount of darks for the night before and I was not able to program a switch of the readout mode of my qhy268c in the scheduling queue. I am able to add the readout mode in the properties area, but the setting is then applied in general for all pictures to be taken in the queue.

**Suggestion:** I would be very happy if the readout mode could be added to the particular job. Like taking 10 darks/lights/etc. in photographic mode (mode 0) at gain 30 and then take another 10 shots in gain 60 at the high gain readout mode (mode 1) for example. I'd be happy to help with testing (windows with indiserver/arm64linux on rpi4/macos).

**P.S. Streaming issue:** I also wanted to kindly ask, if any other qhy268c user around is able to use the streaming function. Mine is not working under linux and mac with kstars, neither with local server nor rpi4 indiserver. It is only working in different Win10 apps (firecapture for example). And thank you all for your effort on Kstars/Ekos, it is great!

SebNik1, 2021-01-23:
P.S. I apologize for the confusion by the number of edits!

Juergen889, 2021-01-25:
I have the same issue. See

https://github.com/indilib/indi-3rdparty/issues/269

and

https://indilib.org/forum/ekos/8462-weird-segmented-image-with-ekos.html

At the moment no changes, I'm waiting to get kstars 3.5.1 and libindi 1.8.8 on astroberry, perhaps this will solve the issue?

SebNik1, 2021-01-26:
> indilib/indi-3rdparty#269

Thank you, that's good to know I am not alone with this phenomenon!

Juergen889, 2021-01-28:
Update: I tried it with stellarmate (all updates until today, kstars 3.5.1, QHY SDK 20.12.23): no change, still the same problem!

SebNik1, 2021-02-04:
Hello everyone, there is a new QHY SDK 
https://www.qhyccd.com/html/prepub/log_en.html#!log_en.md

Log
Fixed: cannot get frame continuously under Linux
Fixed: new chip and driver matching problem under Windows
That sounds very promising!

Kind regards
Sebastian

Juergen889, 2021-03-04:
Hi Sebastian,
I have updated to Kstars 3.5.2, indi 1.8.9 and QHYSDK21.02.11. Streaming still not possible. Only change is that the weired picture dont occur again...

Do you see any changes?

nrackwit, 2021-03-04:
There is some work that I had done in the QHY CCD driver with respect to streaming and it just got pushed into master: 

[Indi QHY streaming work](https://github.com/indilib/indi-3rdparty/pull/341)

I am hoping that this update should address the issues that you are describing and was wondering if you could help test this code with your setup?

Best regards
Niels

Juergen889, 2021-03-05:
Hi Niels,
I would like to test the driver, but I have really no experience with compiling and such things. 

I think its only one file after compiling, right? Is it possible to get the "driver-file" from you, so that I can copy it to my system and backup the old driver? I'm using stellarmate with all updates (kstars 3.5.2, QHY-SDK 21.02.11, indilib 1.8.9) until today. 

Best regards

Jürgen

nrackwit, 2021-03-05:
Hi Jürgen

Thanks much for taking a look. 

Rather than switching out binary files I would recommend to compile from source directly on your system. I unfortunately don't have access to a StellarMate, but I just took a look at the StellarMate user guide and in the FAQ section there is a description for how you could access it on the Linux level (FAQ 17.14. "How do I update StellarMate"). This is really what you would need to do if you wanted to go that far. It shouldn't be too much of a challenge though and really should work straight forward. 

[There is a good description here](https://github.com/indilib/indi-3rdparty) for how to compile an individual indi driver like the qhy_ccd driver. Since the new code is already part of the 3rd-party master branch you would just need to follow the steps as described, i.e. for the qhy_ccd driver first get the source code:

>mkdir -p ~/Projects
>cd ~/Projects
>git clone --depth=1 https://github.com/indilib/indi-3rdparty

Then build the driver:

>mkdir -p ~/Projects/build/indi-qhy
>cd ~/Projects/build/indi-qhy
>cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-qhy
>make

So far this will just build the binaries without  installing them. To install the driver so ECOS can use it, you will need to run:

>sudo make install

This would now be the change that overrides the qhy-ccd driver binary which is currently installed. In order to get back to your installed Indi v1.8.9 you would need to do the same steps as before but instead use the driver code for that particular release in the compilation step. You can get your current source code by using:

>git clone --depth=1 -b v1.8.9 https://github.com/indilib/indi-3rdparty.

To be on the safe side, you could start with getting this version of the driver code and just see if you are able to compile your current version at all, without actually installing it. 

If you don't feel comfortable going that far, please don't. Otherwise I'd of course be looking forward to hear how things worked out for you.

Best regards
Niels

Juergen889, 2021-03-05:
Hi Niels,
I was just working on doing a backup of my stellarmate to try compiling... Before doing this Ithought I first look for updates, and there was an lib-qhy update. I tried streaming again: now it works!!! Thank you for working on the driver :-) 

Best regards

Jürgen

SebNik1, 2021-03-05:
Hi Niels, hi Jürgen, I am sorry I was not able to test anything yet but I made contact with qhy because of the problem about two weeks ago So I will check for the sdk update you mentioned. Thank you very much for your effort!

nrackwit, 2021-03-05:
Thank you folks for taking a look.

@SebNik1, you might still need to update to the new driver code as well as it has some changes in the way the camera gets initialized for going into streaming and back into exposure mode. I'm thinking that for Juergen's test the new driver code also installed when he ran his update. 

@Juergen889, could you take a look at what version number is being displayed for the qhy_ccd driver in the Indi console under the "General Info" tab when you have your camera connected?

Juergen889, 2021-03-05:
Hi Niels,
the driver is version 2.8. I think its the newest from github. 

Best regards 

Jürgen

nrackwit, 2021-03-05:
Ok, yes - v2.8 is the latest revision from the 'QHY streaming work' PR. Thanks for checking!

SebNik1, 2021-03-06:
Hello Niels,

thank you for the description. I build Indi and the qhy driver and installed them successfully according to to terminal output. In kstars I still get Version 2.7 shown. Is that normal? I will try streaming now. 

Kind regards Sebastian

SebNik1, 2021-03-06:
Hello Niels,

in the indi control panel it is now 2.8 after connection to the camera via my rapberry pi. I can start the stream after a longer initialization period. This is a great improvement, thank you!

Right now I got the problem, that from that point on it floods me with "2021-03-06T10:36:36: [WARNING] Frame buffer is full, skipping frame… „ messages. I cannot stop streaming nor can I disconnect indi. Its just stuck in that mode. When I force a shutdown on my Macs kstars and reconnect I cannot connect to the indi terminal anymore but I still get the warning. 

On the raspberrys Kstars the streaming does not start. I tried waiting for about 5 minutes. I enabled debugging but I could not jump to the protocol folder to find it…

Sebastian

SebNik1, 2021-03-06:
P.S. once the streaming from indiserver to my kstars on mac is started I cannot change the framerate or stop the stream anymore. It continues streaming, I cannot disconnect from the indi server. The only way to stop it is to restart the server. Please let me know if and how I can provide accurate data for you if you wish!

nrackwit, 2021-03-06:
Hi Sebastian

Sorry to hear you're having trouble. As a first step let's try to find your log files.
On your Mac the default location is under ~/Library/Application Support/KStars (in Finder enter the "go to" menu and hold down the option key to get the Library location). If I remember correctly on the Raspberry PI by default they are under ~/.local/share/kstars.

Looking at the frame buffer warning that you are describing it looks like you successfully started recording but then ran into some sort of performance limitations. Did you get a .SER file for that session at all (even an empty one)? 

Your QHY286c at 24MP probably has a frame size of around 6000x4000.
If you tried streaming at that resolution maybe start with a smaller subframe, i.e. 600x400 or 1280x960 (please use the "Image settings" tab, not the streaming tab to define that) and to start with also use a low frame rate, i.e. 2fps and then work your way up to what your hardware and camera may be able to handle. I am able to get 2000x2000 frames at 6-10fps on my RP4. 

Also check your USB traffic setting and see if they are set to 0.

I'm not sure what you mean by this:

>When I force a shutdown on my Macs kstars and reconnect I cannot connect to the indi terminal anymore but I still get the warning.

Are you forcing KStars to close via the terminal? That probably wouldn't stop the indiserver process running underneath and when you restart ECOS it will find and reuse it in whichever state it might be. Instead of terminating KStars can you disconnect the camera and kill the indiserver process via the 'Disconnect' and  'STOP' buttons next to the ECOS profile selector? For me too, during my tests INDI would sometimes lock up waiting for the first frame of the camera. Disconnecting the camera, stopping INDI and restarting everything would brings things back for me. This might be a topic for a different discussion though, I don't believe that it's related to the driver code at this time. 

Good luck.

Niels

nrackwit, 2021-03-09:
I looked a bit more into the "Frame buffer is full, skipping frame .." problem and was able to recreate on my MacBook. There is a related setting in the streaming tab that allows you to change the frame buffer size. By default it is set to 512 MB and there is a recommendation in the code to at least set it to 2048MB. For my test this helped  to get a stable stream without skipping frames. 

It is messy, though. Once those messages started coming in, my whole KStars GUI froze, spindump started and I for my part had difficulties reaching the Activity Monitor to kill the driver. Implementing some sort of flood control into the driver unfortunately isn't straight forward as the underlying INDI streamer object doesn't communicate to the camera driver when this is happening. 

With respect to your problem on the Raspberry in which streaming doesn't start, this is indicative of the underlying SDK call not successfully getting it's first frame from the camera. You could test this with the small streaming test app that comes with the qhy-ccd driver. When the camera is connected, in the terminal you can enter 

>$qhy_video_test

This program runs the SDK command sequence to initiate streaming in full frame mode. This test usually fails for the bigger QHY294M on the systems that I have access to (GetQHYCCDSingleFrame failure, error: -1) but works for the smaller QHY5. There are quite a few posts around this on the QHY forum. I would agree that the driver code could probably handle this condition better as there is currently a retry implemented which can take its time. And during this period it doesn't seem to be possible to interrupt or stop this sequence through GUI interactions. 

The best indicator for when the stream has been successfully established is when the frame rate values in the streamer tab start updating and the stream preview shows a life frame rate in the upper right corner. 

Below a screenshot from my RP4:

[image]

Hope that helps.

nrackwit, 2021-03-12:
OK, I see - under libqhy. I could give that a shot. Thanks for pointing it out.

openastroproject, 2021-03-14:
RUN+="/bin/sh -c '/bin/echo 256 >/sys/module/usbcore/parameters/usbfs_memory_mb'"

has always been a poor idea IMNSHO, because it may lower a value that has already been raised.  I have used:

RUN+="/bin/sh -c 'test -f /sys/module/usbcore/parameters/usbfs_memory_mb && test `cat /sys/module/usbcore/parameters/usbfs_memory_mb` -lt 256 && echo 256 >/sys/module/usbcore/parameters/usbfs_memory_mb'"

knro, 2021-03-14:
I agree. Or we can completely leave it out and let the system decide what is best.

nrackwit, 2021-03-17:
One other thought that had crossed my mind was to see if there is a way for streamer::newFrame to report that there are dropped frames, i.e. by changing the method signature from void to int or by adding a new call-by-reference variable (or - for backward compatibility - adding a new method) so the camera driver can implement a check for when it happens and if a threshold number is reached (i.e. another user defined limits setting in the streamer tab) to terminate streaming.

knro, 2021-03-22:
@openastroproject Thank you, I've added your condition now.

closed 2021-03-22 by knro

xsnrg, 2021-10-14:
I maintain a set of COPR repositories for Fedora that build off master. Details are here:  [Fedora INDI/kstars COPR](https://indilib.org/get-indi/download-fedora.html). 1.9.2 should have the fix in it though. Are you sure you don't have another older version of INDI installed somewhere on the system?

Safari77, 2021-10-14:
I have only this one (1.9.2), the error on USB device plugin does not appear when I delete file `/usr/lib/udev/rules.d/99-indi_auxiliary.rules`.

xsnrg, 2021-10-14:
#1546