indilib/indi #1345: PyIndi fails over missing symbol

gnthibault, 2021-02-03:
It is not clear for me wether it is a problem of the binaries hosted on pypi or indi libs themselves.

Problem is exposed here:
https://indilib.org/forum/development/8727-problem-importing-pyindi.html

gnthibault, 2021-02-04:
@pawel-soja  can you please check the problem explained here: https://indilib.org/forum/development/8727-problem-importing-pyindi.html

I suspect the macro stuff you included in github.com/indilib/indi/blob/master/libs/indibase/indiutility.h is breaking swig.

Did you tried to build pyindi against this code with the macro ? I cannot do it personnally

gnthibault, 2021-02-04:
@knro I propose building pyindi should be part of the build test. Otherwise we cannot be aware if code breaks swig.

pawel-soja, 2021-02-04:
Hi, I'll take a look at it.

pawel-soja, 2021-02-05:
It should be OK for the latest version of PyIndi.
If there is any problem, please let me know.

I see the apply/define functions were also crashing, but you already blocked them for another reason ;)
I will inform you when I clear the matter of indiutility and indiproperty.

Please take off your offer on issuehunt.

geehalel, 2021-02-07:
@pawel-soja Thanks for having a look (and solving) this issue. I saw you were refactoring some parts of the indi base library and was waiting you end up before updating the pyindi-client.
@gnthibault Please

1. create issues in the right place, namely pyindi-client
2. read [existing issues](https://github.com/geehalel/pyindi-client/issues/6#issuecomment-768358449) (eventually closed) in the right place:

> Thanks for trying to debug, it's not easy with swig and python...
I don't have time at the moment to have a look into the problem (next stay-at-home here in a few days presumably).
I know that indi is currently rewriting base code, and this happens in the master branch. I really can not follow every indi commits to be in sync, furthermore some client side includes have been put in driver side include files (i should make a copy otherwise the wrapper will include all the driver code for absolutely no use, that sort of thing is not safe at all...)
The solution at the moment may be to use the indi release from november (1.8.7).

3. Creating an issuehunt offer here concerning pyindi-client is completely irrelevant. 5$ is insulting.
4. Considering your knowledge, you should be able to solve this easily.

geehalel, 2021-02-07:
I just tried to build pyindi-client with latest indi core commit on up-to-date Ubuntu 20.04. That compiles fine but the testindiclient.py script crashes when trying to iterate over a switch property:
```
elif p.getType()==PyIndi.INDI_SWITCH:
            tpy=p.getSwitch()
            for t in tpy:
TypeError: 'SwigPyObject' object is not iterable
pure virtual method called
```
It seems that Property get* methods are pure virtual now.  Will have a more precise look to that later.

gnthibault, 2021-02-07:
> @pawel-soja Thanks for having a look (and solving) this issue. I saw you were refactoring some parts of the indi base library and was waiting you end up before updating the pyindi-client.
> @gnthibault Please
> 
> 1. create issues in the right place, namely pyindi-client
> 2. read [existing issues](https://github.com/geehalel/pyindi-client/issues/6#issuecomment-768358449) (eventually closed) in the right place:
> 
> > Thanks for trying to debug, it's not easy with swig and python...
> > I don't have time at the moment to have a look into the problem (next stay-at-home here in a few days presumably).
> > I know that indi is currently rewriting base code, and this happens in the master branch. I really can not follow every indi commits to be in sync, furthermore some client side includes have been put in driver side include files (i should make a copy otherwise the wrapper will include all the driver code for absolutely no use, that sort of thing is not safe at all...)
> > The solution at the moment may be to use the indi release from november (1.8.7).
> 
> 1. Creating an issuehunt offer here concerning pyindi-client is completely irrelevant. 5$ is insulting.
> 2. Considering your knowledge, you should be able to solve this easily.

Ok looks like I definitely haven't set up this issue in the right way

There are actually 2 issues there:
* For a reason I still don't fully understand, there was a  missing symbol in a binary or missing static link step on recent version of pyindi. This issue indeed relates solely to pyindi-project version 0.2.6, I agree, and I have created an issue at the right place afterwards

* While trying to avoid this issue, I tried to downgrade pyindi to a previous version. And one cannot install any pyindi version nor any swig based bindings solution, from what I understand given the newest macros in indilib main project. And for me this is an important concern that needs to be raised.

As "pyindi" is advertised as is on the main indi website, I did not realized the projects were completely separated (in my opinion, if pyindi bindings are not part of indi build, then it should not be advertised on [main indi website](https://indilib.org/support/tutorials/166-installing-and-using-the-python-pyndi-client-on-raspberry-pi.html) ).

Regarding the insulting bounty
At the moment issuehunt for indilib project bounties ranges from 2$ to 20$, with median bounty at 5$ so the bounty cannot objectively considered an outlier, I am very sorry that you felt insulted.
In my view, main problem discussed here, is not a complex feature to implement or bug with complex reproducibility, but it was mostly about correcting a well identified recent commit featuring a macro that introduced a compatibility issue with swig (with all versions of pyindi, or all potential future swig bindings anyone would like to build).

Lastly, I am definitely willing to help, however I have no experience with swig (only a bit with pybind11), with a bit of help, I'll be 100% ok to try to help to fix, lets continue the discussion on pyindi issue.

gnthibault, 2021-02-07:
Thank you @pawel-soja and @knro everything is fine for me with aae54d9c5cc48ce6b074d85cd403e74d6fc47bb5

closed 2021-02-07 by gnthibault