indilib/indi #1350: No virtual destructor in _INumberVectorProperty and others

TallFurryMan, 2021-02-07:
While investigating issues found by the Coverity Scan on KStars, I stumbled upon a report stating that `_INumberVectorProperty` should have a virtual destructor.

```
316  */
    	1. no_virtual_dtor: Class _INumberVectorProperty does not have a virtual destructor.
317 typedef struct _INumberVectorProperty
318 {
319     /** Device name */
320     char device[MAXINDIDEVICE];
321     /** Property name */
322     char name[MAXINDINAME];
323     /** Short description */
324     char label[MAXINDILABEL];
325     /** GUI grouping hint */
326     char group[MAXINDIGROUP];
327     /** Client accessibility hint */
328     IPerm p;
329     /** Current max time to change, secs */
330     double timeout;
331     /** current property state */
332     IPState s;
333     /** Numbers comprising this vector */
334     INumber *np;
335     /** Dimension of np[] */
336     int nnp;
337     /** ISO 8601 timestamp of this event */
338     char timestamp[MAXINDITSTAMP];
339     /** Helper info */
340     void *aux;
341 } INumberVectorProperty;
```

This symbol is a struct, but it is actually used as a class in the KStars client.

Destructor of derived class, which might not be invoked on all instances (`indipropertyview.h`):
```
 37 template <typename T>
 38 struct PropertyView: public WidgetTraits<T>::PropertyType
 39 {
 40     using PropertyType = typename WidgetTraits<T>::PropertyType;
 41     using WidgetType = WidgetView<T>;
 42
 43 public:
 44     PropertyView()                                  { memset(this, 0, sizeof(*this)); }
   	 A1. dtor_in_derived: Class INDI::PropertyView<INumber> has a destructor and a pointer to it is upcast to class _INumberVectorProperty which doesn't have a virtual destructor.
 45     ~PropertyView()                                 { for(auto &p: *this) {p.clear();} free(widget()); }
 46
```

Example where a pointer to the derived class is upcast to a pointer to the base class (`sequencejob.cpp`):
```
233        if (customProp)
234        {
235            QMap<QString, double> numbers = i.value();
236            QMapIterator<QString, double> j(numbers);
   	B1. upcast: Example 1: Casting from a pointer to INDI::PropertyView<INumber> to a pointer to _INumberVectorProperty in customProp->getNumber().
237            INumberVectorProperty *np = customProp->getNumber();
238            while (j.hasNext())
239            {
240                j.next();
241                INumber *oneNumber = IUFindNumber(np, j.key().toLatin1().data());
242                if (oneNumber)
243                    oneNumber->value = j.value();
244            }
245
246            activeCCD->getDriverInfo()->getClientManager()->sendNewNumber(np);
247        }
```

There are also `unique_ptr` of these objects, which are examples where an instance of the base class is deleted.

This object (and some others probably) would benefit from being assigned at least an (empty) virtual destructor to fix the undefined behavior regarding deletion through inheritance.

pawel-soja, 2021-02-24:
Hi,

The INumberVectorProperty structure and the like cannot have constructors / destructors / methods because they are used in C. And PropertyView is an overlay on C structures, offering methods for data management. Unfortunately, the overlay has a destructor, it shouldn't be. However, the overlay is only used for casting to easily manage the data, so there is no serious error. 
Over time, the PropertyView class will expand and I will remove the destructor.

closed 2021-03-12 by knro

TallFurryMan, 2021-04-11:
I now relatively frequently observe exceptions due to "pure virtual method called" when the BaseClient of Ekos terminates a connection. This is making Ekos tests instable on a 1/10 basis. No further precise information, I am not debugging this for now. It would be interesting to have `indiserver` robustness tests in indilib to separate client and server issues.

pawel-soja, 2021-04-11:
In some classes, I noticed that a virtual function is called in the destructor. The problem arises when the derived class is destroyed and the base class destructor calls a virtual function. From what I remember, I have already solved one such problem.