indilib/indi #1352: Pegasus Astro Pocket Power Box Micro not working

jaybo007, 2021-02-10:
**Did you make sure there is no existing driver for your device? Please check [supported devices portal](https://indilib.org/devices].**
Please confirm you verified no drivers exist for your device.

A: Yes i did find a driver (PPB) in the indi list but it does not work with the unit.
I have spoken to Pegasus and they advise that they have referred the error to the indi development team.
They state there is an issue with recognising the device in indi.

**Provide link to product page**
Provide URL to the product description and details.

A: https://pegasusastro.com/pocket-powerbox-micro/

**Describe device family**
All INDI devices fall under one or more of the following categories:

A: Auxiliary (Anything else that does not fit above).

**Provide device communication protocol**
All devices use some sort of communication language between them and the host system. This could be called *Command Set* or *SDK*. It is best to provide a link to a document or PDF with such details from the manufacutrer. It needs to include the following information:
* Communication Method: Serial, USB, TCP...etc. along with the connection parameters.
* Communication Protocol: Detailed description of the language used to communicate with the device along with examples if available.

A: Unsure

**Hardware Sample**
Would it be possible to send the developer a sample hardware unit to develop the driver? If that's not possible, would it be possible to setup a remote environment with the devie attached?

A: unfortunately i cant supply one. Pegasus may be able to help.

knro, 2021-02-11:
IIRC, this was fixed in the nightly version. Can you try to compile from Github or use nightly drivers from the PPA?

jaybo007, 2021-02-12:
Hi Jasem,
Sorry for my lack of knowledge but can i install just the new PPB driver or should i switch over to the nightly build.
I'm not too good navigating github so if you could provide a link and a general install guide that would be awesome.
Kind regards
Jason
--
Sent from myMail for Android Thursday,

closed 2021-04-17 by knro