indilib/indi #1354: Enable Park&Unpark command for ioptron GEM45

JiangXL, 2021-02-12:
**Is your feature request related to a problem? Please describe.**
Ioptron GEM45(firmware 2101) is connected to indi_ioptronv3_telescope(v1.2), but I cannot use functions of Park and Unpark. But ioptron GEM45 actually support Park command.

```
[2021-02-12T13:16:16.297 CST WARN ][                     kf5.ki18n] - "Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered cal
l to '~stripAccel'."
[2021-02-12T13:16:16.297 CST WARN ][                     kf5.ki18n] - "Inter
polation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
[2021-02-12T13:16:16.298 CST INFO ][           org.kde.kstars.indi] - iOptron v3 :  "[ERROR] Parking is not supported in this mount model. 
```

**Describe the solution you'd like**
I found code in https://github.com/indilib/indi/blob/master/drivers/telescope/ioptronv3driver.cpp#L492, it should work if we change some declaration. Thanks!

JiangXL, 2021-02-12:
After a wider research, I find it is more difficult that I know.

https://indilib.org/forum/mounts/7901-parking-the-ioptron-gem45.html

knro, 2021-02-13:
Have you tested this because according to iOptron documentation, the parking/unparking commands are only supported for a subset of mounts.

JiangXL, 2021-02-13:
Thanks for your replay. I'm sure that I can park iOptron GEM45 at iOptron's Go2Nova hand controller.  But I'm not sure whether it is as the same meaning as park function in indi.

knro, 2021-02-13:
The INDI ones uses #:MP1 to park and #:MP0 to unpark. However, as ***per iOptron documentation*** these are limited to a subset of mounts. If you believe that they work on your mount model, you can edit the file to remove the limitation and check if they indeed work or not.

JiangXL, 2021-02-13:
Hi, I just get the iOptron RS-232 command set from iOptron. It may be helpful to update driver.
- [iOptron Mount RS-232 Command Language V2.52 EOL.pdf](https://github.com/indilib/indi/files/5975494/iOptron.Mount.RS-232.Command.Language.V2.52.EOL.pdf)
- [iOptron Mount RS-232 Command Language V3.10.pdf](https://github.com/indilib/indi/files/5975495/iOptron.Mount.RS-232.Command.Language.V3.10.pdf)

knro, 2021-02-13:
Please test the changes and report back if it is working now.

closed 2021-02-13 by knro

JiangXL, 2021-02-13:
Thanks for your hard work, I'll test it at tomorrow daytime.

rishi-garrod, 2021-02-13:
Just tested the ioptronv3 driver and it connects, slews and parks where you tell it to.
Thanks for this update, I was really missing the park.
I have the GEM45G (without the guider). Latest firmware.

JiangXL, 2021-02-14:
Park does works!

commit 162bd940 (2021-02-13): Add CEM26 and GEM28. Migrate CEM40 and GEM45 drivers to iOptronv3 commands and remove parking restrictions. Fixes #1354