indilib/indi #1359: test_base64 test fails on PPC and PPC64

thesamesam, 2021-02-25:
**Describe the bug**
Building indilib 1.8.8 on PPC and PPC64 results in all but one test passing: test_base64 fails on both architectures.

This was reported [downstream](https://bugs.gentoo.org/763126) in Gentoo but I've reproduced it myself.

I've attached the full build log and LastTest.log which is produced by CMake/ctest, please let me know if there's any more information I can provide. Thank you!

Snippet:
```

----------------------------------------------------------
1/7 Testing: test_base64
1/7 Test: test_base64
Command: "/var/tmp/portage/sci-libs/indilib-1.8.8/work/indilib-1.8.8_build/test/core/test_base64"
Directory: /var/tmp/portage/sci-libs/indilib-1.8.8/work/indilib-1.8.8_build/test/core
"test_base64" start time: Feb 25 08:37 -00
Output:
----------------------------------------------------------
Running main() from /var/tmp/portage/dev-cpp/gtest-1.10.0_p20200702/work/googletest-aee0f9d9b5b87796ee8a0ab26b7587ec30e8858e/googletest/src/gtest_main.cc
[==========] Running 3 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 3 tests from CORE_BASE64
[ RUN      ] CORE_BASE64.Test_to64frombits
[       OK ] CORE_BASE64.Test_to64frombits (0 ms)
[ RUN      ] CORE_BASE64.Test_from64tobits
/var/tmp/portage/sci-libs/indilib-1.8.8/work/indi-1.8.8/test/core/test_base64.cpp:56: Failure
Expected equality of these values:
  "FOOBARBAZ"
  (char *)p_outbuf
    Which is: "\x91\x13\xFD\x91\x4\x85\x91\x6\x85"
[  FAILED  ] CORE_BASE64.Test_from64tobits (1 ms)
[ RUN      ] CORE_BASE64.Test_from64tobits_fast
/var/tmp/portage/sci-libs/indilib-1.8.8/work/indi-1.8.8/test/core/test_base64.cpp:72: Failure
Expected equality of these values:
  "FOOBARBAZ"
  (char *)p_outbuf
    Which is: "\x91\x13\xFD\x91\x4\x85\x91\x6\x85"
[  FAILED  ] CORE_BASE64.Test_from64tobits_fast (0 ms)
[----------] 3 tests from CORE_BASE64 (1 ms total)

[----------] Global test environment tear-down
[==========] 3 tests from 1 test suite ran. (1 ms total)
[  PASSED  ] 1 test.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] CORE_BASE64.Test_from64tobits
[  FAILED  ] CORE_BASE64.Test_from64tobits_fast

 2 FAILED TESTS
<end of output>
Test time =   0.10 sec
----------------------------------------------------------
Test Failed.
"test_base64" end time: Feb 25 08:37 -00
"test_base64" time elapsed: 00:00:00
----------------------------------------------------------
```

**To Reproduce**
1. Run tests on a PPC or PPC64 system on (Gentoo) Linux

**Expected behavior**
All tests pass.

**Desktop (please complete the following information):**
 - OS: Gentoo GNU/Linux
 - Version: 1.8.8, but downstream reports say it occurred on 1.8.7 too.

**Log Files**
[build.log](https://github.com/indilib/indi/files/6043354/build.log)
[LastTest.log](https://github.com/indilib/indi/files/6043362/LastTest.log)

ernsteiswuerfel, 2021-03-05:
I tried to bisect the bug but found out the test fails since v1.3 up to git-master (was not able to build v1.4.0 though).
```
[...]
[100%] Linking CXX executable tutorial_seven
[100%] Built target tutorial_seven
Test project /root/build/indi/build/indi/test
    Start 1: test_base64
1/1 Test #1: test_base64 ......................***Failed    0.01 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.02 sec

The following tests FAILED:
	  1 - test_base64 (Failed)
Errors while running CTest
```
[LastTest.log](https://github.com/indilib/indi/files/6090306/LastTest.log)
[CTestCostData.txt](https://github.com/indilib/indi/files/6090307/CTestCostData.txt)

knro, 2021-04-17:
Any ideas on what might be causing this issue?

thesamesam, 2021-09-07:
> Any ideas on what might be causing this issue?

It passes for me on ppc64le so I think it's (big) endian related.

knro, 2021-11-24:
Ok, any possible fix that does not involve performance regressions?

closed 2022-02-19 by knro

thesamesam, 2023-07-28:
Just for completeness: this was fixed in https://github.com/indilib/indi/pull/1593. Thanks all!