indilib/indi #1365: KStars FITS Viewer to display all bits >8 from .arw file captured by Sony A7S

jurgenfd, 2021-03-05:
**Is your feature request related to a problem? Please describe.**
In order to estimate a good exposure it's nice to have the histogram span the full range of values. As in the attached imagery; the viewer only displays the lower 8 bits value. This happens in my workflow when capturing thru Ekos as well as loading it straight up from KStars.

**Describe the solution you'd like**
The same behaviour as when the same data comes to the viewer thru the .fits format, namely: upscale to 16 bits.

**Describe alternatives you've considered**
Move to transfer by .fits but leaves me the problem that I now have to convert from .fits to .dng (or so). On macOS this is a problem as e.g. the Fits Liberaor doesn't run on the latest macOS anymore.

**Additional context**
In the images KStars FITS Viewer and PhotoShop (PS) views of the same attached file are shown. In PS the full range is shown when moved to 16 bit but not in the KStars FITS Viewer, there the cut off is at 255.

Is this the correct place to request the feature?

[DSC00154.ARW.zip](https://github.com/indilib/indi/files/6092978/DSC00154.ARW.zip)
[image]
[image]

knro, 2021-04-17:
Thanks, please file it here since it's a KStars issue: https://invent.kde.org/education/kstars/-/issues

closed 2021-04-17 by knro

jurgenfd, 2021-04-17:
Cool, reposted at: https://invent.kde.org/education/kstars/-/issues/83