indilib/indi #1368: CelestronGPS reporting wrong rates to telescope during pulse guiding

nrackwit, 2021-03-06:
**Describe the bug**
CelestronGPD::Guide() scales the [0..1] user defined sidereal rate to [0..255] instead of the expected [0..100] range when submitting guide pulses to the telescope, causing:
a) For guiding rate settings <0.5  (rate byte n [0..127]) the effective puls rate is 2.55x higher 
b) For guiding rates >0.5 (rate byte in [129..255]) guiding direction is reversed and  scaled by an amount specific to the chosen guiding rate.
c) For guiding rate setting = 0.502 (rate byte = 128) guide rate is zero.

**To Reproduce**
I ran guide calibrations with different guide rate settings, showing reversed RA or DEC direction (parity) for certain rate settings and discrepancies between the setting value of the guiding rate and the actual guiding rates as calculated by PHD2. Guiding can stop working if the calibration was created under a different version of the driver. Otherwise, if the calibration was obtained under the current version PHD2 does a good job of compensating for changed guiding rates and parities. 

I used the following test sequence to reproduce
1) In Indi control panel set guide rates, 
2) In PHD2 create new equipment profile
3) configure calibration sequence to use 25 steps for a max move of 25 px
4) Start calibration at Meridian and DEC=10º
5) Finish calibration and guide for 5 min using uni-direcational guiding based on my PA error.

Equipment: 
Celestron AVX, QHY guide camera with 2.9µm px and 130mm guide scope.
This guide setup provides  ~4.60 arc-s/px.

Test 1: RA guide rate = 0.25, DEC guide rate = 0.25 
RA parity = even
DEC parity = odd
Measured guide rate is ~ 1.8 px/s, which is an effective guiding rate of ~53% sidereal

[image]

Test 2: RA guide rate = 0.55, DEC guide rate = 0.55
RA parity = odd
DEC parity = even
Measured guide rate is ~3.6 px/s, which is an effective guiding rate of ~110% sidereal

[image]
 
Test 3: RA guide rate = 0.50, DEC guide rate = 0.75
RA parity = even
DEC parity = even
Measured RA rate is ~3.8 px/s, which is an effective guiding rate of ~120% sidereal
Measured RA rate is ~2.0 px/s, which is an effective guiding rate of ~60% sidereal

[image]

Test 4: Guide session with RA guide rate = 0.50, DEC guide rate = 0.75 using a calibration obtained with an earlier version of CelestronGPS:

[image]

**Log Files**
CelestronGPS log (with added logging to print out rate value in send_pulse() method):

[log_19-23-13.txt](https://github.com/indilib/indi/files/6094466/log_19-23-13.txt)

stuartatpeasy, 2021-03-08:
I'm not sure this adds any value to the conversation, but I would like to mention that I have encountered exactly the same problem as @nrackwit since upgrading to 1.8.9.  I'm unable to complete guider calibration because the second stage of the calibration process (RA_DEC_DIR) generates pulses in the same direction as RA_INC_DIR, and the guide star drifts away faster and faster as the pulse-length is increased.  I attach some logs below.

Thank you very much for all your work on INDI.

[guide_log_2021-03-07.txt](https://github.com/indilib/indi/files/6101490/guide_log_2021-03-07.txt)

knro, 2021-03-10:
@stuartatpeasy can you test PR #1369 ?

stuartatpeasy, 2021-03-10:
@knro wow, that was a fast turnaround - thanks Jasem.  The weather is bad here for the next day or so, but I will do a build from the PR and test it during clear skies (hopefully Thursday evening).

Thanks again for all the work you do on INDI.

interplanetarychris, 2021-03-10:
The skies cooperated for a bit last night and I had the opportunity to try this out on my CGX, and generally confirmed the observations.  However, perhaps I didn't notice it was a problem, as in my case, the mount was able to successfully calibrate with the parity errors and appears to be able to guide at each of the following rates: 0.25x, 0.50x, 0.80x, 0.85x.

I see that @nrackwit and @esqueletito have submitted fixes.  

I would recommend bumping the version up to 3.6, so we can better pinpoint problems/solutions as the code updates propagate.

https://github.com/indilib/indi/blob/7ce91b24fc333885bd9d474ba98363725c6e41ee/drivers/telescope/celestrongps.cpp#L93

I'll attempt to verify the fixes on my setup this week.

knro, 2021-03-11:
@interplanetarychris Good call, I've bumped the version to 3.6

interplanetarychris, 2021-03-13:
Quick report on trying out this version 3.6 celestrongps guide code.  Everything looks good on quick inspection.

I tried calibrating (and very brief guides) on all the rates combinations listed above.
For each, it calibrated easily, had the same parity for all rates, and the predicted/actual rates were at least within 10% of each other (usually better).

I'll do a more detailed inspection of the logs tomorrow.

It looks to me like this has solved the problem.

interplanetarychris, 2021-03-14:
Here's the series of calibrations on my Celestron CGX with celestrongps version 3.6 (pulled from GIT and compiled).  They were in the following order (same as above):

1. RA guide rate = 0.25, DEC guide rate = 0.25
2. RA guide rate = 0.55, DEC guide rate = 0.55
3. RA guide rate = 0.50, DEC guide rate = 0.75
4. RA guide rate = 0.75, DEC guide rate = 0.75

I didn't take out backlash between each, so the orthogonality varies a bit.

**RA guide rate = 0.25, DEC guide rate = 0.25 (3.76 a-s/s)**
[image]

**RA guide rate = 0.55, DEC guide rate = 0.55 (7.521 a-s/s)**
[image]

**RA guide rate = 0.50, DEC guide rate = 0.75 (3.76,7.521 a-s/s)**
[image]

**RA guide rate = 0.75, DEC guide rate = 0.75 (7.521 a-s/s)**
[image]

nrackwit, 2021-03-17:
Thank you @interplanetarychris for testing and verifying everything. To me the guide rates look ok now. 

Telling from @stuartatpeasy's guide log (the RA_INC_DIR  direction gets reversed from rate byte 128 overflowing into the sign bit) his situation should also be addressed.
 
Together with the earlier change which had already handled the hand controller settings adjustment I think that all should now be working as intended.

Thanks again everyone for the help!

stuartatpeasy, 2021-03-17:
I would like to echo @nrackwit's thanks to @interplanetarychris. If the 
UK skies ever clear, I hope to try out the new fixed build of the 
driver.  I have no doubt that it will work as intended!

Thanks all for your work on this project.

closed 2021-04-17 by knro