indilib/indi #1370: Meridian flip and true/mechanical telescope coordinate

geehalel, 2021-03-06:
Not sure if this is the right place, but that may concern mount drivers, clients and the INDI framework itself.
 
**Is your feature request related to a problem? Please describe.**
Users report meridian flip not occurring with the indi_eqmod driver when the mount is misaligned with a delta RA greater than their HA flip settings. I have a discussion in this [forum thread](https://indilib.org/forum/mounts/8930-meridian-flip-not-happening.html) where I try to explain my understandings.

**Describe the solution you'd like**
Clients should use true/mechanical mount coordinate to initiate a meridian flip. If clients use celestial coordinate to initiate the flip with a goto command, the indi_eqmod driver will prevent the mount to be counterweight up in the above situation and the flip won't occur. Adding a true/mechanical telescope coordinate in the INDI standard properties may help clients to easily determine the mechanical position of german equatorial mount. It may default to celestial coordinate for non german equatorial mount or mounts which can not sync. At the moment the only property related to the mechanical mount position is the side of pier which toggles when the mount crosses its meridian (not the celestial meridian). That's what I had in mind when I wrote the driver.  

**Describe alternatives you've considered**
- Add a flip command implemented with a goto which reverses the side of pier even if this results in counterweight up position. Thinking about that, consider the case where the user wants to start tracking with counterweight up (which is possible with indi_eqmod). The target star is east of celestial meridian. Misalignment may be such that the mount meridian is eastern to the target star. Not sure what will happen then.

**Additional context**
I created this issue because I was wondering how sync/alignment is managed in other mount drivers. Considering the lx200 driver which can sync, my feeling is that the firmware simply syncs the encoders. I believe lx200 mounts are fork mounts thus they do not care about flipping (no mechanical risk?). Another case is the syncan driver. The synscan HC also builds an alignment model I suppose, but I never checked how it behaves regarding meridian flip.
Not really related but I ask here: I see issues with dome slaving and was wondering why a dome driver has to consider the side of pier of a mount. Shouldn't it simply follow the celestial target, whatever is inside the dome ?

knro, 2021-03-06:
@geehalel Thank you for bringing this issue. There has been quite a bit of discussing, especially regarding the **actual** definition of side of meridian. @Chris-Rowland @fcasarramona IIRC discussed this at great lengths along with how it effects the dome tracking.

However, this incident with one user whose mount is not performing a proper flip worth investigating but is not necessarily warrant a complete revision of the current adopted procedure. Numerous users over the users used the EQMod meridian flip quite successfully. I believe there might be another factor for that particular user we may not be aware of. However, this issue can still be discussed here just to make sure all INDI developers are on the same page when it comes to how this information should be interpreted.

fcasarramona, 2021-03-06:
@geehalel The dome driver need the side of the pier because in GEM mount the telescope is not in the center of the dome, and this "missalignment" causes dome azimuth was different from telescope azimuth. This is especially true with big telescopes inside small domes.

geehalel, 2021-03-07:
@knro I don't want to change anything concerning the definition of the side of meridian. My concern is that in a driver which performs alignment (and physically anyway), there are two meridians as there are two coordinate systems (celestial and mount), which are tilted/rotated relative to each other. The driver by design wants to keep counterweight down in a goto and refers to the mount coordinate system for that purpose. That explains those "meridian flip not occuring".
Another point is that, depending on your alignment point set and method (nearest/nstar...), the tilt/rotation that the driver computes is not constant as the mount is tracking, and eventually the side of pier property may toggle several times when crossing the mount meridian (actually for the driver, the mount meridian is moving relative to the celestial meridian unless you use a standard sync, it is constantly moving for Nstar method and is moving "stepwise" for Nearest).
As you stated, those effects may remain negligible when your setup is well leveled and aligned, but for users who can't see the celestial pole, the issue will persist unless they set a fairly large HA flip point.
Lastly, alignment (particularly Nstar) has been made to increase goto accuracy in almost any situations (several degrees/hours misalignment), that's nice for quick visual observing sessions. But it seems that some users believe that alignment automagically provides a perfect mount setup and starts imaging whatever misalignment they get when platesolving/syncing. Imaging clients may warn them in that case and keep considering there is only one meridian for simplicity.

@fcasarramona  Thanks for explaining. Well, at first sight, nice (difficult) trigonometric/algebraic exercise...

fcasarramona, 2021-03-07:
@geehalel No so difficult when you realize it's a line to sphere intersection problem. :-)

Chris-Rowland, 2021-03-08:
I am not convinced that people here understand how meridian mode works.  Talking about meridian mode changing when the mount is tracking is a clear indication.  The pier side must not change as a result on the mount tracking.  Under no circumstances. Regardless of where the mount is pointing, or where the counterweights are.  The pier side only changes as a result of a slew through the pole.

The way to tell what pier side the mount is on is to look as the mount declination axis position.  It does not depend on the hour angle axis position.

Passing the mount axis positions to the user seems like abdicating responsibility for the driver and dumping the problem on the poor user, who probably does not need to know this and will know less than the driver author.

TallFurryMan, 2021-03-09:
> The way to tell what pier side the mount is on is to look as the mount declination axis position. It does not depend on the hour angle axis position.

I genuinely don't understand how this works. Isn't there another meridian on the north of the celestial pole? How do you make the difference between those quadrants and the southern quadrants if you don't look at the hour angle axis?

Chris-Rowland, 2021-03-09:
The only way I know to get people to understand this is for them to try 
it for themselves, look at what is happening, and think.

Get an equatorial mount.
Look at what happens to the axis positions as you track and as you 
change from one side of the meridian to the other doing a pier flip.
Make marks on the mount to see where it's actually moving to if it helps.

See how the mount axes move and over what ranges, as they approach the 
meridian, as they cross the meridian, when a pier flip is done.

Seen and think about how you convert between axis positions and the 
celestial Hour angle and Declination positions.  (Converting between Ra 
and Hour angle doesn't really depend on the mount). Start by assuming 
the mount is perfect.

Other places of confusion are when looking East and West.

Yes, there is another meridian (really the same one) to the North of the 
Celestial pole and a pier flip will be needed there as well.
I've no idea what a southern quadrant is.  There's no difference between 
positive and negative declinations.  The change comes for the axis 
declinations from +90 to 180 to -90.

And of course some things are reversed when you are in the Southern 
hemisphere.  It's rather like bolting your mount to the ceiling.

IMO a large part of the problem is that the terms used, especially pier 
side and expressing it as a direction.  People think they know what that 
means but it's misleading.  I prefer "Pointing state" and using terms 
for the two states such as "normal" and "reversed".  This separates the 
mount pointing state management from the topography.

Chris

geehalel, 2021-03-09:
@Chris-Rowland  Oups, you're totally right. I made a big mistake writing the side of pier property may toggle while tracking. I just wanted to express that the relative positions of true celestial sphere, driver celestial sphere (as displayed in kstars/clients), and driver mount axis sphere may evolve while tracking/guiding.
The problem for the driver is that it does not know what the client is doing: ideally the driver may decide by itself to perform the flip yet (it believes) counterweight is up and it will do that based on the mechanical position of the mount.
@TallFurryMan My understanding for visualizing the pointing state (let's call it the correct name as chris suggested) is to consider the vector product of the optical tube axis (positive to the objective/lens) and the RA axis (positive to the pole): if it points to the same direction as the DEC axis (positive to the mount saddle, negative to the counterweight), you are in the normal state.  In other words, the (optical axis, RA axis, DEC axis) coordinate system is direct in normal state, indirect in 'reversed/beyond the pole' state. I hope that I am correct here.
Anyway, a driver performing alignment manages two coordinate systems. You may completely hide that fact to the user, simply use celestial coordinate in the counterweight up management in gotos. That will suppress this 'meridian flip not occuring' error at the risk of being counterweight up in the real world in some situations. Which may already occur in other cases by the way (home position, location, time setting errors). That might be a solution too. But I would then highly suggest users unable to correctly align their mount to stop using indi-eqmod.
Well, the simplest way is to leave everything as it is now. I'm going back to my business and leave you close the issue (or will do that next week). Sorry for the inconvenience and thanks for taking time to answer.

geehalel, 2021-03-16:
@TallFurryMan I believe that the next time of flip is given using true/mechanical telescope RA and LST. 

Closing the issue.

closed 2021-03-16 by geehalel

knro, 2021-05-14:
Sorry to talk about this issue again, but I am still unsure about how exactly this works from the different sources which seem to add to the confusion. I know that sky coordinates can be reached via the mount in two different mechanical configurations. One is **Normal** and the other is **Reversed** or **Beyond the Pole**. I've actually experimented with this by marking a flag in KStars, then changing HA by 12 hours and then moving the DEC via the pole at twice the sky declination value to return to the same flag, so this part makes perfect sense. However, I got a bit lost at the criteria for normal vs reversed.

The criteria for deciding which is which appears to be related the *mechanical* declination. Presumably, mechanical declination is **zero** if we are looking at the intersection of the local meridian and the celestial equator. If the mechanical declination is within -90 to +90 then we are in the **Normal** state. Otherwise, if the mechanical declination is -180 to -90 or +90 to +180, then we are in the **Reversed** pointing state.

So I put this to the test, command the mount to slew to celestial equator. I selected a star with Declination close to zero, and its azimuth was ~178. The mount was in the pre-flip state as show in the image below. The Side of Pier was **West (Pointing East)**. This, according to [ASCOM](https://ascom-standards.org/Help/Platform/html/P_ASCOM_DeviceInterface_ITelescopeV3_SideOfPier.htm), is the *reversed* state. I also highlighted the vectors. @geehalel it indeed appears that in pre flip configuration, the cross vector product of Optical & RA axis point in the opposite direction of the declination axis, while in the post flip, it points in the same direction.

[image]

However, this does not match the other claim regarding zero mechanical declination and -90 to +90 for **Normal** state and everything else for **Reversed**. In the pre-flip image, why can't we consider this almost zero mechanical declination? It appears rather arbitrary? Both states are pointing at the same sky coordinates, how do we determine which of them is zero mechanical declination?

Would appreciate any insight @geehalel and @Chris-Rowland

Chris-Rowland, 2021-05-14:
> Sorry to talk about this issue again, but I am still unsure about how
> exactly this works from the different sources which seem to add to the
> confusion. I know that sky coordinates can be reached [...]
Yes, this is how the two pointing states should be defined.
> So I put this to the test, command the mount to slew to celestial
> equator. I selected a star with Declination close to zero, and its
> azimuth was ~178. The mount was in the pre-flip state as show [...]
There is, as far as I know, no consensus about which state is normal and 
which is reversed.  That's up to the mount manufacturer.  All that is 
required is that the two states are different.
>
> pointing_state 
> <https://user-images.githubusercontent.com/5986347/118220190-b8424b80-b483-11eb-905f-8f6d0c98bfe5.png>
>
> However, this does not match the other claim regarding zero mechanical 
> declination and -90 to +90 for *Normal* state and everything else for 
> *Reversed*. In the pre-flip image, why can't we consider this almost 
> zero mechanical declination? It appears rather arbitrary? Both states 
> are pointing at the same sky coordinates, how do we determine which of 
> them is zero mechanical declination?
Yes, I think it's arbitratary.  We (ASCOM and now INDI) are going 
slightly further by specifying the states to match the position of the 
OTA relative to the mount when the OTA is pointed towards the meridian 
at a declination of 0.  This is required so that the OTA sight line is 
defined and a dome slit can be position correctly. This is where the 
East and West properties come in.  This also allows the Ha limits and 
pier side checks to be implemented.  The Ha limit should only be active 
in the state where the mount has tracked past the meridian and there is 
a prospect of hitting a hard stop.

Hope that helps.

Chris