indilib/indi #1376: Unmet dependency on ActiveFocuser

rkaczorek, 2021-03-10:
Reported in rkaczorek/astroberry-server#130

Takahashi ActiveFocuser is missing dependency on libhidapi-hidraw0 - user needs to manually install libhidapi-hidraw0. Upon installation the /dev/hidraw0 has the wrong permissions which can't be read needing a new udev rule.

knro, 2021-03-11:
Thanks for the report... so what's the best way to go about this @rkaczorek ?

rkaczorek, 2021-03-11:
libhidapi-hidraw0 should be added to debian packaging control file & new udev rule should be added to 51-activefocuser.rules
`KERNEL=="hidraw*", ATTRS{idVendor}=="20e1", ATTRS{idProduct}=="0002", MODE="0666", GROUP="plugdev"
SUBSYSTEM=="usb", ATTR{idVendor}=="20e1", ATTR{idProduct}=="0002", MODE="0666", GROUP="plugdev"`

knro, 2021-03-11:
Thanks! Can you please submit a PR for this?

rkaczorek, 2021-03-11:
I have reconsidered this one and I think that adding libhidapi-hidraw0 dependency to libindi is not a good idea. It's a sledgehammer to crack a nut. I none of the other core drivers depend on this library.
What about moving this driver to indi-3rdparty? This way the driver will take care of it own dependency.

knro, 2021-04-17:
Sure, indi-3rd party would make sense given the extra dependency.

closed 2021-09-28 by knro