indilib/indi #1384: Conan package support

treaves, 2021-03-15:
[Conan](https://docs.conan.io/en/latest/index.html) is gaining in popularity as a C++ package manager.  I'd be nice to be able to consume libindi in this way.

knro, 2021-04-17:
Thanks, but there is no need to switch frameworks now given how stable CMake is.

closed 2021-04-17 by knro

treaves, 2021-04-17:
What does Conan package management have to do with CMake build system?  Well, nothing. One is a package manager - it manages finding, downloading, and availability of packages, and the other builds a product with source code.

knro, 2021-04-17:
Sorry, I confused it with a build system. At any rate, we have lots of packages, and are under staffed as it is. So currently it's released as source, with some Debian/Fedora packages.