indilib/indi #1388: Telescope Simulator RA wrong on first slew

TallFurryMan, 2021-03-19:
**Describe the bug**
When asked to slew for the first time, the Telescope Simulator is consistently wrong in centering the required RA coordinate.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start the Simulators profile in Ekos
2. Request Ekos to slew to a random object using the Telescope Simulator
3. Compare the object coordinates and the INDI panel coordinates
4. Request Ekos to slew again to the same object
5. Compare the object coordinates and the INDI panel coordinates

**Expected behavior**
3. Coordinates should match

**Screenshots**
First slew never matches coordinates:
[image]

Second and subsequent slews match coordinates:
[image]

**Desktop (please complete the following information):**
 - OS: Lubuntu 20.04
 - Version: git f6d8ecd9

**Log Files**
An example with KStars/Ekos `test_ekos_simulator`, in which I only logged the mismatch:
```
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos: "Telescope Simulator is online."
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos: "Guider port from Telescope Simulator is ready."
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Mount is unparked. "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos: All INDI devices are now connected.
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos.align: "Idle."
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos: "CCD Simulator is online."
WARNING: TestEkosSimulator::testMountSlew(M 41) HACK HACK HACK: Reset clock to initial conditions when starting Ekos
   Loc: [/home/tallfurryman/Documents/kstars/Tests/kstars_ui/test_ekos_simulator.cpp(41)]
QDEBUG : TestEkosSimulator::testMountSlew(M 41) Test slewing to 'M 41' RA '06h 46m 00s' DEC '-20° 45' 15"'
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos: "Guider port from CCD Simulator is ready."
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos.align: "World Coordinate System (WCS) is enabled. CCD rotation must be set either manually in the CCD driver or by solving an image before proceeding to capture any further images, otherwise the WCS information may be invalid."
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos: "CCD Simulator filter is online."
QWARN  : TestEkosSimulator::testMountSlew(M 41) QObject::disconnect: Unexpected null parameter
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Focuser Simulator :  "[INFO] No previous configuration found. To save driver configuration, click Save Configuration in Options tab. "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Loading device configuration... "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Dome Policy set to: Dome ignored. Mount can park or unpark regardless of dome parking state. "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Observer location updated: Latitude 51:28:05.9 (51.4683) Longitude  0:00:00.0 (0) "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Device configuration applied. "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos.mount: "Telescope info updated successfully."
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Saving device configuration... "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Device configuration saved. "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Saving device configuration... "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Device configuration saved. "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: CCD Simulator :  "[INFO] Loading device configuration... "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: CCD Simulator :  "[INFO] Upload settings set to client only. "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: CCD Simulator :  "[INFO] Setting current filter to slot 1 "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: CCD Simulator :  "[INFO] Device configuration applied. "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Slewing to RA:  6:46:00 - DEC: -20:45:15 "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: CCD Simulator :  "[INFO] World Coordinate System is enabled. "
WARNING: TestEkosSimulator::testMountSlew(M 41) Target 'M 41', RA 24360, slewed to RA 24368 (06h 46m 08s) with offset RA 8
   Loc: [/home/tallfurryman/Documents/kstars/Tests/kstars_ui/test_ekos_simulator.cpp(153)]
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Telescope slew is complete. Tracking... "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos: "Disconnecting INDI devices..."
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.indi: Telescope Simulator :  "[INFO] Telescope simulator is offline. "
QINFO  : TestEkosSimulator::testMountSlew(M 41) org.kde.kstars.ekos: "Telescope Simulator is disconnected."
```

TallFurryMan, 2021-03-20:
Keeping Ekos `TestEkosSimulator` in the `unstable` group for now.

knro, 2021-03-20:
@Chris-Rowland Could this be related to the alignment class mount to apparent HA, Dec conversion?

TallFurryMan, 2021-03-20:
As a side note, the default mount type is `CGEM`, but I don't think this has an impact. It's just slightly unexpected.

Chris-Rowland, 2021-03-20:
The difference is in Ra so probably in the Ha to Ra conversion, maybe a time change - and there's a warning about a clock reset.

TallFurryMan, 2021-03-20:
The clock reset happens at the beginning of the test, so that its execution is deterministic in terms of where stellar objects are positioned in the sky. The log is only informative here, it is what brought the issue to my attention. Screenshots, in particular the INDI panel, show that the issue appears at the driver level, and only on the first attempt. It might be an uninitialised variable somewhere.

TallFurryMan, 2021-03-20:
I will run a bisect on this driver when time allows.

Chris-Rowland, 2021-03-20:
None the less a difference in Ra only points towards a difference in 
time or longitude.  It's the area//I would start by looking at.

TallFurryMan, 2021-03-20:
That's right, the Telescope Simulator uses the Alignment class, which uses the system clock and not the date/time provided by Ekos.

I noticed this mainly because of seemingly erratic meridian flips. I was not getting the same test results in the morning and in the evening :roll_eyes: I must have been developing many tests without this change in my INDI simulator setup. No wonder some of them are unstable. :disappointed:

knro, 2021-04-17:
Any luck tracking this issue down?

TallFurryMan, 2021-04-17:
None yet.

closed 2024-04-20 by github-actions[bot]