indilib/indi #1390: The Driver for Pegasus Astro Ultimate Power Box crashes when run on Fedora 33

ebessant, 2021-03-19:
When running KStars/EKOS under Fedora 33 (KDE and Gnome), every time I start the driver for the "indi_pegasus_upb" (Pegasus Astro Ultimate Power Box V1), the process gets up to the full display of the "Connection" screen, and thenjust crashes.  While this unit works fine running under Ubuntu 20.10, my much preferred OS environment is Fedora.

All devices connected to the UPB operate normally from the Ekos menu, just NOT the UPB itself.  The attached Screen Recording shows the successful connection of the iOptron Mount, The Altair Astro CCD and the Pegasus Focuser Cube before I try to connect to the Power Box, which faithfully crashes.  The power box driver crashes when I try to connect to it before any of the other devices.

The attached screen recording shows the full process used to get the crash.  There are also 3 log files (1 for kstars itself and the other 2 for the Pegasus UPB driver crashes) produced for the same session as the screen recording.  Also attached are the ".zst" core files produced by the 2 driver crashes and saved under the "/var/lib/systemd/coredump/".  All of the above-mentioned files are contained in the zip file.

A couple of other points to note.

1) I repeat that this driver works fine under Ubuntu 20.10, but I had to do a special rebuild of one of my smaller machines to do the actual testing.  I will use this a temporary solution if I must, but I really don't get on well with Debian machines of any kind.
2) I did a rebuild of another of my machines with Fedora 33 but WITHOUT your repositories, and just used the out of the box Kstars/Ekos/INDI software available from the Fedora repositories.  The exact same driver crash occurs with these bits of software.
3) Jjust to further emphasise  the device version, note that it is the V1 version the the Pegasus Astro UPB that is crashing.  I do NOT have a V2 version of the device, so I don't know whether that crashes or not.
4) Finally, I have also reported this bug on the KDE bugzilla reporting site.  This link should point to that.  The reason I reported it first in KDE is because that is where "KStars --> Help --> Report BUG" sent me to.

https://bugs.kde.org/show_bug.cgi?id=434208

STEPS TO REPRODUCE
Make sure that a Pegasus UPB V1 power box is connected to the computer.

1. Start kStars from the "Applications" menu
2. Start the Ekos application (Tools -> Ekos)
3. Create a minimal profile with "Simulator" devices for Telescope, CCD and Guide, and an Auxillary Device for the Pegasus UPB.
4. Uncheck the Automatic Connect option in the profile editor screen.
5. Save the new profile, and close the Profile editor window.
6. Back in the Ekos main window, click on the play button.
7. Click on the top "Pegasus UPB"
8) In the "Main Control" window, click the "Connect" button.

The full list of properties and options for the device will be displayed before the device quickly crashes and is then rebooted by Ekos presumably.

Thanks very much,

Ed Bessant

**Expected behavior**
I expect the control panel for the UPB to remain running and allow me to monitor the electrical loads and so on for the attached devices.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

uname -a
Linux <hostname>  5.10.23-200.fc33.x86_64 #1 SMP Thu Mar 11 22:18:30 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

dnf list kstars indi-libs
Last metadata expiration check: 0:00:35 ago on Fri 19 Mar 2021 23:07:13 GMT.
Installed Packages
indi-libs.x86_64               1.9.0.git-20210319140858.fc33                  @copr:copr.fedorainfracloud.org:xsnrg:libindi-bleeding
kstars.x86_64                  1:3.5.3.git-20210319003411.fc33                @copr:copr.fedorainfracloud.org:xsnrg:kstars-bleeding 

**Log Files**
[2021-03-19.zip](https://github.com/indilib/indi/files/6174486/2021-03-19.zip)

knro, 2021-03-20:
@xsnrg Can you please take a look at this? why would it work on Ubuntu and not Fedora?

ebessant, 2021-10-04:
Hi there,
I am surprised that you cannot reproduce this.  While letting you schedule the issue, I tried a large number of different things here to see if I could get it working.  This included building a number of different virtual machines of Fedora 31, Fedora 32, Fedora 33 and Fedora 34 on Ubuntu and WinDROSS devices on which I knew the Pegasus UPB worked.  All to no avail.

After getting the email from your above report and closure of my issue, I then tried running things on a full Fedora 34 box, hoping that it has been sorted with the most recent bleeding release, but alas, no go.  In fat, it was even worse in that rather than crashing just the window for the UPB, it actually crashed everything right down to kStars.

Is it possible that you were testing the UPB Version 2, and NOT the UPB Version 1 that I use?  If you did test against the Version 1 UPB, can you please provide instructions on how you got it to work?

Sorry for the inconvenience and thanks in advance,

E.

Ed Bessant

knro, 2021-10-05:
Right, I have the V1 release and it works fine. This is an issue with the packaging in Fedora.

xsnrg, 2021-10-05:
I missed this somehow. I will take a look to see if I can reproduce it.

ebessant, 2021-10-07:
As requested, I attach a backtrace file created using the above command piped through "tee <FILENAME>.  Hopefully this will prove beneficial/useful for your analysis.

Thanks again,

E.

[Issue_1390_backtrace-PegasusUPB-20211007-3245.txt](https://github.com/indilib/indi/files/7298492/Issue_1390_backtrace-PegasusUPB-20211007-3245.txt)

knro, 2021-10-07:
Try to delete under file for UPB under ~/.indi and try again?

ebessant, 2021-10-09:
Oops.  My bad.  I missed that one.  Sorry about that.

Anyway, reverting to the default configuration did nothing.  Note that I did have to enable Port Scanning, since the default "/dev/ttyUSB0" was assigned to one of the Focus Cubes attached to the UPB.

Thanks and have a good weekend,

E.

ebessant, 2021-10-09:
Sorry for the delay in response, but the broadband to my boat has been a bit suspect today.

Thanks for the instructions.  Alas, I have not used docker before.  Reading between the lines, I presume I have to install "podman" AND "podman-docker" on the device,  Is this correct, or are there other things to install as well?

E.

xsnrg, 2021-10-09:
Yes, dnf install podman and podman-docker on the host machine

xsnrg, 2021-10-09:
No problem. I am not home now, and see I must have mucked up the first line of the docker file. Try adding this:

from amd64/fedora:34

ebessant, 2021-10-09:
Well, I got further on, but now I am getting a "manifest unknown" failure.

========

[ebessant@******** Docker]$ sudo docker build . 
[sudo] password for ebessant:
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
STEP 1/13: FROM amd64/fedora:34
✔ registry.fedoraproject.org/amd64/fedora:34
Trying to pull registry.fedoraproject.org/amd64/fedora:34...
Error: error creating build container: initializing source docker://registry.fedoraproject.org/amd64/fedora:34: reading manifest 34 in registry.fedoraproject.org/amd64/fedora: manifest unknown: manifest unknown
[ebessant@******** Docker]$ 

========

It's at time like this when I wish I had done more of these modern-day development techniques.  But Operational DBAs seldom have the time for studying things outside their day to day remit and stay sane.

E.

ebessant, 2021-10-09:
Hopefully another simple one.  Once the container build had completed, I ran the 2nd command you provided above, and I get provided with the list for images again, but it is the same as the list provided for the build.  Choosing "docker.io" again causes the below to be issued.  What am I missing?

========

[ebessant@******** Docker]$ sudo docker run -ti -e DISPLAY --rm -v /tmp/.X11-unix/:/tmp/.X11-unix/ -e XDG_RUNTIME_DIR=/run/user/1000 -e PULSE_SERVER=/run/user/1000/pulse/native --ipc host --net=host fedora34
[sudo] password for ebessant: 
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
? Please select an image: 
  ▸ registry.fedoraproject.org/fedora34:latest
    registry.access.redhat.com/fedora34:latest
    docker.io/library/fedora34:latest
    quay.io/fedora34:latest
 [sudo] password for ebessant: 
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
✔ docker.io/library/fedora34:latest
Trying to pull docker.io/library/fedora34:latest...
Error: initializing source docker://fedora34:latest: reading manifest latest in docker.io/library/fedora34: errors:
denied: requested access to the resource is denied
unauthorized: authentication required
========

ebessant, 2021-10-09:
Sorry.  Don't know how that font change happened.

ebessant, 2021-10-09:
The use of the hash did the trick, and I successfully got a session running in the container, and fired up kStars/Ekos.  Just one last problem, and that is getting the container to see the "/dev" entries from the main host machine on which the container was built.  Google search here I come.

ebessant, 2021-10-09:
[image]

xsnrg, 2021-10-09:
The docker container had 2 purposes. The first was to see if the bare container would run it. This is a very stripped down Fedora install. This helps rule out any local configurations you may have that could have affected the operation. The other reason for the docker container is this should let @knro see the failure, as I believe he has the same device. He won't have to load Fedora to reproduce it. Without the hardware, I am not sure what else I can do at this point.

ebessant, 2021-10-10:
OK.  Thanks for what has been done thus far.  Back to the Senators @ Maple Leafs before the start of regular season next week.

Have a good one,

E.

ebessant, 2021-10-20:
A quick follow-up to last weekend's investigations.  In last weekend's session, I built the Dockerosed Fedora 24 container on a full Fedora 34 machine.  Since then, I have successfully built the same Fedora 34 container on an Ubuntu 20.10 machine, and run kStars/Ekos/INDI Control Panel.  Agaiin, I have successfully reproduced the crash of the Pegasus UPB V1 device.

Thanks,

E.

xsnrg, 2021-10-20:
@knro , you have the same hardware, correct? Would you be willing to try the Docker container?

xsnrg, 2021-10-24:
So the only differences that I see are libtheoraenc and libogg, both of which are multi-media libraries. I will see if I can remove them from the INDI build, though I am not sure why they would crash a driver. They likely don't need to be there for INDI though.

xsnrg, 2021-10-24:
I rebuilt the INDI RPM from a branch that removed those two libraries. kstars is building now, using this new indilib. Please try the two together and see if there is anything different in operation.

https://copr.fedorainfracloud.org/coprs/xsnrg/kstars-bleeding/build/2905553/

ebessant, 2021-11-11:
So, last week, Fedora rolled out Fedora 35, and I initially installed/upgraded to this newest version on the off chance that the freshly installed bleeding indi libraries ran under 35.  Alas, nope, again.  What I have also done, though is create a script that will clone/cmake/make and install from the Git site.  I am attaching said script, as well as the output from a full indi-lib installation process.  Hopefully, the output from these processes can be compared with similar output from the make processes used in creating the bleeding "rpm" files.

Thankfully, the make install processing of the libraries from the GIT site still produces a working module for the Pegasus UPB V1, and thus a work-around for the non-working "bleeding rpm".

Thanks again,

E.

[build_20211111_011918.log](https://github.com/indilib/indi/files/7517082/build_20211111_011918.log)
[build_indi-core.bash.txt](https://github.com/indilib/indi/files/7517083/build_indi-core.bash.txt)

knro, 2021-11-24:
Okay, this is clearly a downstream issue then. Will close.

closed 2021-11-24 by knro

ebessant, 2021-12-11:
I am not sure what you mean by "download issue" and no longer worth worrying about.  Are you saying it's my fault for not downloading the rpm files properly with "dnf"?  Or is it my fault for using the software on Fedora?  Or are you simply cannot be bothered with trying to discover what is going wrong with the rpm builds for the indi libraries?

I sincerely apologise for doing something so heinous as preferring Fedora to Ubuntu, and for using an older piece of kit that you no doubt can't be bothered to try to support any more.

If it is my fault, then please explain to me what I am doing wrong with the command "dnf upgrade" that would cause the software for a single piece of hardware to break?

Sorry for being so much trouble,

E..

xsnrg, 2021-12-20:
I spent a good bit of time trying to figure this out. I do not have the hardware, and do not use the driver. As is builds locally for you and works, but does not build in the Fedora COPR system, there must be a difference in the compile system between your machine and Redhat's.  I sense you are frustrated with this by the sarcasm in your response, but there really is not much else I can do. All the build logs are available in COPR if you want to try to look through them to see what might be different. I attempted to do this as well, but did not see much to go on.

ebessant, 2021-12-22:
Hello again.  Sorry for the sarcasm, but, yeah, there is some frustration here.

With regard to going through the build logs, how do I get access to COPR?  I have time on my hands now, and can try to help out in any way I can.

E.

xsnrg, 2022-01-06:
All of the logs are available here: https://copr.fedorainfracloud.org/coprs/xsnrg/indi-3rdparty-bleeding/ for 3rd party, and here for indi base: https://copr.fedorainfracloud.org/coprs/xsnrg/libindi-bleeding/

My notifications still are not working so did not see your reply, or maybe it does not send any on a closed issue