indilib/indi #1395: Indi Parking issue / affects iOptron

seventi71, 2021-03-22:
**Describe the bug**
When indi issues a park command to a iOptron mount, the mount parks, but does not always update indi variable to say it is parked. Sometimes it does work, but not possible yet to isolate why it does work sporadically.
To get the mount to park ... tracking needs to be enabled and disabled in quick succession then it parks.
I have tried to modify the driver to do this on parking, but it still hangs so something tells me it is in the Indi server code.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start kstars (stable) start indi (stable) , connect iOptron V3 driver (stable)
2. Unpark and the Park ... it will say parking and just hangs.
3. See error highlighted in log below, where i force the tracking to stop, which then allows it to park.

**Expected behavior**
When the mount has completed it's park routine, the software needs to be set to park.

**Screenshots**
NA

**Desktop (please complete the following information):**
see log

**Log Files**
Pruned log, to show the issue
...
Kstars startup
...
[2021-03-22T12:16:37.816 GMT INFO ][                org.kde.kstars] - Welcome to KStars 3.5.2 Stable
[2021-03-22T12:16:37.816 GMT INFO ][                org.kde.kstars] - Build: 2021-03-01T11:47:36Z
[2021-03-22T12:16:37.816 GMT INFO ][                org.kde.kstars] - OS: "pop"
[2021-03-22T12:16:37.816 GMT INFO ][                org.kde.kstars] - API: "x86_64-little_endian-lp64"
[2021-03-22T12:16:37.816 GMT INFO ][                org.kde.kstars] - Arch: "x86_64"
[2021-03-22T12:16:37.816 GMT INFO ][                org.kde.kstars] - Kernel Type: "linux"
[2021-03-22T12:16:37.816 GMT INFO ][                org.kde.kstars] - Kernel Version: "5.8.0-7642-generic"
[2021-03-22T12:16:37.816 GMT INFO ][                org.kde.kstars] - Qt Version: 5.12.8
[2021-03-22T12:16:37.818 GMT INFO ][                org.kde.kstars] - KStars is started in paused state.
[2021-03-22T12:16:37.896 GMT DEBG ][                org.kde.kstars] - Opened the User DB. Ready.
[2021-03-22T12:16:38.028 GMT INFO ][                org.kde.kstars] - Processing  "unnamedstars.dat" , HTMesh Level 3
[2021-03-22T12:16:38.028 GMT INFO ][                org.kde.kstars] -   Sky Mesh Size:  512
[2021-03-22T12:16:38.036 GMT INFO ][                org.kde.kstars] - Loaded DSO catalog file:  "unnamedstars.dat"
[2021-03-22T12:16:38.156 GMT WARN ][                org.kde.kstars] - "Star HD20,794 not found."
[2021-03-22T12:16:38.162 GMT WARN ][                org.kde.kstars] - "Star HD98,230 not found."
[2021-03-22T12:16:38.167 GMT DEBG ][                       default] - File opened:  "/usr/share/kstars/ngcic.dat"
[2021-03-22T12:16:38.167 GMT INFO ][                org.kde.kstars] - Loading NGC/IC objects
[2021-03-22T12:16:38.366 GMT DEBG ][                       default] - File opened:  "/home/astro/.local/share/kstars/comets.dat"
...
Indi Startup
...
[2021-03-22T12:16:42.821 GMT INFO ][           org.kde.kstars.ekos] - "Starting INDI services..."
[2021-03-22T12:16:42.821 GMT DEBG ][           org.kde.kstars.indi] - INDI: Starting local drivers...
[2021-03-22T12:16:42.821 GMT DEBG ][           org.kde.kstars.indi] - Starting INDI Server:  ("-v", "-p", "7624", "-m", "300", "-r", "0", "-f", "/tmp/indififoa476d06c") -f "/tmp/indififoa476d06c"
[2021-03-22T12:16:42.830 GMT DEBG ][           org.kde.kstars.indi] - INDI Server Started?  true
[2021-03-22T12:16:42.830 GMT DEBG ][           org.kde.kstars.indi] - INDI: INDI Server started locally on port  7624
[2021-03-22T12:16:42.830 GMT DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_ioptronv3_telescope"
[2021-03-22T12:16:42.830 GMT DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_v4l2_ccd"
[2021-03-22T12:16:42.830 GMT DEBG ][           org.kde.kstars.indi] - Adding managed driver "iOptronV3"
[2021-03-22T12:16:42.830 GMT DEBG ][           org.kde.kstars.indi] - Adding managed driver "V4L2 CCD"
[2021-03-22T12:16:42.830 GMT DEBG ][           org.kde.kstars.indi] - INDIListener: Adding a new client manager to INDI listener..
[2021-03-22T12:16:42.830 GMT DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
[2021-03-22T12:16:42.831 GMT DEBG ][           org.kde.kstars.indi] - INDI server connected.
[2021-03-22T12:16:42.831 GMT DEBG ][           org.kde.kstars.indi] - Connection to INDI server is successful
[2021-03-22T12:16:42.831 GMT INFO ][           org.kde.kstars.ekos] - "INDI services started on port 7624. Please connect devices."
[2021-03-22T12:16:42.833 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-22T12:16:42: startup: /usr/bin/indiserver -v -p 7624 -m 300 -r 0 -f /tmp/indififoa476d06c "
[2021-03-22T12:16:42.833 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-22T12:16:42: listening to port 7624 on fd 3"
[2021-03-22T12:16:42.833 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_ioptronv3_telescope -n \"iOptron v3\""
[2021-03-22T12:16:42.833 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "With name: iOptron v3"
[2021-03-22T12:16:42.833 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_ioptronv3_telescope"
[2021-03-22T12:16:42.833 GMT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-03-22T12:16:42: Driver indi_ioptronv3_telescope: pid=6282 rfd=4 wfd=7 efd=8"
... 
Mount Start up
...
[2021-03-22T12:16:45.242 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] RES <210105xxxxxx> "
[... 309 lines cut ...]
[2021-03-22T12:17:50.354 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] CMD <:GLS#> "
[2021-03-22T12:17:50.377 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] RES <-0004170050763000060811> "
[2021-03-22T12:17:50.377 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] CMD <:GEP#> "
[2021-03-22T12:17:50.411 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] RES <+3240000003396156021> "
[2021-03-22T12:17:51.425 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] CMD <:GLS#> "
[2021-03-22T12:17:51.452 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] RES <-0004170050763000060811> "
[2021-03-22T12:17:51.452 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] CMD <:GEP#> "
[2021-03-22T12:17:51.484 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] RES <+3240000003396307521> "
[2021-03-22T12:17:52.487 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] CMD <:GLS#> "
[2021-03-22T12:17:52.522 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] RES <-0004170050763000060811> "
[2021-03-22T12:17:52.522 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] CMD <:GEP#> "
[2021-03-22T12:17:52.557 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] RES <+3240000003396457521> "
[2021-03-22T12:17:53.557 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] CMD <:GLS#> "
[2021-03-22T12:17:53.594 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] RES <-0004170050763000060811> "
[2021-03-22T12:17:53.594 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] CMD <:GEP#> "
[2021-03-22T12:17:53.629 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] RES <+3240000003396607521> "
[2021-03-22T12:17:54.636 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] CMD <:GLS#> "
[2021-03-22T12:17:54.670 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] RES <-0004170050763000060811> "
[2021-03-22T12:17:54.670 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] CMD <:GEP#> "
[2021-03-22T12:17:54.702 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] RES <+3240000003396757521> "
[2021-03-22T12:17:55.707 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] CMD <:GLS#> "
[2021-03-22T12:17:55.724 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] RES <-0004170050763000060811> "
[2021-03-22T12:17:55.724 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] CMD <:GEP#> "
[2021-03-22T12:17:55.756 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[SCOPE] RES <+3240000003396909021> "
[2021-03-22T12:17:56.759 GMT DEBG ][           org.kde.kstars.indi] - iOptron v3 : "[DEBUG] CMD <:GLS#> "

knro, 2021-03-22:
This issue is known and reported to iOptron on Jan 13th, but we didn't get any feedback on it yet. It's a bug in the firmware. Please follow up with iOptron. I sent this email to them:

> This is an excerpt from the log with the firmware versions:
> 
> [2020-12-26T11:13:30.668 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[DEBUG] CMD <:MountInfo#> "
> [2020-12-26T11:13:30.668 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[DEBUG] RES <0122> "
> [2020-12-26T11:13:30.668 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[DEBUG] CMD <:FW1#> "
> [2020-12-26T11:13:30.668 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[DEBUG] RES <191120200110> "
> [2020-12-26T11:13:30.668 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[DEBUG] CMD <:FW2#> "
> [2020-12-26T11:13:30.668 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[DEBUG] RES <190820190820> "
> 
> After parking is started and finished (mount physically stops at parking position), the status report is wrong as you can see below:
> 
> [2020-12-26T11:15:12.017 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[DEBUG] CMD <:GLS#> "
> [2020-12-26T11:15:12.031 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[DEBUG] RES <-3821580045014100220631> "
> [2020-12-26T11:15:12.032 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[SCOPE] CMD <:GEP#> "
> [2020-12-26T11:15:12.062 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[SCOPE] RES <+3240000003112720521> "
> [2020-12-26T11:15:13.062 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[DEBUG] CMD <:GLS#> "
> [2020-12-26T11:15:13.085 MST DEBG ][           org.kde.kstars.indi] - iOptron CEM120 : "[DEBUG] RES <-3821580045014100220631> "
> 
> According to documentation, :GLS# returns system status. Here is the bit from documentation:
> 
> The 18th digit indicate system status: 0 means stopped at non-zero position, 1 means tracking with
> PEC disabled, 2 means slewing, 3 means auto-guiding, 4 means meridian flipping, 5 means tracking
> with PEC enabled (only for non-encoder edition), 6 means parked, 7 means stopped at zero position
> (home position).
> 
> You can see the 18th digit above as "2", whereas it is expected to be "6" (parked). Hence, this is the issue. This issue is confirmed on both CEM70 and CEM120 mounts.

seventi71, 2021-03-22:
Okay, this makes sense, because it does not happen on old firmware.  So the V3 firmware has inherited the bug. 

As a FYI, there are cases where it does park ... i will keep investigating these. That way if we can nail down a definite park, then maybe we can put a workaround in place until they fix the firmware. One known way to force the park is to issue the ST1 then ST0 commands. I think in the firmware, logic it thinks it is still slewing. If you explicitly issue a command to end the slew, then it changes to a 6. (i.e. tell it to track (ends the slew) then tell it to stop tracking, then it parks.) 

I will look at a work around on this to get it bullet proof.

knro, 2021-07-24:
Any response from iOptron about this? This is an upstream issue and not an INDI bug.

closed 2021-07-24 by knro