indilib/indi #1397: Raspberry Pi HQ Camera FITS have green tint. BGGR <> RGGB

callimero, 2021-03-24:
**Describe the bug**

Trying to use Raspicam HQ with Astroberry/KStars/Ekos/Indi. Resulting FITS have a green tint.

FITS also have RGGB in the Header, but I checked in Siril and BGGR fits much better, however still with a green tint.

RAWs done with raspistill and converted  to DNG are perfect.

**To Reproduce**

Take some longish exposures, tint will also be in Fitsviewer.

**Expected behavior**
[image]
[image]

Good FITS :)

**Desktop (please complete the following information):**
 - OS: Raspian/Astroberry
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
Follows

knro, 2021-03-24:
@lboclboc Can you please check?

lboclboc, 2021-03-24:
Ill have a check. RGGB in fits sounds very wrong yes. A green tint is as expected since there are twice as many green pixels than red (or blue). Siril has image filters for that.

callimero, 2021-03-24:
Ok, great. Seems that other software takes that pixel counts into account. No prob.

closed 2021-04-17 by knro