indilib/indi #1403: Allow for physical disconnection of cameras without restarting INDI server

marcocipriani01, 2021-03-29:
Sometimes, either my DSLR or QHY guide camera go crazy. No big deal - the Nikon might get stuck and stops responding (probably a firmware issue) or the QHY might remain busy after a KStars or PHD2 crash. So I have to disconnect one of them from INDI, go outside and physically unplug it, then try re-connect from the INDI control panel. Unfortunately, this last step never works (the driver keeps asking "did you power it up?"). Since I run all my equipment using the INDI Web Manager, I try to use the restart button next to each driver, but I have no luck with this too (same question). I have to stop and restart the entire INDI server, which termintes PHD2 and Ekos and shuts down the mount.

From time to time, I am able to connect after I restart the driver using the INDI Web Manager, but in this situation the affected camera simply refuses to expose (as if it didn't really connect🤔) and KStars crashes. Just to isolate the problem from KStars, I tried this using my [Telescope.Touch](https://github.com/marcocipriani01/Telescope.Touch) app, and it shows that the camera driver disappears after I press "Expose" (but at least the app doesn't crash😁).

It would be great if I could just reconnect from the INDI control panel instead of starting from scratch all the time...😅
Kind regards,
Marco Cipriani

knro, 2021-03-29:
For some of these cameras, sometimes a hard USB reset would work. Otherwise, only a power cycle is sufficient to bring it back into a usable state.

marcocipriani01, 2021-03-29:
Yes, that's what I try every time, the problem is connect the camera back to INDI after a power cycle. I am able to connect again only after restarting the entire INDI server

xsnrg, 2021-03-29:
I have seen this happen when the system cannot re-use the dev/tty that it was previously configured to use. The OS assigns a new one. I don't know if this is your case or not, but restarting INDI would cause it to re-scan if you are set to automatically configure the driver port. Perhaps check /dev/tty for a new device when this happens again, and see if you can go into the INDI control panel and re-scan for it, or set it manually?

marcocipriani01, 2021-03-29:
> I have seen this happen when the system cannot re-use the dev/tty that it was previously configured to use.

That's the case for serial ports. I am able to disconnect the EQMod mount and then connect it again, changing manually the port in the INDI Control Panel (from `/dev/ttyACM0` to `/dev/ttyACM1`). Same for the focuser. But cameras don't use serial ports, so they don't appear as `tty` devices. There's no manual "Port" property for INDI cameras, the driver automatically connects to it, or reports an error.

xsnrg, 2021-03-29:
I have never used a DSLR on my scopes, so did not  know if that was the case there as well. Perhaps it is something similar, just not an exposed tty device.

closed 2021-04-17 by knro