indilib/indi #1412: Escaping initial Text

knro, 2021-04-03:
I just ran across an issue whereas the initial text needs to be escapes (It was from DSLR camera, so dybamic). IUFillText escapes name and label, but not initial text

https://github.com/indilib/indi/blob/bd9110b4a35f6848ad7880d7b74cae7c3a79bbfc/indidriver.c#L504

I thought about replacing it with this:

```
if (initialText && initialText[0])
    {
        char *escapedInitialText = malloc(strlen(initialText)*2);
        escapeXML2(escapedInitialText, initialText, sizeof(escapedInitialText));
        IUSaveText(tp, escapedInitialText);
        free(escapedInitialText);
    }
```

Do you think this is sufficient? is there a better way to do it?

pawel-soja, 2021-04-03:
All escape should be done when converting to XML and from XML. Unfortunately, this has not been done from the beginning and is now problematic. Now you have to make hotfixes that will solve and create new problems (if you want, I can describe it).

I propose to create a function that will do escape only when required.

For short: 
```cpp
escape(escape("Hel&lo")) == escape("Hel&lo") 
```
and 
```cpp
escape("He&amp;lo") == "He&amp;lo"
```
Then put the function in the IUSaveText function.

I'll be making fixes tonight.

pawel-soja, 2021-04-03:
My suspicions have been confirmed.
I quickly renamed the button in the example to `"Control Rain <FAIL>"`.
```cpp
bool RainDetector::initProperties()
{
    // Must init parent properties first!
    INDI::DefaultDevice::initProperties();

    IUFillLight(&RainL[0], "Status", "", IPS_IDLE);
    IUFillLightVector(&RainLP, RainL, 1, getDeviceName(), "Rain Alert", "", MAIN_CONTROL_TAB, IPS_IDLE);

    IUFillSwitch(&RainS[0], "On", "", ISS_OFF);
    IUFillSwitch(&RainS[1], "Off", "", ISS_OFF);
    IUFillSwitchVector(&RainSP, RainS, 2, getDeviceName(), "Control Rain <FAIL>", "", MAIN_CONTROL_TAB, IP_RW, ISR_1OFMANY, 0,
                       IPS_IDLE);

    return true;
}
```
```
2021-04-03T19:54:11: Driver ./tutorial_rain: tutorial_rain dispatch error: Property Control Rain <FAIL> is not defined in Rain Detector.
```
The structure holds the text `"Control Rain &lt;FAIL&gt;"`.

I can move the use of the escape function where XML is built. We'll see what happens.

pawel-soja, 2021-04-04:
KStars has double unescap?
[image]
```cpp
void escapeXML_fputs(const char *src, FILE *stream);
void escapeXML_vprintf(const char *fmt, va_list ap);
void escapeXML_vfprintf(FILE *stream, const char *fmt, va_list ap);

#define prints(STR)  fputs(STR, stdout)
#define printsx(STR) escapeXML_fputs(STR, stdout)
#define printfx(...) escapeXML_vfprintf(stdout, __VA_ARGS__)
```

What's going on here?!

knro, 2021-04-04:
I need to double check but perhaps QLabel is interpreting the text as such? Maybe it's a Qt thing.

pawel-soja, 2021-04-04:
This is not a QT problem. I improved the INDI library and recompiled KStars. It looks okay now.

closed 2021-04-12 by knro