indilib/indi #1413: Pegasus DMFC no longer connects

DerPit, 2021-04-04:
I've upgraded my INDI installation to the latest git version.  I use an Pegasus DMFC.  Since this upgrade, I cannot connect to it anymore.  All I get is "Error retrieving data from Pegasus DMFC Device, please ensure device is powered and the port is correct. "  Yes, it is.

I am running INDI on a separate computer at the mount, and connect to it via EKOS from my laptop, indiserver is started manually with the needed drivers.  I noticed that it is now regarded as a new device.  Before it was called 'Pegasus DMFC', now it is 'Pegasus DMFC Device'.  

The weird thing is that when I connect the DMFC directly to the laptop and run it in local mode (i.e., EKOS starts indiserver and the drivers) it will still have the old name, and works:  I started a session with only the DMFC focuse and the simulator cam+telescope.  It works.  Then I only change the setup to use 'remote' on 'localhost' and start indiserver manually.  Again it fails to connect.  Switching now back to the 'local' setting in EKOS, it will also not work anymore.  I have to power cycle the DMFC, then the local mode works again.  As if something has messed up the device.

I ran it with debug enabled:
`[2021-04-04T16:19:31.231 WEST INFO ][           org.kde.kstars.indi] - Pegasus DMFC Device :  "[INFO] Debug is enabled. "
[2021-04-04T16:19:31.231 WEST DEBG ][           org.kde.kstars.indi] - < Pegasus DMFC Device >: < DEBUG_LEVEL >
[2021-04-04T16:19:31.232 WEST DEBG ][           org.kde.kstars.indi] - < Pegasus DMFC Device >: < LOGGING_LEVEL >
[2021-04-04T16:19:31.232 WEST DEBG ][           org.kde.kstars.indi] - < Pegasus DMFC Device >: < LOG_OUTPUT >
[2021-04-04T16:19:31.276 WEST DEBG ][           org.kde.kstars.indi] - Pegasus DMFC Device : "[DEBUG] Toggle Debug Level -- Driver Debug "
[2021-04-04T16:20:04.110 WEST DEBG ][           org.kde.kstars.indi] - < Pegasus DMFC Device >: < SYSTEM_PORTS >
[2021-04-04T16:20:13.086 WEST DEBG ][           org.kde.kstars.indi] - Pegasus DMFC Device : "[DEBUG] Auto search is disabled because /dev/focus is not a system port. "
[2021-04-04T16:20:37.264 WEST INFO ][           org.kde.kstars.indi] - Pegasus DMFC Device :  "[INFO] Saving device configuration... "
[2021-04-04T16:20:37.265 WEST INFO ][           org.kde.kstars.indi] - Pegasus DMFC Device :  "[INFO] Device configuration saved. "
[2021-04-04T16:20:37.265 WEST DEBG ][           org.kde.kstars.indi] - Pegasus DMFC Device : "[DEBUG] Configuration successfully saved. "
[2021-04-04T16:20:46.597 WEST DEBG ][           org.kde.kstars.indi] - Pegasus DMFC Device : "[DEBUG] Connecting to /dev/focus @ 19200 "
[2021-04-04T16:20:46.715 WEST DEBG ][           org.kde.kstars.indi] - Pegasus DMFC Device : "[DEBUG] Port FD 3 "
[2021-04-04T16:20:46.715 WEST DEBG ][           org.kde.kstars.indi] - Pegasus DMFC Device : "[DEBUG] Connection successful, attempting handshake... "
[2021-04-04T16:20:46.715 WEST DEBG ][           org.kde.kstars.indi] - Pegasus DMFC Device : "[DEBUG] CMD <0X23> "
[2021-04-04T16:20:46.727 WEST DEBG ][           org.kde.kstars.indi] - Pegasus DMFC Device : "[DEBUG] RES <OK_DMFCN> "
[2021-04-04T16:20:46.755 WEST INFO ][           org.kde.kstars.indi] - Pegasus DMFC Device :  "[INFO] Error retrieving data from Pegasus DMFC Device, please ensure device is powered and the port is correct. "
[2021-04-04T16:20:46.755 WEST DEBG ][           org.kde.kstars.indi] - Pegasus DMFC Device : "[DEBUG] Handshake failed. "
[2021-04-04T16:20:46.756 WEST DEBG ][           org.kde.kstars.ekos] - "Pegasus DMFC Device"  is disconnected.
`
So it did get an initial answer (if I interpret the OK_DMFCN correct), but somehow isn't happy with it?
Attached is the logfile with a few attempts in both local and remote mode

[log_12-57-22.txt.gz](https://github.com/indilib/indi/files/6254866/log_12-57-22.txt.gz)

chrysikos, 2021-04-04:
@DerPit It's possible to tell me the full command you execute in order to start the Indiserver manually.

DerPit, 2021-04-04:
Sure.  I just run "indiserver indi_simulator_telescope indi_simulator_ccd indi_dmfc_focus"

chrysikos, 2021-04-04:
Please try to run the indiserver in FIFO mode and let me know:

mkfifo /tmp/myFIFO
indiserver -f /tmp/myFIFO
start indi_simulator_telescope
start indi_simulator_ccd
start indi_dmfc_focus -n "Pegasus DMFC"

DerPit, 2021-04-04:
Yes, this is basically what EKOS does when starting the indiserver locally by itself.  It works (though the first try had failed, I had to re-plug USB).

But the direct mode is one official way to start drivers, and so far (with the old driver) that worked without problems...

chrysikos, 2021-04-04:
@DerPit I feel sorry for this trouble, In the next version I'll fix and the "Pegasus DFMC" will be the default label in order to start indiserver in direct mode. @knro Can we close this? or will keep it open until the fix?

knro, 2021-04-04:
@chrysikos you can close this in the commit itself. Just say "fixes #1413" some where in the commit log and it will automatically close it.

DerPit, 2021-04-04:
No worries - it's devel version, and that's why we test it - is it :)
I don't mind if the name changes, but I wonder if that alone would fix the issue?  After all, it connects (handshake), gets the initial OK response but then fails?  Is it expecting some other device?
Thanks for looking at it!

chrysikos, 2021-04-04:
@DerPit  Yes exactly, the indi_dmfc_focus driver corresponds to three different devices Pegasus DMFC, Pegasus FocusCube and Pegasus ScopsOAG.

DerPit, 2021-04-04:
Ah, and the '-n "Pegasus DMFC" tells it what it is?  Can't it detect which device actually is connected? Maybe an option would be to call the binary indi_dmfc_focus_device, but if you call it (via a link) as indi_dmfc_cube or indi_dmfc_scops it expects the respective one?  Else people with the other devices might get issues?  
(Sorry, just guessing around there....)

chrysikos, 2021-04-04:
Yes the label "-n" determines the driver's behavior. I do not know if binary link can solve this, @knro did you have any idea?.

chrysikos, 2021-04-04:
@DerPit  By the way it's preferable to start the indiserver in FIFO mode. In case someone had two FocusCube one for its main telescope and the other for the guiding. It is one-way the FIFO mode. I prefer the Indi-Web manager https://github.com/knro/indiwebmanager to handle the start up process.

DerPit, 2021-04-04:
@chrysikos: Yes, the switch-behavior-on-executable-name is a very common trick/method to affect how a program behaves.  As for FIFO vs. direct start:  That's a different question of course.  But I think you would agree that - as long as both ways are officially supported - an update shouldn't break this and work with only one method.
Cheers,
  Pit

DerPit, 2021-04-05:
> @DerPit By the way it's preferable to start the indiserver in FIFO mode. 
@chrysikos I'm experimenting with indi-web, but it seems it doesn't solve the problem:  I start indi-web, then activate 'INDI Web Manager' radio box.  This will however not add the '-n "Pegasus DMFC"' option to the start command, and it will also come up as "Pegasus DMFC Device", and fail to connect, as before.
Is there specific code to add the '-n' option in EKOS when in local mode?  Seems that is then also needed for the web manager case?
Should I open a separate issue for that?

PR #1415 (2021-04-06): separate focuscube and scopsoag behavior to new drivers
fixes #1413
closed 2021-04-06 by knro