indilib/indi #1416: indi_webcam_cdd crashes in KStars under macos

offermannu, 2021-04-07:
Not sure whether this is the right place for this bug report. I also opened a ticket at the KStars issue tracker: https://bugs.kde.org/show_bug.cgi?id=435390

**Describe the bug**

INDI server crahes in KStars for INDI Webcam driver. KStars shows "INDI Driver INDI Webcam crashed. Restart it?"

**To Reproduce**
Exact steps to reproduce the behavior.
In KStars:
1. Use Internal INDI Server
1. Create new Ekos Profile with CCD: INDI WebCam (Auto Connect off) as only device
2. Start Ekso

**Expected behavior**
Running Ekos
[image]

**Screenshots**
see attachment

**Desktop (please complete the following information):**
 - OS: macOS  and 
 - Version 10.15.7 (19H524) 11.2.3 (20D91)
 - KStars version: 3.5.2

**Log Files**
macOS Log console app reveals:

Process: indi_webcam_ccd:
This app has crashed because it attempted to access privacy-sensitive data without a usage description.  The app's Info.plist must contain an NSCameraUsageDescription key with a string value explaining to the user how the app uses this data.
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html
[log_16-47-55.txt.zip](https://github.com/indilib/indi/files/6272692/log_16-47-55.txt.zip)

offermannu, 2021-04-07:
I tried to launch indi server directly in the terminal which seems to work fine:

> /Applications/KStars.app/Contents/MacOS/indi
> ==> ./indiserver -v -p 7624 -m 300 -r 0 -f /tmp/indififo31c28859
> 2021-04-07T15:25:48: startup: ./indiserver -v -p 7624 -m 300 -r 0 -f /tmp/indififo31c28859 
> 2021-04-07T15:25:48: listening to port 7624 on fd 3

and then:

> ==> nc localhost 7624
> 2021-04-07T15:28:15: Client 4: new arrival from 127.0.0.1:50307 - welcome!

knro, 2021-04-07:
@rlancaste Any idea is this is limited to MacOS? it works fine under Linux as far as I can tell.

rlancaste, 2021-04-07:
I just tried this and yes, @offermannu is correct, it does refuse to launch INDI Webcam.  This appears to be a new change in OS X.  But @offermannu also provided the solution.  Nice Work!  I tried just adding the following to KStars's info.plist:

<key>NSCameraUsageDescription</key>
	<string>To take Astronomy Photos of Galaxies and Nebulae</string>

(This info.plist is right inside the kstars.app bundle, I just edited it with a text editor.)

And it worked!  I don't know if the string that I put there is exactly what we should say, but I can definitely add something like that to the next release.  It just frustrates me when random changes like this just get made and break stuff that we made without warning.

offermannu, 2021-04-07:
@rlancaste the developer's pain ;-)

I'm not a macos developer (I'm a Java developer) and don't know much about Info.plist and the like but encouraged by  @rlancaste reply I tried the same - voila :-)

[image]

xsnrg, 2021-04-07:
>  It just frustrates me when random changes like this just get made and break stuff that we made without warning.

Can a test be added to the code to check for such a condition?

openastroproject, 2021-04-07:
> It just frustrates me when random changes like this just get made and break stuff that we made without warning.

It's the Apple way :)

offermannu, 2021-04-07:
@rlancaste now I'm trying to connect to DroidCam but with no success.
I'm following your guide at https://www.indilib.org/ccds/indi-webcam.html and I am able to connect to the DroidCam in the browser while indi_webcam_ccd is complaining:

2021-04-07T17:02:53: [INFO] Failed to open source. Check your settings: Invalid data found when processing input 
2021-04-07T17:02:52: [INFO] Attempting to connect 
2021-04-07T17:02:13: [INFO] Session log file 2021-04-07T17:02:52: [INFO] Trying to connect to IP Camera at: http://udo:odu@192.168.7.29:4747 

Maybe this should be placed into a new ticket?

rlancaste, 2021-04-07:
Jim, I am not sure how you could add a test to test for this particular condition.  If you started INDI Webcam from the command line, it would have worked fine.  In this case, in order to show the problem, it required starting the driver from an app launched with a graphical user interface.  It wasn't an issue with the driver or with INDI, it was an issue with the App bundle in KStars.

xsnrg, 2021-04-07:
I see. Yes, that would be difficult. More of a regression test suite thing, and even then hard to do without simulating all the hardware.

rlancaste, 2021-04-07:
From a testing perspective, I would really like to see the CircleCI/Travis tests that used to be in place for testing the INDI build on OS X to be restored somehow.  About a year ago, the Linux builds and the OS X builds were all tested, but then the OS X one was removed.  Ever since, every time I go to build a DMG, I have to submit a couple of pull requests to fix little bugs like forgotten includes etc. to make it work on OS X again.

rlancaste, 2021-04-07:
Of course, that wouldn't have helped in this case, but it would really help in many others

offermannu, 2021-04-07:
Would it possible to integrate the logs into macos Console app? 
In case of errors you could ask people to open the Console app and search for "indi" or "kstars" and so you would get KStars' and macos logs...

knro, 2021-04-08:
> From a testing perspective, I would really like to see the CircleCI/Travis tests that used to be in place for testing the INDI build on OS X to be restored somehow. About a year ago, the Linux [...]

IIRC, @pawel-soja added MacOS testing. Can you please confirm?

closed 2021-07-24 by knro

elnappo, 2021-08-31:
INDI webcam driver still crashes on my machine but with a different error:

```
Driver indi_webcam_ccd: dyld: Library not loaded: @rpath/libx265.192.dylib"
Driver indi_webcam_ccd:   Referenced from: /Applications/kstars.app/Contents/Frameworks/libavcodec.58.dylib"
Driver indi_webcam_ccd:   Reason: no suitable image found.  Did find:"
Driver indi_webcam_ccd: \t/Applications/kstars.app/Contents/Frameworks/libx265.192.dylib: mach-o, but wrong filetype"
Driver indi_webcam_ccd: \t/Applications/kstars.app/Contents/MacOS/indi/../../Frameworks/libx265.192.dylib: mach-o, but wrong filetype"
Driver indi_webcam_ccd: stderr EOF"
```

```
INFO ][                org.kde.kstars] - Welcome to KStars 3.5.4 Stable
INFO ][                org.kde.kstars] - Build: 2021-07-08T01:54:02Z
INFO ][                org.kde.kstars] - OS: "osx"
INFO ][                org.kde.kstars] - API: "x86_64-little_endian-lp64"
INFO ][                org.kde.kstars] - Arch: "x86_64"
INFO ][                org.kde.kstars] - Kernel Type: "darwin"
INFO ][                org.kde.kstars] - Kernel Version: "20.6.0"
INFO ][                org.kde.kstars] - Qt Version: 5.15.1
[...]
INFO ][           org.kde.kstars.ekos] - "Starting INDI services..."
DEBG ][           org.kde.kstars.indi] - INDI: Starting local drivers...
DEBG ][           org.kde.kstars.indi] - Starting INDI Server:  ("-v", "-p", "7624", "-m", "1024", "-r", "0", "-f", "/tmp/indififobbf8a7ef") -f "/tmp/indififobbf8a7ef"
DEBG ][           org.kde.kstars.indi] - INDI Server Started?  true
DEBG ][           org.kde.kstars.indi] - INDI: INDI Server started locally on port  7624
DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_webcam_ccd"
DEBG ][           org.kde.kstars.indi] - Adding managed driver "INDI Webcam"
DEBG ][           org.kde.kstars.indi] - INDIListener: Adding a new client manager to INDI listener..
DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: startup: /Applications/kstars.app/Contents/MacOS/indi/indiserver -v -p 7624 -m 1024 -r 0 -f /tmp/indififobbf8a7ef "
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: listening to port 7624 on fd 3"
DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_webcam_ccd -n \"INDI Webcam\""
DEBG ][           org.kde.kstars.indi] - INDI Server:  "With name: INDI Webcam"
DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_webcam_ccd"
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: Driver indi_webcam_ccd: pid=63541 rfd=4 wfd=7 efd=8"
DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
DEBG ][           org.kde.kstars.indi] - Connection to INDI server is successful
INFO ][           org.kde.kstars.ekos] - "INDI services started on port 7624. Please connect devices."
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: Driver indi_webcam_ccd: dyld: Library not loaded: @rpath/libx265.192.dylib"
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: Driver indi_webcam_ccd:   Referenced from: /Applications/kstars.app/Contents/Frameworks/libavcodec.58.dylib"
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: Driver indi_webcam_ccd:   Reason: no suitable image found.  Did find:"
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: Driver indi_webcam_ccd: \t/Applications/kstars.app/Contents/Frameworks/libx265.192.dylib: mach-o, but wrong filetype"
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: Driver indi_webcam_ccd: \t/Applications/kstars.app/Contents/MacOS/indi/../../Frameworks/libx265.192.dylib: mach-o, but wrong filetype"
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: Driver indi_webcam_ccd: stderr EOF"
DEBG ][           org.kde.kstars.indi] - INDI Server:  "Child process 63541 died"
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: Driver indi_webcam_ccd: Terminated after #0 restarts."
DEBG ][           org.kde.kstars.indi] - INDI Server:  "2021-08-31T18:04:27: Client 4: new arrival from 127.0.0.1:56937 - welcome!"
DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
CRIT ][           org.kde.kstars.indi] - INDI driver  "indi_webcam_ccd"  crashed!
DEBG ][           org.kde.kstars.indi] - INDI: Stopping local drivers...
DEBG ][           org.kde.kstars.indi] - Removing managed driver "INDI Webcam"
DEBG ][           org.kde.kstars.indi] - INDIListener: Removing client manager for server localhost @ 7624
DEBG ][           org.kde.kstars.indi] - Stopping INDI Driver  "indi_webcam_ccd"
DEBG ][           org.kde.kstars.indi] - INDI server disconnected. Exit code: 0
DEBG ][           org.kde.kstars.indi] - Stopping INDI Server  "localhost" @ "7624"
DEBG ][           org.kde.kstars.ekos] - Resetting Ekos Manager...
WARN ][                       default] - QSqlDatabasePrivate::removeDatabase: connection 'filter_db' is still in use, all queries will cease to work.
WARN ][                       default] - QSqlDatabasePrivate::addDatabase: duplicate connection name 'filter_db', old connection removed.
INFO ][           org.kde.kstars.ekos] - "INDI services stopped."
```

shayneoneill, 2021-12-16:
Yeah I'm getting the no Dylib one. Macos Montery on an M1 mac.  Maybe its an arch issue?