indilib/indi #1417: INDI WebCam driver fails to connect to DroidCam

offermannu, 2021-04-08:
**Describe the bug**
I want to use the INDI WebCam Driver in KStars to connect to my Android smartphone.
I follwoed the guide at https://www.indilib.org/ccds/indi-webcam.html but the INDI WebCam driver is not able to connect to DroidCam

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start DroidCam and provide userid+pw
2. Start INDI WebCam Driver
3. Configure IP Camera, IP addresse, port and userid+pw
4. Connect -> fails

**Expected behavior**
An established connection.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: macos
 - Version 10.15.7 (19H524)
 - KStars version: 3.5.2

**Log Files**

I activated all log & debug levels but there is not really much in the logs:

> DEBUG	307.631123 sec	: Toggle Debug Level -- Alignment Subsystem
> INFO	314.143753 sec	: Trying to connect to IP Camera at: http://udo:odu@192.168.7.29:4747
> INFO	314.143793 sec	: Attempting to connect
> INFO	314.363190 sec	: Failed to open source. Check your settings: Invalid data found when processing input
> INFO	350.425331 sec	: Trying to connect to IP Camera at: http://udo:odu@192.168.7.29:4747
> INFO	350.425388 sec	: Attempting to connect
> INFO	350.588150 sec	: Failed to open source. Check your settings: Invalid data found when processing input
> INFO	526.952674 sec	: Trying to connect to IP Camera at: http://udo:odu@192.168.7.29:4747
> INFO	526.952731 sec	: Attempting to connect
> INFO	527.136602 sec	: Failed to open source. Check your settings: Invalid data found when processing input
> INFO	537.955563 sec	: Trying to connect to IP Camera at: http://udo:odu@192.168.7.29:4747
> INFO	537.955635 sec	: Attempting to connect
> INFO	538.122637 sec	: Failed to open source. Check your settings: Invalid data found when processing input
> 

Note that I can open the URL in the browser without any problems.

Maybe it's a problem with DroidCam - in this case could you please tell me which Android IP camera should work with the INDI WebCam driver? Or is there another way to integrate my smartphone camera into KStars/Ekos?

knro, 2021-09-28:
@rlancaste Any insights on this issue?

knro, 2021-11-24:
This should be filed in INDI 3rdparty Github.

closed 2021-11-24 by knro