indilib/indi #1419: Deprecation of guide command in latest ioptronv3

simont77, 2021-04-11:
This is a reminder for future.

According to latest ioptron_v3 specs, the present command “:MnXXXXX#”, “:MeXXXXX#”, “:MsXXXXX#” or “:MwXXXXX#” for guiding is deprecated and substituted by the new command  “:ZSXXXXX#”, “:ZQXXXXX#”, “:ZEXXXXX#” or “:ZCXXXXX#”

I don't know if the new command is already recognized by any mount.

knro, 2021-04-11:
Thanks for the reminder, that's good to know. Any idea which firmwares/mounts are affected by this change?

simont77, 2021-04-11:
It is introduced in version 3.10 (January 4th, 2021) of the command set, so I suppose that affects GEM45/CEM40 updated to the new firmware requiring v3 command set and to GEM28/CEM26. I don't know if also CEM70 and CEM120 were updated. In any case, it's a deprecation warning, so for the time being both commands should still work, the old one will be removed in a future release of the command set, I suppose.

closed 2021-11-10 by knro