indilib/indi #1420: Possible fix for incorrect slew rate in ioptronv3 driver

simont77, 2021-04-11:
It has been reported on indi forum that slew rate is improperly set in ioptronv3 driver.

Here https://github.com/indilib/indi/blob/605180db41dfa3c1974c37ac5bb33a690e0ef6b4/drivers/telescope/ioptronv3.cpp#L1083 there seems to be an extra "+1", since the setSlewRate function in the rolling line already increases the index when preparing the string to be sent to the mount (which should include a number from 1 to 9). The extra "+1" is indeed not present in ieq driver, and the command syntax did not change in v3 protocol.

knro, 2021-04-12:
The index is from 0 to 8

simont77, 2021-04-12:
Yes, but there is a +1 here and another +1 inside setSlewRate function, so at the end we send to the mount a range from 2 to 10

knro, 2021-04-12:
oh right you're absolutely right, will fix this.

commit 339ec271 (2021-04-12): Fix slew rate index. Fixes #1420
closed 2021-04-12 by knro