indilib/indi #1422: possible race conditions from userio

knro, 2021-04-12:
Config files are getting corrupted since <message> messages are getting written the the config file as it gets saved:

e.g.

```
<INDIDriver>
    <message device="NightCrawler" timestamp="2021-04-12T08:43:21" message="[INFO] Saving device configuration..."/>
    <newSwitchVector device="NightCrawler" name="DEBUG">
        <oneSwitch name="ENABLE">
Off
        </oneSwitch>
        <oneSwitch name="DISABLE">
On
        </oneSwitch>
    </newSwitchVector>

```

It appears there needs to be a mutex to protect against contamination of different streams? @pawel-soja can you please check? This is a critical issue since it renders config files unusable now and corrupted.

pawel-soja, 2021-04-12:
The appearance of userio should not require mutex.
The "message" element should not be in config? Is there no closures "</INDIDriver>"?

knro, 2021-04-12:
No message is not part of the config file, this is intended for clients like all other messages. The config has one root node element <INDIDriver>...</INDIDriver>

knro, 2021-04-12:
Oh I see IUUserIOConfigTag now. How can we redirect  IDUserIOMessage from file io to network io?

pawel-soja, 2021-04-12:
Yes, my little mistake, give me a minute.

knro, 2021-04-12:
Thanks, it works great now!

closed 2021-04-12 by knro