indilib/indi #1424: Baseclient SIGPIPE crash when disconnected abnormally

knro, 2021-04-12:
This is a rather old issue, but it still happens in GIT. When an INDI client server connection is terminated due to server termination, it appears the sockfd is not properly closed/recycled. On next connection, the whole app crashes in SIGPIPE, this is repeatable. Attached in the backtrace.

@pawel-soja I'm going to start investigating this more, but thought you might already have some idea on this?

[sigpipe.txt](https://github.com/indilib/indi/files/6296256/sigpipe.txt)

pawel-soja, 2021-04-12:
It is easy to solve. I'm on vacation now. I'll have a look at it over the weekend.

TallFurryMan, 2021-04-12:
I think I had a look to this at some point. It was related to the management of signals through the socket. Basically, let the program manage signals instead of propagating them as real signals. And there was a difference between *nix and Win32 apis there.

pawel-soja, 2021-04-15:
I see we need to take care of the *nix / Windows and Qt versions. I can separate sockets into a separate class that will be portable. But a small version of the Qt library is made.
It would be a lot easier to switch to Qt completely. - all project (INDI / INDI-3rdParty).
Both ways are OK for me, as I create a similar project to Qt, using a newer approach to slot signals (vtable manipulation). Libraries will be opened in the future. The library is modular and lightweight.

What do you think?

knro, 2021-04-16:
So I thought about this long ago, and adding Qt dependency is an overkill for INDI. Especially many clients do not use Qt and do not want to use Qt in their systems.

knro, 2021-04-17:
At any rate, any way to dealing with this now so it handles abnormal disconnects more gracefully? It's important for v1.9.0 INDI release.

pawel-soja, 2021-04-17:
@knro When do you want to release this version?

When analyzing the code, I can see more socket problems.

I can throw in some hotfixes to fix some problems or upload a completely new implementation consisting of:
- EventDispatcher - you can dynamically add sockfd / timers to listen for events
- EventLoop - starts a loop to handle EventDispatcher
- Thread - thread support - you can use multiple EventDispatchers.
- TcpSocket - as the name suggests.

The network connection operation is as follows:
```cpp
int main(int, char*[])
{
    EventLoop e;

    TcpSocket socket;

    socket.callOnReadyRead([&](){
        char data[1024];
        size_t size = socket.read(data, sizeof(data));

        cout << "Got data " << size << " Bytes" << endl;
        cout << data << endl;
    });

    socket.callOnConnected([&](){
        cout << "Connected" << endl;
        const char *msg =
                "GET / HTTP/1.1\r\n"
                "Host: google.com\r\n"
                "Connection: close\r\n"
                "\r\n";
        socket.write(msg, strlen(msg));
    });

    socket.callOnDisconnected([](){
        cout << "Disconnected" << endl;
    });

    socket.connectToHost("google.com", 80);

    return e.exec();
}
```
To add the libraries, I still lack tests on several platforms.

Ultimately, there may also be classes such as:
```
AbstractServer    DataStream                ForEach             Iterator             RefCount           Timer
AbstractSocket    DateTime                  FreeList            LinearFunction       Reflection         TimerEvent
AccessArray       Debug                     Function            LinearInterpolation  Semaphore          TimerInfo
ArrayData         Destroyer                 FunctionThread      List                 SharedPointer      UrlParser
Atomic            Dir                       HexDump             MakeReference        SiUnits            Vector
Buffer            DirIterator               HostAddress         Map                  SocketNotifier     WaitCondition
ByteArray         Dsp                       Http                Math                 SslServer          WebSocket
ByteArrayMatcher  DynamicTarball            HttpHeader          Memory               SslSocket          WebSocketServer
Cast              Event                     HttpRequestHeader   MimeDatabase         String             WifiConfigFile
Class             EventDispatcher           HttpResponseHeader  MimeType             SystemInformation  WifiNetwork
Color             EventLoop                 HttpServer          Mutex                TcpServer          WifiStation
Combine           FifoBuffer                IndexSequence       MutexLocker          TcpSocket          ZipIterator
ContainerHelper   File                      Ini                 NetworkInterface     TextStream
CoreApplication   FileDescriptorFileSystem  Integers            Object               Thread
CoreLib           FileInfo                  IODevice            ObserveDataPointer   ThreadData
```

pawel-soja, 2021-04-18:
if you want to release today, wait for my PR today, I have critical fixes.

knro, 2021-04-18:
Thanks! I've already pushed a small fix for this issue, but if you have more, than that's great!!

closed 2021-06-25 by knro