indilib/indi #1427: Random crashes involving IText

knro, 2021-04-15:
Been getting some random crashes recently that appear to be related to IText changes. The client receives an invalid IText. In the INDI server thread, it stops here at (*myfree) while in the client thread, a newINDIText is causing the crash as it sends an invalid point for ITextVectorProperty:

```
static void freeString(String *sp)
{
    if (sp->s)
        (*myfree)(sp->s);
    sp->s  = NULL;
    sp->sl = 0;
    sp->sm = 0;
}
```

Could this be related to recent changes with escaping XML? or with perhaps the IUSaveText? This appears to happen more often when a BLOB is getting downloaded from the camera as well. I will see if I can have a pattern for this, but something is definitely amiss in string handling.

knro, 2021-04-15:
Another thing I just noticed is that this happens in line 549 of baseclient.cpp

delXMLEle(root)

What's interesting is that the buffer it is processing is invalid, it's missing opening tag for XML and just appears to be somewhere in the middle. I think this causes to the parses to go boom.

pawel-soja, 2021-04-15:
@knro can you provide more details? What does corrupt XML look like?
I cannot reproduce the problem. I tested lilxml by analyzing memory allocation and freeing. Additionally, I tested parseXMLChunk by inserting one character each.

I only found a bug in function 'IUUserIONewBLOBStart' which should not be related to this issue.
Commit https://github.com/indilib/indi/commit/32025368ae8844e9fd33e84440cf6d7469d5a61a

Previously:
```cpp
void IUUserIONewBLOBStart(
    const userio *io, void *user,
    const char *dev, const char *name, const char *timestamp
)
{
    userio_prints    (io, user, "<newBLOBVector\n"
                                "  device='");
    userio_xml_escape(io, user, dev);
    userio_prints    (io, user, "'\n"
                                "  name='");
    userio_xml_escape(io, user, name);
    userio_prints    (io, user, "'\n");  // <-- double apos
    if (timestamp != NULL)
    {
        userio_prints    (io, user, "  timestamp='");
        userio_xml_escape(io, user, timestamp);
        // missing apos
    }
    userio_prints    (io, user, "'>\n"); // <-- double apos if timestamp == NULL
}
```

knro, 2021-04-16:
Actually, nevermind. I found the issue was caused by something else, completely unrelated to this. One driver was causing a 100% CPU utilization, so that's fixed now.

closed 2021-04-16 by knro