indilib/indi #1431: iOptron iPolar captures using the indi_v4l2_ccd driver show major vertical banding

jbradberry, 2021-04-20:
Attempting to make use of the iOptron iPolar polar alignment camera with KStars using the indi_v4l2_ccd driver results in unusable images that show heavy vertical streaking.  Plate solving consistently fails regardless of the camera settings used.

My particular hardware is known to work with the iOptron proprietary software, and also appears to have reasonable results using the Gnome desktop's webcam capture program (which presumably uses the same underlying v4l2 device as the indi driver).

**To Reproduce**
1. Configure the iPolar device in KStars Ekos, and connect
2. In the CCD tab, set an exposure time <= 0.5 seconds, then either capture a preview or queue and start an exposure
3. In the FITS viewer, we now see an image that appears similar to a barcode

**Expected behavior**
A capture showing a flat dark field, if taken with the lens cap on (as the ones below were), with only a manageable amount of noise and bias.  Otherwise with stars showing in a manner that can successfully be plate solved.

**Screenshots**

FITS file captured in KStars (exported as PNG from the Gimp in order to be attachable here):
[image]

Capture using Gnome Cheese using the same hardware:
[image]

**Desktop (please complete the following information):**
 - OS: Debian Linux Testing/Unstable
 - Version: KStars 3.4.3, libindi 1.8.8

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2021-04-20:
Maybe it's stretching issue? If you turn off auto-stretch? Can you attach a link to the FITS image?

jbradberry, 2021-04-20:
Ok, here's the FITS image:

[Light_001.fits.txt](https://github.com/indilib/indi/files/6343812/Light_001.fits.txt)

I've added a .txt extension so that Github doesn't complain.

knro, 2021-04-20:
Thanks, so it's not a stretching issue. The image indeed looks like that. Have you tried using INDI Webcam driver as well? or tried to play with the V4L2 settings? there are plenty.

Unfortunately, I don't have access to this device to test locally.

jbradberry, 2021-04-20:
@knro fair enough.

I've played around with the `brightness`, `contrast`, `gain`, and `gamma` settings, though only `gain` seems to have any visible effect (which kind of surprised me).  I've also toggled settings having to do with white balance and exposure, to no noticeable effect.  I'll go ahead and try configuring it with the webcam driver after work today.

jbradberry, 2021-04-20:
@knro I see a possibility.  Here are the results of running `$ v4l2-ctl -l` at various stages of the process:

iPolar first plugged in to the computer, KStars not yet running:

```
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=25 value=25
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=48 value=48
                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=0 value=0
                            hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=72 max=300 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=64 max=480 step=32 default=64 value=64
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=2 value=2
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=32 value=32
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0
```

after initial Ekos connection:

```
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=25 value=25
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=48 value=48
                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=0 value=0
                            hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=72 max=300 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=64 max=480 step=32 default=64 value=64
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=2 value=2
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=32 value=5000
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0
```

after setting 1280x960 and manually setting `exposure_absolute=32`, then taking a preview exposure:
```
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=25 value=25
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=48 value=48
                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=0 value=0
                            hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=72 max=300 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=64 max=480 step=32 default=64 value=64
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=2 value=2
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=32 value=5000
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0
```

after setting only `exposure_absolute=32`, no other settings, then preview:
```
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=25 value=25
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=48 value=48
                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=0 value=0
                            hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=72 max=300 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=64 max=480 step=32 default=64 value=64
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=2 value=2
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=32 value=5000
[... cut ...]
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=2 value=2
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=32 value=32
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0
```

at every step once KStars is involved and I try to take an exposure, the `absolute_exposure` gets set to maximum, even if I try to explicitly set it otherwise.

jbradberry, 2021-04-21:
Looking at the UVC 1.5 specs, I see that this is as expected.  0.5 seconds -> `absolute_exposure=5000`.

I'll try messing around with the webcam driver instead a bit later.

jbradberry, 2021-04-21:
I don't seem to have the option for a webcam driver.

[image]

knro, 2021-11-24:
I actually had access to iPolar remotely and it worked quite well with PAA all the way, so this issue is either stretching issue or perhaps other settings within the driver. But I can confirm it works and captures well.

closed 2021-11-24 by knro