indilib/indi #1433: New repos for openSUSE

paolostivanin, 2021-04-25:
Hello all,
sorry if this is not the appropriate place, but I couldn't think of a better alternative :)
I wanted to inform you that both `indi` and `indi-3rdparty` are now (almost) fully packaged for openSUSE 15.2, 15.3 and Tumbleweed for both x64 and aarch64 architectures.
I wrote "almost" because:

- some indi-3rdparty drivers are still missing (slowly working on packaging them all. The most used are there though)
- ATIK couldn't be built on 15.2 due to the glibc 2.27 requirement (and 15.2 is using 2.26). This is not a big issue since 15.3 will be released in June as the new stable release.
- indi-pentax couldn't be built on aarch64 due to missing upstream support

How to install the repo:

For 15.2:
```
zypper ar -f https://download.opensuse.org/repositories/home:/polslinux:/astrophotography/openSUSE_Leap_15.2/ Astrophotography_Tools
````

For 15.3:
```
zypper ar -f https://download.opensuse.org/repositories/home:/polslinux:/astrophotography/openSUSE_Leap_15.3/ Astrophotography_Tools
````

For Tumbleweed:
```
https://download.opensuse.org/repositories/home:/polslinux:/astrophotography/openSUSE_Tumbleweed/ Astrophotography_Tools
```

And then:
```
zypper ref
zypper in ...
```

where `...` must be replaced with the wanted driver(s) (e.g. `indi-asi`, `indi-qhy`, `indi-qsi`, etc).
One can search for all indi packages with:

```
zypper se indi
```

I'm the current and only maintainer for those repo ATM. I'm working on integrating those packages into the main repos, but this could take a while :)

knro, 2021-04-25:
Thanks! I need to add this to INDI website soon.

paolostivanin, 2021-05-07:
Small update. I have an official devel project now, so please do not put the above outdated info on the website :)
The new official repos are:

openSUSE Leap 15.2 (stable version):
```
sudo zypper ar -f https://download.opensuse.org/repositories/Application:/astrophotography/openSUSE_Leap_15.2/ Astrophotography_Software
sudo zypper ref
```

openSUSE Leap 15.3 (beta version):
```
sudo zypper ar -f https://download.opensuse.org/repositories/Application:/Astrophotography/openSUSE_Leap_15.3/ Astrophotography_Software
sudo zypper ref
```

openSUSE Tumbleweed (rolling release):
```
sudo zypper ar -f https://download.opensuse.org/repositories/Application:/Astrophotography/openSUSE_Tumbleweed/ Astrophotography_Software
```

Please note that packages are being moved from the old to the new repo, so it might take a couple of days to have the whole set of packages built :)

knro, 2021-06-25:
Thank you, INDI download section updated.

closed 2021-06-25 by knro