indilib/indi #1438: Broken indi_asi_efw driver after upgrade

rseidler2019, 2021-04-28:
**Describe the bug**
After upgrade to the newest version (stable and bleeding) the indi_asi_efw driver is not responding to any calls, preventing
it from being shown in indi control panel of Kstars and the driver does not report anything in the log files.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver indi_asi_efw
2. Connect to indiserver using Kstars EKOS
3. No ASI EFW panel in indi control panel

**Expected behavior**
should see the ASI EFW in indi control and see logs on the server

**Desktop:**
 - OS: Ubuntu 20.04 on Raspi 4
 - Version 1.9.1

**Log Files**
[asi-efw-error.txt](https://github.com/indilib/indi/files/6391985/asi-efw-error.txt)

knro, 2021-04-28:
I can confirm this issue. It turned out to be more complex than I initially thought. Why would debian packaging breaks the drivers? The package (build log)[https://launchpadlibrarian.net/535942048/buildlog_ubuntu-focal-amd64.libindi_1.9.0~202104281314~ubuntu20.04.1_BUILDING.txt.gz] shows some warnings for missing symbols, but I've seen those before. 

@pawel-soja Could it be related to the way we changed ISXXX function in DefaultDevice?  Did we miss to include a file in the debian package?? I checked and couldn't find anything amiss. It's not only indi_asi_efw, it happens with indi_lx200gps...

I just noticed a pattern... all drivers with multiple inheritance (multiple parents) are affected by this. I don't think this is a coincidence. @pawel-soja could this somehow affect the drivers? It again doesn't explain this problem on arises when packaging debian files.

@xsnrg Do you have this issue on Fedora?

xsnrg, 2021-04-28:
The Fedora build appears to be working okay, at least for the devices I have been able to hook up (ASI 1600MC with EFW mini).

https://indilib.org/forum/focusers-filter-wheels/9595-zwo-efw-not-connecting.html?start=12#70792

knro, 2021-04-28:
Just to be sure, you tested these on a fresh Fedora installation and install the packages? The thing is, it works if I BUILD libindi and install it, but it doesn't work from the Deb packages. I tested this on a 20.04 VM under VirtualBox.

xsnrg, 2021-04-28:
It was not fresh, but did not have the asi drivers installed before. I am spinning up a Docker container to try it there, but my connection is slow. Here is a Dockerfile if anyone wants to try:
`from amd64/fedora:33
USER root
RUN dnf upgrade -y --refresh
RUN dnf install -y dnf-utils
RUN dnf copr enable -y xsnrg/stellarsolver-bleeding
RUN dnf copr enable -y xsnrg/kstars-bleeding
RUN dnf copr enable -y xsnrg/libindi-bleeding 
RUN dnf copr enable -y xsnrg/indi-3rdparty-bleeding 
RUN dnf install -y stellarsolver kstars indi-eqmod indi-asi xeyes
RUN useradd user -G wheel
RUN echo "user	ALL=(ALL)	NOPASSWD: ALL" >> /etc/sudoers
USER user
CMD ["/bin/bash"]
`

it will drop you to a prompt where you can run commands, including kstars, on Fedora 33

rseidler2019, 2021-04-28:
Thx for the fast reply! I also found an issue with the indi_eqmod that seems to be broken, too. It connects, than crashes and responds further with unable to open serial port, because it is allready used by an other process (the same PID as the process indi_eqmod is reported in lsof). Do I need to open another issue or could that be related?

beckrd, 2021-04-28:
FWIW, I just ran "indiserver driver indi_asi_efw" on Kubuntu 20.10 having just recompiled libasi and indi-asi on an amd64 machine, and I got the following messages:
`2021-04-28T19:41:11: startup: indiserver driver indi_asi_efw `
`2021-04-28T19:41:11: Driver driver: 2021-04-28T19:41:11: Driver driver: execlp: No such file or directory`
`Child process 29476 died`
`Child process 29477 died`
`2021-04-28T19:41:11: Driver driver: restart #1`
`2021-04-28T19:41:11: Driver indi_asi_efw: stderr EOF`
`2021-04-28T19:41:11: Driver indi_asi_efw: restart #1`
`Child process 29478 died`
This repeated for 10 restarts.

xsnrg, 2021-04-28:
I had a problem with the docker image not able to do SQL commands even though sqlite was installed... to look at later. 

I fired up a VirtualBox VM with Fedora 33 from scratch, and ran these commands: 
``` dnf upgrade -y --refresh
 dnf install -y dnf-utils
 dnf copr enable -y xsnrg/stellarsolver-bleeding
 dnf copr enable -y xsnrg/kstars-bleeding
 dnf copr enable -y xsnrg/libindi-bleeding 
 dnf copr enable -y xsnrg/indi-3rdparty-bleeding 
 dnf install -y sqlite stellarsolver kstars indi-eqmod indi-asi xeyes
```
The camera and the filter wheel work.

xsnrg, 2021-04-28:
@beckrd I think you may have the command wrong. Try this:

```indiserver indi_asi_wheel```

beckrd, 2021-04-28:
Does this help?

`indiserver -vvv indi_asi_wheel`
`2021-04-28T21:37:44: startup: indiserver -vvv indi_asi_wheel `
`2021-04-28T21:37:44: Driver indi_asi_wheel: pid=30857 rfd=3 wfd=6 efd=7`
`2021-04-28T21:37:44: listening to port 7624 on fd 4`
`2021-04-28T21:37:44: Driver indi_asi_wheel: sending msg copy 1 nq 1:`
`<getProperties version='1.7'/>`

I'm not sure how I connect to a running instance of indiserver, but the above was all I see on the console.

xsnrg, 2021-04-28:
With it running, does your filter wheel show up in Ekos?

beckrd, 2021-04-28:
If I bring up EKOS and start it and leave the existing indiserver running, I get the same tabs as if it weren't running (that is, no tab for the EFW).  The console information above includes this case.

xsnrg, 2021-04-28:
I am not sure where to go from here. Hopefully there are some hints here that help @knro get to the issue. It seems like your driver is running, but Ekos doesn't want to acknowledge it, but there are probably more steps to determine if that is the case for certain.

pawel-soja, 2021-04-28:
I can confirm.
**Doesn't work(ppa):**
```bash
#!/bin/bash

set -e

docker build -t indi-asi-ubuntu -<<EOF
FROM ubuntu:latest

RUN apt-get update
RUN apt-get install -y software-properties-common 

RUN add-apt-repository ppa:mutlaqja/indinightly
RUN DEBIAN_FRONTEND=noninteractive apt-get install -y indi-asi 
RUN dpkg -l | egrep "indi"
EOF

docker run --rm -it --privileged indi-asi-ubuntu /bin/bash
```
Run in docker:
```bash
indiserver indi_asi_wheel
```
Version:
```
ii  indi-asi                      1.9~202104281558~ubuntu20.04.1    amd64        INDI Driver for ZWO Optics ASI cameras
ii  indi-bin                      1.9.1~202104272038~ubuntu20.04.1  amd64        INDI server, drivers and tools
ii  libindi-data                  1.9.1~202104272038~ubuntu20.04.1  all          Instrument-Neutral Device Interface library -- shared data
ii  libindi1:amd64                1.9.1~202104272038~ubuntu20.04.1  amd64        Instrument-Neutral Device Interface library -- shared library
```

**Works(latest git):**
```bash
#!/bin/bash

set -e

docker build -t indi-git-ubuntu -<<EOF
FROM ubuntu:latest
WORKDIR /app

RUN apt-get update
RUN DEBIAN_FRONTEND=noninteractive apt-get install -y \
    software-properties-common git

RUN git clone --depth=1 https://github.com/indilib/indi.git
RUN ./indi/scripts/requisites-install.sh
RUN ./indi/scripts/googletest-build.sh
RUN ./indi/scripts/googletest-install.sh
RUN ./indi/scripts/indi-core-build.sh
RUN ./indi/scripts/indi-core-install.sh

RUN git clone --depth=1 https://github.com/indilib/indi-3rdparty.git
RUN ./indi-3rdparty/scripts/requisites-install.sh
RUN ./indi-3rdparty/scripts/indi-3rdparty-libs-build.sh
RUN ./indi-3rdparty/scripts/indi-3rdparty-libs-install.sh
RUN ./indi-3rdparty/scripts/indi-3rdparty-build.sh
RUN ./indi-3rdparty/scripts/indi-3rdparty-install.sh
EOF

docker run --rm -it --privileged indi-git-ubuntu /bin/bash
```
Run in docker:
```bash
LD_LIBRARY_PATH=/usr/local/lib indiserver indi_asi_wheel
```
_(*) LD_LIBRARY_PATH - scripts install libraries to /usr/local/lib_ 

Back to the main problem, I can fix this if I recreate the package build process.

pawel-soja, 2021-04-29:
I already know what's going on. Here is an example of how to solve the problem (asi wheel).
https://github.com/indilib/indi-3rdparty/tree/hotfix-loader
The linking order when building packages is not correct.

I will try to repair all libraries within 24 hours.

pawel-soja, 2021-04-29:
I found the source of the problem, why it works on debian and not on ubuntu.
https://lists.ubuntu.com/archives/ubuntu-devel/2018-June/040369.html
There is an additional flag when building packages `-Wl,-Bsymbolic-functions` It causes 'ISXXX' functions not to be called.

The ultimate solution is to get rid of these functions, I described it in: https://github.com/indilib/indi/pull/1375
INDI 3rd-Party library has been modified on a separate branch.
https://github.com/indilib/indi-3rdparty/compare/hotfix-loader

I can also remove this flag for backward compatibility.

knro, 2021-04-29:
Wow thanks for digging this up. This explains why I have no issues on Raspbian and it's only reported in Ubuntu. We need to get rid of them also in indi along with 3rdparty. Thank you so much for the quick fix.

pawel-soja, 2021-04-29:
Thanks, @knro If I re-create the problem, I am able to fix most critical problems within 24 hours.

I propose to remove `-Wl,-Bsymbolic-functions` https://github.com/indilib/indi/pull/1439
Then, just rewrite all drivers so that they don't duplicate the ISXX function implementation.

I have already done for INDI-3rdParty (https://github.com/indilib/indi-3rdparty/tree/hotfix-loader), but you can still test it with caution.
INDI Core drivers, I'll be rewriting this evening.

knro, 2021-04-29:
Thanks, I think it's better to wait for the fix than to rebuild without the -Wl,-Bcymbolic-functions 

It's greatly appreciated, it's a monumental task to do it for all the drivers.

closed 2021-05-01 by knro