indilib/indi #1443: Sestosenso2 hang

knro, 2021-05-01:
As reported in detail here: https://indilib.org/forum/focusers-filter-wheels/9347-esatto-2-focusing-hangout.html

I checked the driver, it appears there is no timeout for read(..) at line 1233. I believe there should always be a timeout for read to prevent any cases where it can hang the driver.

@zyziuk Can you please check?

zyziuk, 2021-05-01:
Hello,

I wrote about it in comment below 
<https://github.com/karlrees/indi/commit/7ef61760e451bd0038d479edf13ee18e0994e219>
Therefore I still use version 1.8.9.

Piotr

knro, 2021-05-02:
Thanks, I just noticed this. @karlrees can you please check what broke this since many users are unable to use the driver properly now.

zyziuk, 2021-05-02:
I noticed that some auto update installs newer version of  
*libindidriver.so.1*then focuser is not working. To fix it I restore 
only libindidriver.so.1 to previous version (1.8.9), focuser works fine.

Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/bin/indiserver
-- Up-to-date: /usr/lib/aarch64-linux-gnu/libindiclient.a
-- Up-to-date: /usr/include/libindi/baseclient.h
-- Up-to-date: /usr/lib/aarch64-linux-gnu/libindidriver.a
-- Up-to-date: /usr/lib/aarch64-linux-gnu/libindidriver.so.1.8.9
*-- Installing: /usr/lib/aarch64-linux-gnu/libindidriver.so.1*
-- Up-to-date: /usr/lib/aarch64-linux-gnu/libindidriver.so
-- Up-to-date: /usr/bin/indi_imager_agent
-- Up-to-date: /usr/lib/aarch64-linux-gnu/libindilx200.so.1.8.9
...

Piotr

karlrees, 2021-05-02:
I don't understand what's going on here.  I never touched this file.

knro, 2021-05-02:
Sorry about that. I just checked the history and @lantictac had the most recent changes. Can you please check?

karlrees, 2021-05-02:
So I _think_ my involvement here is that zyziuk commented on this in a commit I made in my fork when I was trying to sync my fork up with master.  I don't recall exactly what I was doing or thought I was doing at the time, but I ended up messing up my fork, and eventually just did a hard reset.  I've looked over the history, and I'm pretty sure I didn't do anything that actually made it out of my fork.  But, in any event, I've never deliberately touched this file, and don't have the relevant equipment to test it.

lantictac, 2021-05-04:
Sorry for the delay, I only just saw this issue.

My changes to the driver were only tested on the SS2 and the CommandSet::send() function largely mirrors the code that was formerly in CommandSet::sendCmd() in the previous revision. Notably there has never been a timeout value.

That said, my updated code assumes a response for every command issued (while the former would skip the blocking read() call if it didn't care about a response). Under my tests with SS2 using the latest firmware (v1.3) every command would result in a response from the device - it certainly didn't hang (I've been using it for the past couple of months without issue).

I guess a safe starting point might be to add a timeout to the read() call. And perhaps an error output that could indicate what command doesn't get a response within the allotted time. This - at least - should prevent a hang and give us an idea of what might be causing it.

Also, can people experiencing the issue confirm which firmware version they're using?

gilesco, 2021-05-04:
I experienced the problem with the SS2 (firmware 1.3) at one point with a git compiled indi / kstars in late March. I think the comment that reverting to libindidriver.so.1.8.9 resolved the problem was interesting, perhaps this is a focuser polling issue?
Unfortunately, I don't have a permanent setup, and various factors (weather, latitude, work, inability to easily move between INDI versions) prevent me from setting up and testing more right now.

knro, 2021-05-04:
Thanks. I was hesitant to modify the driver since I didn't want to introduce any regressions in the code. @lantictac What do you suggest in order to resolve this? Can you replicate the issue in your current setup with the latest updates from INDI?

knro, 2021-05-05:
I can confirm this issue as well with Essato focuser. It takes two commands to bring it to position. @lantictac Any ideas thus far on how to resolve this?

lantictac, 2021-05-06:
> I can confirm this issue as well with Essato focuser. It takes two commands to bring it to position. @lantictac Any ideas thus far on how to resolve this?

When you say it takes two commands? The first command silently fails or...?

I've attempted to resurrect my dev environment but I haven't had a chance to dig any deeper just yet. That said, if you can identify the command that doesn't get a response that might be a useful starting point. I assume it's the "MOT1" :  { "GOTO" : NNN } command that is failing to generate a response?

commit 079ab23a (2021-05-06): Using the new improved INDI::Timer instead of legacy timers. For some reason the older timers stopped working, so this needs to be investigated further. However, the driver now is confirmed to work correctly. Fixes #1443
commit 4f6ff7e2 (2021-05-06): Using the new improved INDI::Timer instead of legacy timers. For some reason the older timers stopped working, so this needs to be investigated further. However, the driver now is confirmed to work correctly. Fixes #1443
closed 2021-05-06 by knro