indilib/indi #1445: Unguided Dithering not following a random pattern

fenriques, 2021-05-05:
When the checkbox 'Non-guide dither pulse' (Ekos Guide Tab -> options->guide) is checked, dither seems to occur in one direction only and not follow the expected  random pattern.
It was reported by several users in the forum:
https://indilib.org/forum/ekos/9345-unguided-dithering.html

Seems not mount related: CEM and 10Micron are affected.
It was tested with several pulse durations from 0.3s to 2.0s.
Also, changing setting in the upper 'dither' box doesn't affect the behavior.

Attached a screenshot of the settings and an image of the dither pattern over 6 hours
 
 - OS: Ubuntu
 - Version Kstars 3.5.2

[image]

[image]
(6hrs integration of not registered subs with 0.5s dither pulse)

knro, 2021-05-05:
Thanks for the report, but this is not an INDI bug. Should be filed at KStars: https://invent.kde.org/education/kstars/-/issues

closed 2021-05-05 by knro

fenriques, 2021-05-05:
ok. i will post it there.