indilib/indi #1446: Pegasus drives fail in HandShake

jmocenigo, 2021-05-05:
### Describe the Bug
In my experience, both the Pegasus PPBA and Pegasus DMFC drivers fail to connect. Failure occurs immediately during HandShake with the following message: [ERROR] Serial read error: Timeout error).

### To Reproduce:
1. Start kstars
2. Load profile with a PPBA device
3. Try connecting PPBA device
4. Failure occurs, log shows following lines:
```plaintext
[2021-05-05T13:22:52.946 EDT DEBG ][           org.kde.kstars.indi] - Pegasus PPBA : "[DEBUG] Port FD 3 "
[2021-05-05T13:22:52.946 EDT DEBG ][           org.kde.kstars.indi] - Pegasus PPBA : "[DEBUG] Connection successful, attempting handshake... "
[2021-05-05T13:22:52.946 EDT DEBG ][           org.kde.kstars.indi] - Pegasus PPBA : "[DEBUG] CMD <P#> "
[2021-05-05T13:22:54.826 EDT INFO ][           org.kde.kstars.indi] - Pegasus PPBA :  "[ERROR] Serial read error: Timeout error "
[2021-05-05T13:22:54.826 EDT DEBG ][           org.kde.kstars.indi] - Pegasus PPBA : "[DEBUG] Handshake failed. "
[2021-05-05T13:22:54.830 EDT DEBG ][           org.kde.kstars.ekos] - "Pegasus PPBA"  is disconnected.
[2021-05-05T13:22:54.831 EDT INFO ][           org.kde.kstars.ekos] - "Pegasus PPBA failed to connect.\nPlease ensure the device is connected and powered on."
```

### Expected behavior
Successful connection to PPBA and DMFC

### Additional comments
Failure to connect also occurs when attempting to connect Pegasus DMFC, but log has less information.

See attached log for both connection attempts.
[log_13-22-32.txt](https://github.com/indilib/indi/files/6429177/log_13-22-32.txt)

If you need me to run indiserver and indi_pegasus_ppba or indi_dmfc_focus from the command line, please send instructions for what you want executed.

### Environment
 - OS: Ubuntu Mate
 - Version: 20.10
 - Architecture: amd64
 - Computer: System76 Meerkat

### Log Files
[log_13-22-32.txt](https://github.com/indilib/indi/files/6429184/log_13-22-32.txt)

jmocenigo, 2021-05-08:
I have some more information on this matter.

In the case of the PPBA, the fault appears to be in my stars. I had set the BAUD rate to 19200, but only 9600 works. So, with that change, I am able to connect to PPBA.

The case of the focuser is harder to explain. OK, so I had been using "Pegasus DMFC" driver with version 1.8.9, but apparently part of the 1.9.0 changes is that there is now a "Pegasus FocusCube" driver. My unit (a Pegasus FocusCube 2) returns OK_FC as the ack string and the "Pegasus DMFC" driver now requires "OK_DMFCN" as the ack string, so one would think that changing my driver to "Pegasus FocusCube" would fix things, but oddly it does not. Even though the code for DMFC driver (dmfc.cpp) is very similar to the FocusCube driver code (pegasus_focuscube.cpp), the tty_read_section call during pegasus_focuscube.cpp ack() processing times-out, whereas the dmfc.cpp ack() tty_read_section does not?!? Of course, the dmfc.cpp ack() then fails because "OK_FC" != "OK_DMFCN" (i.e., the strstr() call fails). If I change the dmfc.cpp ack() string from "OK_DMFCN" to "OK_FC", then I am able to connect and all is well.

I realize the above is just a hack and I am not at all suggesting it as an answer, but I cannot figure out why the tty_read_section behavior is different in these two drivers (all else is the same). I tried using tty_set_debug to get IDLog to generate some output, but either no IDLog was triggered or the output is going to some other log, because I saw nothing related to it in the log. I also tried tty_nread_section (which seems a safer choice to me since res is a small, fixed-size buffer), but it made no difference.

So, bottom-line:
1. Apparently no bug with the PPBA code (sorry about the red-herring)
2. Weird bug in Pegasus FocusCube code
3. My hack to dmfc.cpp appears to indicate that the bug is definitely somewhere in the Pegasus FocusCube driver.

knro, 2021-05-09:
Or just try to connect to FocusCube alone and see if that works directly.

jmocenigo, 2021-05-09:
My testing was done with a profile that only had focuscube plus simulated mount & ccd. The focuscube was plugged directly into the computer. The ppba was only distributing power, I was not using its USB hub feature. 

Let me know if you need me to try something else.

jmocenigo, 2021-05-09:
Also, with the focuser-only profile, if I do nothing except change it to use the (hacked) DMFC driver vs the FocusCube driver, upon connection it works - and when I reverse it (i.e., switch back to FocisCube driver) upon connection it fails — that’s with no change to anything except changing the driver in the profile.

justanenduser, 2021-05-09:
I too was using the wrong driver (DMFC) and Jasem directed me to the correct one (FocusCube). However, I had to delete my profile and recreate it in order for the driver to properly work with the focuser. I am using Patrick Chevalley's INDIstarter rather than EKOS.

jmocenigo, 2021-05-09:
Did you delete all profiles or just the one?

justanenduser, 2021-05-09:
INDIstarter allows me create a single profile listing all the drivers for all the devices used on a particular telescope setup (newtonian vs refractor for instance). So the profile delete (newtonian in this case) necessitated a complete recreation of that profile to include the drivers for all the devices used on that setup.

jmocenigo, 2021-05-09:
@justanenduser : you are correct... I created an entirely new profile and was able to connect to FocusCube. Thanks... but why that should make a difference remains a mystery, but such is life.

jmocenigo, 2021-05-09:
Final note from me on this matter: once I was able to successfully connect using a new profile (with just the Focuser and the CCD simulator in it), I was then able to change the driver in my existing full-profile to Pegasus Focuser and it, too, connected and worked.

closed 2021-06-26 by knro