indilib/indi #1448: iOptron CEM70 mount time&location not correct in INDI/Kstars

paolostivanin, 2021-05-06:
Hello,
I don't know whether this is an INDI or Kstars issue. So if this is not the right place for this bug, please let me know and I'll open it on the right issues tracker :)

When setting the option "Mount update Kstars for both time and location", Kstars doesn't have the correct time but, instead, it's off by 1 hour.

The current setup is:
* iOptron CEM70
* RPi 4b 4GB
* StellarmateOS 1.5.8 with updates applied as of yesterday morning

On the mount side, the HC is setup like the following:
* DST: Y
* UTC offset: +060
The reported time and date on the HC are perfect. Even when the GPS OK'd, settings are not changed (obviously) and the information presented are fine.

Then I head over to Kstars, start the CEM70 profile in EKOS and the time it's set 1 hour off (behind current time). Therefore, if on the HC the time is 15:30:00, on Kstars the time is 14:30:00.

A workaorund I've tried and that partially worked, was to set DST to N and UTC offset to +120 in the HC. In this way, Kstars had the correct time. But the problem is that as soon as the mount's internal GPS is OK'd, the info are set back to DST Y and UTC offset +060.
Therefore, this is not really a solution, because it requires always having the HC attached to the mount and always changing the settings after the GPS OK'd.

Would it be possible for INDI/Kstars to get the correct time based on the mount info? Considering that UTC time is correct, UTC offset is correct, and DST option is correct, INDI/Kstars should be able to set the right time (something like `UTC+offset+DST`).

Example:
Mount UTC time: 13:30:00
UTC offset: +060
DST: Y
Correct time: 13:30+1:00+1:00 = 15:30:00

paolostivanin, 2021-05-10:
I've also tried the "Mount update only location", but the time still get synced to the wrong value.

knro, 2021-05-10:
is it safe to assume that day light saving always increases offset by 1 hour? I live in a region where there is no DST offset (thankfully), so not sure if that's a correct assumption. If yes, the UTC offset can be increased by 60 minutes if the DST flag is active.

knro, 2021-05-10:
Maybe post about this in INDI forum? Let's find out if it happens to more users. I recall some users reporting similar issue in the past.

paolostivanin, 2021-05-10:
As expected, there seem to some exceptions to the 1h rule out there :(
This complicates things even further...

paolostivanin, 2021-05-10:
@knro ok! Tomorrow morning I'll open a new post there!

paolostivanin, 2021-05-11:
Opened a new post in the forum, let's see: https://indilib.org/forum/mounts/9662-we-need-your-feedback-do-you-also-face-this-issue.html

paolostivanin, 2021-05-11:
I was thinking of a "location and mount agnostic" solution, and got this idea:

[image]

What do you think?

In this way we wouldn't force anyone to have a fix, but it would be an "opt-in" solution decided by the user, in case he/she should be facing the issue.

knro, 2021-05-11:
I'm afraid this could be used to hide real issues with the mount (if it was really off by sometimes) and then users would use this just to mask off the issue. A better solution is required.

glundby, 2021-05-11:
I definitely have been hit by this also. Even more complicated, I'm running two RPi, the second to fire a Piggy-Back camera in asynchronous parallell. So one RPi owns the mount, the other has a simulator mount. Then I got this problem. The RPi with the  IOptron GEM45 mount always ended up one hour different from the other (both has a real time clock and both get time from internet but the setting of "mount updates all" overrides this. Very confusing and not good for me trying to make jobs go in parallell on the two RPi so they could change target kind of in sync.

paolostivanin, 2021-05-11:
As reported by PDB on the forum, the issue may reside here: https://github.com/indilib/indi/blob/a6ffc654e24e0fb7a1ab0af631b0955f5567f377/drivers/telescope/ieqdriverbase.cpp#L543-L551 It seems that the UTC value is not parsed correctly.

IOptron RS232 manual:
```
Command: “:GLT#”
Response: “sMMMYYMMDDHHMMSS#”
...
The 4th digit indicate the Daylight-Saving Time, 0 means Daylight Saving Time has not been
observed, 1 means Daylight Saving Time has been observed.
...
```

Full answer and explanation [here](https://www.indilib.org/forum/mounts/9662-we-need-your-feedback-about-mount-kstars-indi-time-sync-do-you-also-face-this-issue.html#71231)

knro, 2021-05-12:
I pushed a hack fix to ioptron_dst_fix branch, maybe you can test this now?

```
git checkout ioptron_dst_fix
cd ~/Projects/build/indi-core
sudo make -j8 install

```

paolostivanin, 2021-05-12:
Thanks a lot @knro :) tomorrow I'll test it and report back!

paolostivanin, 2021-05-13:
@knro can confirm that your fix solved the issue. In case anyone would like to try, I've compiled libindi for opensuse 15.3 and tumbleweed: https://build.opensuse.org/project/show/home:polslinux:teststuff
It's the stable 1.9.0 branch + the small DST fix.

knro, 2021-05-13:
Thanks, I'll merge to master to give access to more people to try to nightlies. If any trouble is found, we can always revert back.

paolostivanin, 2021-05-14:
@knro just to double check, I see that you've bumped the driver version only for certain mounts (https://github.com/indilib/indi/commit/85d200d1a280364afbef813a9f164ed0e4b6dede). Is this expected? Thanks :)

knro, 2021-05-14:
@paolostivanin Good call! I updated the rest, thanks! :-)

knro, 2021-05-28:
Closing this issue since it's fixed.

closed 2021-05-28 by knro

ButtCrackBob, 2023-02-24:
Nothing has been fix for me. I have tested both APT and Kstars and continue to get incorrect Longitude numbers.
[image]

[image]

knro, 2023-02-25:
That's actually correct. INDI longitude is 0 to +360 eastwards. It's converted to iOptron format when sent over, so it's not an issue.