indilib/indi #1449: Update BaseMediator and BaseClient

knro, 2021-05-06:
Along with the recent changes to Properties, the BaseMediator, BaseDevice, and BaseClient classes need to be updated accordingly. Of the important things that caused issues in the past is use of raw pointers to communicate INDI::Property. We should use shared_ptr so that the object is reserved until all parties stop using it.

@pawel-soja Besides the migration to shared_ptr, do you have any suggestions? Let's try to find a way to deprecate the existing interface while allow existing clients to be build against it without requiring to be re-written again. Might be a good option as well to put the D pointer to use here as well.

pawel-soja, 2021-05-07:
@knro 
Currently
- [INDI::Property](https://github.com/indilib/indi/blob/master/libs/indibase/property/indiproperty.h)
- [INDI::PropertyBasic template](https://github.com/indilib/indi/blob/master/libs/indibase/property/indipropertybasic.h) See below
- [INDI::PropertyBlob](https://github.com/indilib/indi/blob/master/libs/indibase/property/indipropertyblob.h)
- [INDI::PropertyLight](https://github.com/indilib/indi/blob/master/libs/indibase/property/indipropertylight.h)
- [INDI::PropertyNumber](https://github.com/indilib/indi/blob/master/libs/indibase/property/indipropertynumber.h)
- [INDI::PropertySwitch](https://github.com/indilib/indi/blob/master/libs/indibase/property/indipropertyswitch.h)
- [INDI::PropertyText](https://github.com/indilib/indi/blob/master/libs/indibase/property/indipropertylight.h)

They have a built-in `shared_ptr`.
So the copies of the classes store the same data with a counted reference to the data.
```cpp
class EasySharedVirtualClass
{
public:
    EasySharedVirtualClass() = default;
    virtual ~EasySharedVirtualClass() = default;

private:
    void * d_ptr; // or std::shared_ptr
};
/* ... */
std::cout << "sizeof void *                 " << sizeof(void *) << std::endl;
std::cout << "sizeof EasySharedVirtualClass " << sizeof(EasySharedVirtualClass) << std::endl;
std::cout << "sizeof INDI::Property         " << sizeof(INDI::Property)         << std::endl;
std::cout << "sizeof INDI::PropertySwitch   " << sizeof(INDI::PropertySwitch)   << std::endl;
```
Output:
```
sizeof void *                 8
sizeof EasySharedVirtualClass 16
sizeof INDI::Property         16
sizeof INDI::PropertySwitch   16
```

INDI is prepared to use `INDI::PropertyXXX` like this:
```cpp
#include <list>
#include <indipropertyswitch.h>
/* ... */

INDI::PropertySwitch switchSP {3};
switchSP[0].fill("NAME", "LABEL", ISS_OFF);
/* ... */

std::list<INDI::PropertySwitch> listOfSwitches;
listOfSwitches.push_back(switchSP); // copy built-in shared_pointer

std::list<INDI::Property> listOfProperty;
listOfProperty.push_back(switchSP); // copy build-in shared_pointer
```

knro, 2021-05-07:
@pawel-soja Thanks for the explanation. I think [INDI::BaseMediator](https://github.com/indilib/indi/blob/1a4b61c2e7c0d8a7fcdbe932d27831343de0e5c2/libs/indibase/indibase.h#L80) should no longer use raw pointers to pass INDI::BaseDevice or INDI::Property. Similarly for all the newXXX functions that report the values. A single newPropertyValue(...) could probably suffice, though we should retain newXXX for backward compatibility at any rate.

pawel-soja, 2021-05-08:
@knro I'll clean up a bit and see what I can do.
https://github.com/indilib/indi/tree/feature-d_ptr-baseclient

pawel-soja, 2021-05-08:
@knro The feature below is quite problematic to easily switch to new implementations.
https://github.com/indilib/indi/blob/1a4b61c2e7c0d8a7fcdbe932d27831343de0e5c2/libs/indibase/baseclient.h#L134-L149
I also don't see it being used anywhere.
Can I remove it?

I can also suggest:
```cpp
BaseDeviceList getDevices(uint16_t driverInterface) const;
// or
BaseDeviceList getDeviceList(uint16_t driverInterface) const;
```

pawel-soja, 2021-05-09:
## In brief
Current implementation in INDI is correct, but it is not resistant to improper use.
```cpp
virtual void removeProperty(INDI::Property *property) = 0;
```
If you want to use asynchronous calls, implementing this function, you should ensure a copy of the data.
```cpp
void Implementation::removeProperty(INDI::Property *property)
{
    emit someSignal(*property); // copy constructor, See the description at the end
    // where someSignal:
    // void someSignal(INDI::Property property);
}
```

## An example of correct and incorrect use of current INDI libraries
```cpp
#include <QCoreApplication>
#include <QDebug>
#include <QTimer>
#include <memory>

// Current implementation in INDI
class PropertyPrivate
{
public:
    PropertyPrivate()  { qDebug() << __PRETTY_FUNCTION__; }
    ~PropertyPrivate() { qDebug() << __PRETTY_FUNCTION__; }
};

class Property
{
public:
    Property()
        : d_ptr(new PropertyPrivate)
    { }

protected:
    std::shared_ptr<PropertyPrivate> d_ptr;
};

// API
class Mediator
{
public:
    virtual ~Mediator() = default;
    virtual void removeProperty(Property *property) = 0;
    /* ... */
};

class ManagerBroken: public QObject, public Mediator
{
    Q_OBJECT
public:
    ManagerBroken()
    {
        QObject::connect(this, &ManagerBroken::signal, this, &ManagerBroken::slot, Qt::QueuedConnection);
    }

    void removeProperty(Property *property) override
    {
        qDebug() << __PRETTY_FUNCTION__;
        emit signal(property);
    }

    void slot(Property *) { qDebug() << __PRETTY_FUNCTION__; }

signals:
    void signal(Property *prop);
};

class ManagerFixed: public QObject, public Mediator
{
    Q_OBJECT
public:
    ManagerFixed()
    {
        QObject::connect(this, &ManagerFixed::signal, this, &ManagerFixed::slot, Qt::QueuedConnection);
    }

    void removeProperty(Property *property) override
    {
        qDebug() << __PRETTY_FUNCTION__;
        emit signal(*property); // copy built-in shared_ptr
    }

    void slot(Property) { qDebug() << __PRETTY_FUNCTION__; }

signals:
    void signal(Property prop);
};

Q_DECLARE_METATYPE(Property)
#include "main.moc"

int main(int argc, char *argv[])
{
    QCoreApplication a(argc, argv);
    ManagerBroken managerBroken;
    ManagerFixed  managerFixed;

    // scope
    {
        Property *prop = new Property; // Current, incorrect implementation in INDI
        managerBroken.removeProperty(prop);
        //managerFixed.removeProperty(prop);
        delete prop;
    }

    QTimer::singleShot(100, [&]{ a.quit(); });
    return a.exec();
}
```

#### ManagerBroken output:
```
PropertyPrivate::PropertyPrivate()
virtual void ManagerBroken::removeProperty(Property*)
PropertyPrivate::~PropertyPrivate()
void ManagerBroken::slot(Property*)  <============== Reference to data that no longer exist :(
```

#### ManagerFixed output:
```
PropertyPrivate::PropertyPrivate()
virtual void ManagerFixed::removeProperty(Property*)
void ManagerFixed::slot(Property) <============== All OK :)
PropertyPrivate::~PropertyPrivate()
```
## Problems in KStars
I see that the code has evolved and I admit, I can not find problems.

I guess that to workaround problems, it is copying field data before `INDI:: Property` will be removed...
```cpp
void ClientManager::removeProperty(INDI::Property *prop)
{
    const QString name = prop->getName();
    const QString device = prop->getDeviceName();
    emit removeINDIProperty(device, name);

    // If BLOB property is removed, remove its corresponding property if one exists.
    if (blobManagers.empty() == false && prop->getType() == INDI_BLOB && prop->getPermission() != IP_WO)
    {
        for (QPointer<BlobManager> bm : blobManagers)
        {
            const QString bProperty = bm.data()->property("property").toString();
            const QString bDevice = bm.data()->property("device").toString();
            if (bDevice == device && bProperty == name)
            {
                blobManagers.removeOne(bm);
                bm.data()->disconnectServer();
                bm->deleteLater();
                break;
            }
        }
    }
}
```
...passing fields...
```cpp
// file: clientmanager.h
        void newINDIDevice(DeviceInfo *dv);
        void removeINDIDevice(const QString &name);

        void newINDIProperty(INDI::Property *prop);
        void removeINDIProperty(const QString &device, const QString &name);

        void newBLOBManager(const char *device, INDI::Property *prop);

        void newINDIBLOB(IBLOB *bp);
        void newINDISwitch(ISwitchVectorProperty *svp);
```
@knro So where is the problem?
I saw timer with removing INDI::Property ;)

## API change - Idiot-proof
Of course, we can do this simpler to use by modifying API (INDI::BaseMediator).
```cpp
virtual void removeProperty(INDI::Property property) = 0;
```
I will repeat, copying `INDI::Property` is not expensive because it is `shared_ptr`.
```cpp
// file: indiproperty.h
class Property
{
    DECLARE_PRIVATE(Property)
public:
/* ... */

protected:
    std::shared_ptr<PropertyPrivate> d_ptr; // <===========
    Property(PropertyPrivate &dd);
};
```

@knro I can take care of the new API for Indi and Indi-3rdparty, and send modifications for KStars. Is it ok?

pawel-soja, 2021-05-10:
From:
https://github.com/indilib/indi/blob/86d32f2a02308ccd954a810e1f427d8554ecb717/libs/indibase/basedevice.h#L157-L161
To:
https://github.com/indilib/indi/blob/c7699c63f4826a8c1661871097d2d85c54028bfe/libs/indibase/basedevice.h#L156-L160

From:
https://github.com/indilib/indi/blob/86d32f2a02308ccd954a810e1f427d8554ecb717/libs/indibase/indibase.h#L98-L104
To:
https://github.com/indilib/indi/blob/c7699c63f4826a8c1661871097d2d85c54028bfe/libs/indibase/indibase.h#L101-L106

- INDI Core https://github.com/indilib/indi/tree/feature-d_ptr-baseclient
- INDI 3rdParty https://github.com/indilib/indi-3rdparty/tree/upgrade-indi-api-2
- KStars https://invent.kde.org/pawel-soja/kstars/-/tree/upgrade-indi-api-2 or https://github.com/pawel-soja/kstars/tree/upgrade-indi-api-2

Not tested.

knro, 2021-05-10:
Wow, bravo for the detail explanation @pawel-soja !! This is awesome! I think I never wanted to copy before because it would have been expensive (and also not really possible because we lacked a copy ctor) but now with shared_ptr, it make sense to copy and forget about it!

I think we can proceed with the current changes, but there are two issues:

1. We cannot remove existing removeProperty(INDI::Property*)..etc just yet. This would break a lot of existing clients outside of the ones in INDI & KStars. Same thing for the basedevice functions. We should perhaps mark it as deprecated and keep using it somehow alongside the update function?
2. I added the hack to delete INDI::Property pointer with a timer which is terrible I know. So while internally we have a shared_ptr, INDI::Property is still managed as a list of raw pointers in BaseDevice. Should this change? I guess we can now safely delete the pointer since we're sending a copy and we don't need to worry about client accessing a dangling pointer anymore.

knro, 2021-05-14:
@pawel-soja Thank you for the work thus far, great progress have been made to transition to the new standard Property/Properties format. I think next item to tackle is indibase/basedevice/baseclient. We can now safely copy the properties. However, which signature should we opt for? i.e.

`virtual void newProperty(INDI::Property property) override;`

or

`virtual void newProperty(const INDI::Property &property) override;`

Wouldn't this also make newXXX (e.g. newSwitch) functions obsolete? Looks like it's going to be challenging to keep backward compatibility in indibase/baseclient.

pawel-soja, 2021-05-14:
A reference should be used, but should not be used for asynchronous signals / slots in qt.
By the way, I see you got rid of `thread delete property hack`
 (https://github.com/indilib/indi/commit/e37dcf4f8d42a2689f109b581308cfd73f57eb47).
Is the dereference done before entering the asynchronous part in KStars?

I would also like to complete widget access implementation.
So as to create an easy-to-use INDI::PropertyXXX and INDI::WidgetXXX interface.
Currently, they are WidgetView, which will enable easy refactoring in the future.
https://github.com/indilib/indi/blob/e37dcf4f8d42a2689f109b581308cfd73f57eb47/libs/indibase/property/indipropertybasic.h#L27-L31
I think it will first reimplement all functions where the signature will not cause any compatibility problems. I will analyze the sources of KStars and clean up a bit. I will then be able to suggest the some solution.

knro, 2021-05-18:
Thanks! Today, while I was writing driver for AstroTrac mount using the new INDI::Property syntax, I ran across a new requirements. Looks like we need to add .loadConfig to the property and then use IUGetConfigXXX(..) to read from the config file if it exists instead of manually doing it for each property.

I actually started doing this, but wanted to make sure about something. If there is a driver-only side implementation (like this is), how does it affect clients? would loadConfig be visible from a client side of view?

pawel-soja, 2021-05-18:
I would like to standardize this in the future, as I see no reason why these functions should not be available on the client side.
The C to C++ transition layer is the INDI::PropertyView/INDI::WidgetView classes.

Please, see:
https://github.com/indilib/indi/blob/master/libs/indibase/property/indipropertyview_client.cpp
https://github.com/indilib/indi/blob/master/libs/indibase/property/indipropertyview_driver.cpp

closed 2021-09-28 by knro