indilib/indi #1451: Handshake failure on PPBA and FocuCube 2 connection attempts

justanenduser, 2021-05-07:
Handshake failure when indiserver attempts to talk to FocusCube 2. 

SSH to PI 4.
Start remote indiserver (on PI 4) with needed device drivers using local INDIstarter with Auto-Connect on all drivers.
Veriify device driver status on remote using ps -eH | grep indi.
Verify driver connections to devices using INDI Client tab on INDIstarter server window.
Click CONNECT on the Main Control tab of any unconnected device to initiate any necessary connection.
Observe connection status of device on Main Control tab.
Observe messages in bottom window.

Expected:
Successful connection to FocusCube 2.

Desktop OS: 64 bit Fedora 33, distribution provided. Up-to-date a/o 07 May 2021.
Remote OS: 64 bit Fedora 33 on PI 4, distribution provided. Up-to-date a/o 07 May 2021.
Connection to remote: ethernet

Error log summary:
DEBUG	637.619119 sec	: Connecting to /dev/ttyUSB1 @ 19200
DEBUG	637.623495 sec	: Port FD 4
DEBUG	637.623645 sec	: Connection successful, attempting handshake...
DEBUG	637.623789 sec	: CMD <0X23>
DEBUG	637.638939 sec	: RES <OK_FC>
INFO	637.639077 sec	: Error retrieving data from Pegasus DMFC, please ensure device is powered and the port is correct.
DEBUG	637.639140 sec	: Handshake failed.

-or-

DEBUG	647.741358 sec	: Connecting to /dev/serial/by-id/usb-Pegasus_Astro_USB_Control_HUB_revA_UCH5V2TJH-if00-port0 @ 19200
DEBUG	647.745276 sec	: Port FD 4
DEBUG	647.745385 sec	: Connection successful, attempting handshake...
DEBUG	647.745449 sec	: CMD <0X23>
ERROR	650.748595 sec	: Ack error: Timeout error.
INFO	650.748765 sec	: Error retrieving data from Pegasus DMFC, please ensure device is powered and the port is correct.
DEBUG	650.748827 sec	: Handshake failed.

-or-

DEBUG	651.273078 sec	: Connecting to /dev/serial/by-id/usb-Pegasus_Astro_FocusCube_2_PA5EJ9CE-if00-port0 @ 19200
DEBUG	651.277003 sec	: Port FD 4
DEBUG	651.277108 sec	: Connection successful, attempting handshake...
DEBUG	651.277170 sec	: CMD <0X23>
DEBUG	651.288074 sec	: RES <OK_FC>
INFO	651.288185 sec	: Error retrieving data from Pegasus DMFC, please ensure device is powered and the port is correct.
DEBUG	651.288243 sec	: Handshake failed.

Complete debug log:
[indi_dmfc_focus.log](https://github.com/indilib/indi/files/6443137/indi_dmfc_focus.log)

knro, 2021-05-07:
Please use "Pegasus FocusCube" driver.

justanenduser, 2021-05-08:
Thank you. I seem to have missed the driver change. Things work as expected. :-)

closed 2021-05-08 by justanenduser