indilib/indi #1456: indi_paramount_telescope driver takes too long to report pulseguide move result

interplanetarychris, 2021-05-16:
**Describe the bug**
I've posted details about this problem on the [PHD2 forum](https://groups.google.com/g/open-phd-guiding/c/1ezYaVY_ss0)

I am opening this issue while I look into the code to determine what might be causing excessive delay for short-duration pulse-guide requests (eg., 9.1 seconds to implement a 37ms pulse guide request).  

It appears that DirectGuide requests are made **synchronously** - although that may or may not be problematic.

**To Reproduce**
Any guiding which requests a guide correction.

**Expected behavior**
I would expect guiding to occur at the frequency of the guide imaging duration, but instead, PHD2 often waits several multiples of the guide-image duration before hearing a response from the driver, and initiating the next image.

**Environment:**
 - OS: Ubuntu 20.04.2 LTS
 - Version: indi_paramount_telescope v1.1

**Log Files**
https://www.dropbox.com/s/l6n42p9wvy26798/indi_paramount_telescope_guiding.zip?dl=0

interplanetarychris, 2021-05-16:
I compiled and tested a version that sends the pulses asynchronously — but it appeared to have no effect.  PHD2 is still sometimes waiting ~8 seconds for completion status of a 90ms pulse.

Examining the PHD2 response, it looks like the paramount driver needs to more promptly send a "GuideComplete" call in order to provide a IDSetNumber result to PHD2, but this looks like unimplemented functionality in the paramount driver.  I'll examine the "timed guide" code in other drivers to try and figure out an implementation appropriate for the Paramount.

knro, 2021-05-16:
Thank you for the report. Please check how the other drivers are implementing this functionality as well to see if there is any lag from the driver itself.

interplanetarychris, 2021-05-18:
Per the PR https://github.com/indilib/indi/pull/1458, I believe I've addressed the bug.

closed 2021-05-18 by knro