indilib/indi #1466: indi_eqmod_telescope crash on Ubuntu 20.10 just after it connects

atoy40, 2021-05-24:
**Describe the bug**
When connecting the eqmod driver, it start to work (all the tabs appears with coherent values) then one second later, it crash

**To Reproduce**
Exact steps to reproduce the behavior.
1. run indiserver with eqmod driver
2. connect ekos to the indiserver (with auto connect unchecked)
3. connect the eqmod device from the indi control panel

**Expected behavior**
No crash :)

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.10
 - Version : 1.9.0 (from stable PPA)

**Log Files**
I've generated a backtrace using gdb and the indi-eqmod-dbg symbols package :

```
Thread 2.1 "indi_eqmod_tele" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff540bb40 (LWP 46769)]
__strcmp_avx2 () at ../sysdeps/x86_64/multiarch/strcmp-avx2.S:102
102	../sysdeps/x86_64/multiarch/strcmp-avx2.S: Aucun fichier ou dossier de ce type.
(gdb) bt
#0  __strcmp_avx2 () at ../sysdeps/x86_64/multiarch/strcmp-avx2.S:102
#1  0x00007ffff7e47e0c in IUFindSwitch (svp=svp@entry=0x5555555da318, name=0x555000369573 <error: Cannot access memory at address 0x555000369573>) at ./libs/indicom.c:1364
#2  0x00007ffff7e31d96 in IUUpdateSwitch (svp=svp@entry=0x5555555da318, states=states@entry=0x5555555f88a0, names=names@entry=0x555555603ab0, n=n@entry=4) at ./indidriver.c:186
#3  0x00007ffff7e503f8 in INDI::DefaultDevice::ISNewSwitch (this=this@entry=0x5555555dafc0, dev=dev@entry=0x55555560afa0 "EQMod Mount", name=name@entry=0x55555560aea0 "CONFIG_PROCESS", 
    states=states@entry=0x5555555f88a0, names=names@entry=0x555555603ab0, n=n@entry=4) at ./libs/indibase/defaultdevice.cpp:536
#4  0x00007ffff7e986b7 in INDI::Telescope::ISNewSwitch (this=this@entry=0x5555555dafc0, dev=dev@entry=0x55555560afa0 "EQMod Mount", name=name@entry=0x55555560aea0 "CONFIG_PROCESS", 
    states=states@entry=0x5555555f88a0, names=names@entry=0x555555603ab0, n=n@entry=4) at ./libs/indibase/inditelescope.cpp:1552
#5  0x0000555555576eb7 in EQMod::ISNewSwitch (this=0x5555555dafc0, dev=0x55555560afa0 "EQMod Mount", name=<optimized out>, states=0x5555555f88a0, names=0x555555603ab0, n=4) at ./indi-eqmod/eqmodbase.cpp:3221
#6  0x00007ffff7e4e775 in ISNewSwitch (dev=0x55555560afa0 "EQMod Mount", name=0x55555560aea0 "CONFIG_PROCESS", states=0x5555555f88a0, names=0x555555603ab0, n=4) at ./libs/indibase/defaultdevice.cpp:65
#7  0x00007ffff7e30da0 in dispatch (root=root@entry=0x5555555f0660, msg=msg@entry=0x7fffffffd6e0 "") at ./indidriver.c:944
#8  0x00007ffff7e325e5 in clientMsgCB (fd=<optimized out>, arg=0x5555555e4db0) at ./indidrivermain.c:84
#9  0x00007ffff7e32bbf in callCallback (rfdp=0x7fffffffdf30) at ./eventloop.c:409
#10 oneLoop () at ./eventloop.c:501
#11 0x00007ffff7e32cab in eventLoop () at ./eventloop.c:109
#12 0x00007ffff7e32792 in main (ac=<optimized out>, ac@entry=1, av=av@entry=0x7fffffffe118) at ./indidrivermain.c:131
#13 0x00007ffff7531cb2 in __libc_start_main (main=0x7ffff7e32684 <main>, argc=1, argv=0x7fffffffe118, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe108)
    at ../csu/libc-start.c:314
#14 0x00005555555639ee in _start ()
```

Anthony.

atoy40, 2021-05-24:
After more investigation, I can probably tell that the "bug" is coming from kstars/ekos, because if :
- I launch the gdb session with the eqmod driver
- I connect the driver using indi_setprop

It doesn't crash, and start to send update of  mount coords.

Then if I start kstars/ekos and connect to the indi server, no crash at all. So the problem only appears when connecting the driver from ekos. but why ......

Anthony.

knro, 2021-05-24:
Make sure you're running the latest of both. I use EQMod on daily basis (just used it with EQ8 10 minutes ago) and it works fine.

atoy40, 2021-05-24:
@knro yes, i'm using both from thé stable ppa.

atoy40, 2021-05-25:
Hello,

a new really strange test. I've upgraded my ubuntu from 20.10 to 21.04. Then I removed/purge all indi/kstars packages to be sure all 20.10 packages has been removed.
Then I reinstalll indi/kstars, and a new really strange thing happen :

If I start an ekos profile with "local" indiserver (with just the eqmod and a cdd simulator), the eqmod tabs (in indi panel) only contains two tabs : alignement and connection, and the connection tabs only contains mode and system ports props.

If I manually start an indiserver with thoses two drivers, and then swap my ekos profile to "remote" ... the device appears, and the eqmod is well populated .. and more : It connects perfectly and doesn't crash anymore.

But why almost all properties are not visible when using the local server ??

Here is the log :
[log_18-48-27.txt](https://github.com/indilib/indi/files/6541049/log_18-48-27.txt)

Anthony.

atoy40, 2021-05-25:
Ok my last problem is probably identified, I realize kstars-bleeding is not yet packaged for 21.04. So the version I was using was the 3.4.3 included with this ubuntu.

atoy40, 2021-05-25:
I've also seen a difference : the version of indi available for Ubuntu 21.04 is 1.8.9. So I suspect changes between 1.8.9 and 1.9.0 that can impact serial connection.
The segfault I've shared in the first message seems to happen because the eqmod driver loose tty connection with the mount while Ekos is sending vector updates. One thing I've noticed between 1.8.9 and 1.9.0 is the name of the serial ports. In 1.8.9, system ports are ttyUSBX, but in 1.9.0, this becomes the symlinks in /dev/serial/by-id/
I've tried to connected my EQ6-R Pro using two methods :
- With the Hand control port and a Pegasus eqdir cable at 9600 bauds
- With the integrated USB port of the mount at 115200 bauds.

In both cases the mount disconnects a short time after connexion with 1.9.0. With 1.8.9 .. both works perfectly well.

Is there a big difference in serial connection handling between thoses versions ?
Is it possible to upload last indi and kstars version for 21.04 to the stable ppa ?

Anthony

knro, 2021-05-26:
It used to be available but Launchpad (where we build these packages) had recently some issues so I restricted the stable PPA to LTS releases. Nightlies are still in all Ubuntu releasese so you can try that.

atoy40, 2021-05-28:
Yes, I already check the nightly ppa, but some packages failed to build with 21.04, and some other a not even built for this release.
I started to check why libindi package won't build on 21.04, it seems to have a problem in the linking step :
```
/usr/bin/ld: /tmp/libindidriver.so.1.9.0.3Y30fP.ltrans0.ltrans.o: relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
```

It's strange because all source seems to be compile with the PIC flag

Any idea ?

knro, 2021-06-30:
This is fixed in GIT.

closed 2021-06-30 by knro